RPi-Distro/vlc #1: Upstreaming of the mmal patch

kwizart, 2019-03-29:
HI,

I was wondering if there is a plan to get the mmal patch upstream to vlc's VideoLAN ?
There is nothing much in the headers to know the author, so I'm assuming it's handled in this project.

Hopefully this mmal support can get into vlc-4.0.

Thx in advances.

XECDesign, 2019-04-01:
I don't know what the long term plan is, but the mmal patch is still a WIP.

kwizart, 2019-04-01:
Thx for the answer.

Been in WIP is probably an important state to consider, but it shouldn't prevent this patch serie to be at least "seen" by the VideoLan devs (1).  They will certainly provide comments and review that will help to leave the WIP state on the "libvlc" side of things (if any).

Anyway please try to get them involved early than later.

FYI, I'm interested as a vlc package maintainer for Fedora where we try to maintain a RPI specific port (including this patch). Will be provided at https://rpmfusion.org

(1) https://wiki.videolan.org/git

XECDesign, 2019-04-01:
It's just an issue of finding time to fit it in. It's just one dev working on vlc, chromium and other bits and pieces. The patch is available for anybody who wants to grab it, and we understand the benefits of upstreaming, but I don't know if it's the top priority right now.