RPi-Distro/vlc #2: VLC Video Filters don't work

Jow2meeM, 2019-08-27:
Adjusting the brightness has no effect on the video. 

When I enable 'Image Adjust', I got these debug messages:
```
main debug: Adding 'adjust' as interactive
main debug: looking for video filter module matching "adjust,chain": 51 candidates
main debug: Buffering 0%
main debug: Buffering 83%
adjust debug: Unsupported input chroma (MMAL)
chain debug: Trying to use chroma I420 as middle man
main debug: looking for video converter module matching "any": 20 candidates
chain debug: Trying to use chroma I422 as middle man
main debug: looking for video converter module matching "any": 20 candidates
chain error: Too high level of recursion (3)
main debug: no video converter modules matched
main error: Failed to create video converter
chain debug: Trying to use chroma I0AL as middle man
main debug: looking for video converter module matching "any": 20 candidates
chain error: Too high level of recursion (3)
main debug: no video converter modules matched
main error: Failed to create video converter
chain debug: Trying to use chroma I0AB as middle man
main debug: looking for video converter module matching "any": 20 candidates
chain error: Too high level of recursion (3)
main debug: no video converter modules matched
main error: Failed to create video converter
chain debug: Trying to use chroma I0FL as middle man
main debug: looking for video converter module matching "any": 20 candidates
chain error: Too high level of recursion (3)
main debug: no video converter modules matched
main error: Failed to create video converter
chain debug: Trying to use chroma RV32 as middle man
main debug: looking for video converter module matching "any": 20 candidates
mmal_codec debug: OpenConverter: (hvs) MMAL/OPQV,1920x1088 [(0,0) 1920/1080] sar:1/1->RV32/BGRA,1920x1088 [(0,0) 1920x1080] rgb:0xff0000:0xff00:0xff sar:1/1 (gpu=79691776)
main debug: Stream buffering done (500 ms in 50 ms)
main debug: using video converter module "mmal_codec"
main debug: Filter 'MMAL converterer' (0x61580c98) appended to chain
main debug: looking for video converter module matching "any": 20 candidates
swscale debug: 1920x1080 (1920x1088) chroma: RV32 -> 1920x1080 (1920x1088) chroma: I420 with scaling using Bicubic (good quality)
main debug: using video converter module "swscale"
main debug: Filter 'Swscale' (0x735c5230) appended to chain
main debug: using video converter module "chain"
main debug: Filter 'chain' (0x61580780) appended to chain
main debug: looking for video filter module matching "adjust,chain": 51 candidates
main debug: using video filter module "adjust"
main debug: Filter 'adjust' (0x6150b4f8) appended to chain
main debug: using video filter module "chain"
main debug: Filter 'adjust' (0x615804d0) appended to chain
main debug: Adding a filter to compensate for format changes
main debug: looking for video converter module matching "any": 20 candidates
chain debug: Trying to use chroma I422 as middle man
main debug: looking for video converter module matching "any": 20 candidates
swscale debug: 1920x1080 (1920x1088) chroma: I420 -> 1920x1080 (1920x1088) chroma: I422 with scaling using Bicubic (good quality)
main debug: using video converter module "swscale"
main debug: Filter 'Swscale' (0x6150be20) appended to chain
main debug: looking for video converter module matching "any": 20 candidates
chain debug: Trying to use chroma I420 as middle man
main debug: looking for video converter module matching "any": 20 candidates
swscale debug: 1920x1080 (1920x1088) chroma: I422 -> 1920x1080 (1920x1088) chroma: I420 with scaling using Bicubic (good quality)
main debug: using video converter module "swscale"
main debug: Filter 'Swscale' (0x6153abd8) appended to chain
main debug: looking for video converter module matching "any": 20 candidates
chain error: Too high level of recursion (3)
main debug: no video converter modules matched
main error: Failed to create video converter
main debug: removing module "swscale"
main debug: Filter 0x6153abd8 removed from chain
chain debug: Trying to use chroma I0AL as middle man
main debug: looking for video converter module matching "any": 20 candidates
swscale debug: 1920x1080 (1920x1088) chroma: I422 -> 1920x1080 (1920x1088) chroma: I0AL with scaling using Bicubic (good quality)
[... 587 lines cut ...]
main debug: removing module "swscale"
main debug: Filter 0x6150be20 removed from chain
main debug: no video converter modules matched
main error: Failed to create video converter
main error: Failed to compensate for the format changes, removing all filters
main debug: removing module "chain"
main debug: removing module "chain"
main debug: removing module "mmal_codec"
main debug: Filter 0x61580c98 removed from chain
main debug: removing module "swscale"
main debug: Filter 0x735c5230 removed from chain
main debug: Filter 0x61580780 removed from chain
main debug: removing module "adjust"
main debug: Filter 0x6150b4f8 removed from chain
main debug: Filter 0x615804d0 removed from chain
```

Raspberry Pi 3 Model B Plus Rev 1.3
Raspbian GNU/Linux 10 (buster)
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
VLC media player 3.0.8 Vetinari (revision 3.0.8-0-gf350b6b5a7)

Related: 
raspberrypi/userland#571
oaubert/python-vlc#102

XECDesign, 2019-09-26:
Response from the dev:

> The issue here is that the filter
is trying to insert itself into a pipeline that uses MMAL opaque buffers
from end to end.  It successfully converts out of opaque but has no
filter for returning to opaque.  I can (probably) fix this, but the
workaround would be to force use of the (s/w) avcodec video decoder
rather than our mmal h/w one - this might end up with a faster pipeline
than having to repeatedly convert / copy anyway.

dasl-, 2021-02-11:
Potential workaround: https://forum.videolan.org/viewtopic.php?t=149455