RPi-Distro/vlc #3: VLC & CVLC don't work in fullscreen on screen HDMI-2 on Raspberry Pi 4

mobluse, 2019-08-29:
VLC & CVLC don't work in fullscreen on screen HDMI-2 on Raspberry Pi 4 B with full-upgraded Raspbian Buster.
```
vlc --fullscreen /opt/vc/src/hello_pi/hello_video/test.h264
cvlc --fullscreen /opt/vc/src/hello_pi/hello_video/test.h264
```This is so even if I have the LXTerminal in the screen connected to HDMI-2. What happens is that the video is shown on HDMI-1 anyway.

```
$ vlc --version
VLC media player 3.0.8 Vetinari (revision 3.0.8-0-gf350b6b5a7)
VLC version 3.0.8 Vetinari (3.0.8-0-gf350b6b5a7)
Compiled by serge on arm-build.pitowers.org (Aug 21 2019 14:42:50)
Compiler: gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)
```

XECDesign, 2019-08-29:
This doesn't seem to be anything specific to our patch. If you go to preferences and select the appropriate output device, it will use that.

mobluse, 2019-08-31:
Even if I choose HDMI-2 in preferences on Video tab it uses HDMI-1 for fullscreen. I also tried HDMI-1 in preferences and that shows on HDMI-1, so it's not that the cables are crossed.

BTW I didn't know that you could sett fullscreen output in preferences, but now that I tried it, it didn't work.

I use two 1280x1024 screens.

XECDesign, 2019-08-31:
It worked when I tried it, but I'll try to reproduce it again on Monday.

XECDesign, 2019-09-11:
Initially, I had my 'Output' set to 'automatic'. After testing a few of the other options, I could see the problem and switching back to automatic didn't help. Looks like some of them work and some don't. I'll pass it on, thank you.

XECDesign, 2019-09-26:
A fix is in the works.

Luciano2018, 2019-12-14:
Maybe my answer is late but, VLC has a menu to select fullscreen Video device, doesn't work. But Output option Yes, it works!!!
Tools->Preferences->Video->Output->OpenGL for Embedded Systems 2 video output

With this options, and if we want, the option Fullscreen Video Device it works, in the 3 modes:
Automatic (the player go fullscreen in the screen that it is)
HDMI-1 (the player go fullscreen in the first screen, it is or not the player in that screen)
HDMI-2 (the player go fullscreen in the second screen, it is or not the player in that screen)

Hope this little guide help!

das800, 2020-07-19:
this solved the same problem for me. vlc on raspberry pi 4 would show blank video when going into fullscreen and this solution worked

Bastoon38, 2020-11-04:
Hello,

UP!

I have the same problem that @mobluse explain...
Has anyone found a solution from the command line?

Thank you!

jc-kynesim, 2020-11-04:
--mmal-display hdmi-2
should do the trick

jc-kynesim, 2020-11-04:
vlc -H will list all the options - pipe the output to a file as it is huge and search through it for --mmal- to find all the Pi (mmal) specific switches.  Beware that some of them are debug grade switches so use with care.

Bastoon38, 2020-11-12:
Yes !

Many thanks 👍

kleurbleur, 2022-02-05:
The above mentioned solutions don't work for me on RPI4 and fresh install of bullseye. Also VLC is not capable of looping a video file (big buck bunny) without stuttering. 

Solution:  edit /boot/config.txt the line dtoverlay=vc4-kms-v3d to dtoverlay=vc4-fkms-v3d 
Then run rpi-update and you can use omxplayer again. Smooth as silk.