RPi-Distro/vlc #4: MediaListPlayer window interruptions while using python-vlc

Jow2meeM, 2019-09-19:
When using python-vlc the player closes and reopens between playlist items. This occurs on Raspian but not on another computer with Ubuntu. Do you have an idea how to prevent this behaviour and have a seamless transition instead?

Example code:
```
#!/usr/bin/env python3
import time, vlc

instance = vlc.Instance()
media_list = instance.media_list_new(['./video_01.mp4', './video_02.mp4'])
list_player = instance.media_list_player_new()
list_player.set_media_list(media_list)

list_player.play()
time.sleep(5)
list_player.next()
list_player.play()

while True:
    pass
```
python-vlc 3.0.7110
Raspberry Pi 3 Model B Plus Rev 1.3
Raspbian GNU/Linux 10 (buster)
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
VLC media player 3.0.8 Vetinari (revision 3.0.8-0-gf350b6b5a7)

See also: oaubert/python-vlc#106

XECDesign, 2019-09-20:
I tried your first snippet and I'm not seeing the behaviour you've described.

Jow2meeM, 2019-09-21:
Unfortunately even on a freshly installed Raspian the described behaviour persists.

Here is the error log for code snippet two (python-vlc repeat). For better readability I divided the log in four parts (Initialization, Playback, Repeat1, Repeat2).

Initialization:
```
[013436d8] main libvlc debug: VLC media player - 3.0.8 Vetinari
[013436d8] main libvlc debug: Copyright © 1996-2019 the VideoLAN team
[013436d8] main libvlc debug: revision 3.0.8-0-gf350b6b5a7
[013436d8] main libvlc debug: configured with ./configure  '--build=arm-linux-gnueabihf' '--prefix=/usr' '--includedir=${prefix}/include' '--mandir=${prefix}/share/man' '--infodir=${prefix}/share/info' '--sysconfdir=/etc' '--localstatedir=/var' '--disable-silent-rules' '--libdir=${prefix}/lib/arm-linux-gnueabihf' '--runstatedir=/run' '--disable-maintainer-mode' '--disable-dependency-tracking' '--disable-debug' '--config-cache' '--disable-update-check' '--enable-fast-install' '--docdir=/usr/share/doc/vlc' '--with-binary-version=3.0.8-0+deb10u1+rpt1' '--enable-a52' '--enable-aa' '--enable-aom' '--enable-aribsub' '--enable-avahi' '--enable-bluray' '--enable-caca' '--enable-chromaprint' '--enable-chromecast' '--enable-dbus' '--enable-dca' '--enable-dvbpsi' '--enable-dvdnav' '--enable-faad' '--enable-flac' '--enable-fluidsynth' '--enable-freetype' '--enable-fribidi' '--enable-gles2' '--enable-gnutls' '--enable-harfbuzz' '--enable-jack' '--enable-kate' '--enable-libass' '--enable-libmpeg2' '--enable-libxml2' '--enable-lirc' '--enable-live555' '--enable-mad' '--enable-matroska' '--enable-mod' '--enable-mpc' '--enable-mpg123' '--enable-mtp' '--enable-ncurses' '--enable-notify' '--enable-ogg' '--enable-opus' '--enable-pulse' '--enable-qt' '--enable-realrtsp' '--enable-samplerate' '--enable-sdl-image' '--enable-sftp' '--enable-shine' '--enable-shout' '--enable-skins2' '--enable-sndio' '--enable-soxr' '--enable-spatialaudio' '--enable-speex' '--enable-svg' '--enable-svgdec' '--enable-taglib' '--enable-theora' '--enable-twolame' '--enable-upnp' '--enable-vdpau' '--enable-vnc' '--enable-vorbis' '--enable-x264' '--enable-x265' '--enable-zvbi' '--with-kde-solid=/usr/share/solid/actions/' '--disable-d3d11va' '--disable-decklink' '--disable-directx' '--disable-dsm' '--disable-dxva2' '--disable-fdkaac' '--disable-fluidlite' '--disable-freerdp' '--disable-goom' '--disable-gst-decode' '--disable-libtar' '--disable-macosx' '--disable-macosx-avfoundation' '--disable-macosx-qtkit' '--disable-mfx' '--disable-opencv' '--disable-projectm' '--disable-schroedinger' '--disable-sparkle' '--disable-srt' '--disable-telx' '--disable-vpx' '--disable-vsxu' '--disable-wasapi' '--enable-alsa' '--enable-dc1394' '--enable-dv1394' '--enable-libplacebo' '--enable-linsys' '--enable-nfs' '--enable-omxil' '--enable-udev' '--enable-v4l2' '--enable-wayland' '--enable-libva' '--enable-vcd' '--enable-smbclient' '--disable-oss' '--disable-crystalhd' '--disable-mmx' '--disable-sse' '--enable-neon' '--enable-mmal' '--disable-mmal-avcodec' '--disable-altivec' 'build_alias=arm-linux-gnueabihf' 'CFLAGS=-g -O2 -fdebug-prefix-map=/build/vlc-wDSKZq/vlc-3.0.8=. -fstack-protector-strong -Wformat -Werror=format-security ' 'LDFLAGS=-Wl,-z,relro -Wl,-z,now -Wl,--as-needed' 'CPPFLAGS=-Wdate-time -D_FORTIFY_SOURCE=2' 'CXXFLAGS=-g -O2 -fdebug-prefix-map=/build/vlc-wDSKZq/vlc-3.0.8=. -fstack-protector-strong -Wformat -Werror=format-security ' 'OBJCFLAGS=-g -O2 -fdebug-prefix-map=/build/vlc-wDSKZq/vlc-3.0.8=. -fstack-protector-strong -Wformat -Werror=format-security'
[013436d8] main libvlc debug: searching plug-in modules
[013436d8] main libvlc debug: loading plugins cache file /usr/lib/arm-linux-gnueabihf/vlc/plugins/plugins.dat
[013436d8] main libvlc debug: recursively browsing `/usr/lib/arm-linux-gnueabihf/vlc/plugins'
[013436d8] main libvlc debug: plug-ins loaded: 528 modules
[012ddf80] main logger debug: looking for logger module matching "any": 4 candidates
[012ddf80] main logger debug: using logger module "console"
[013436d8] main libvlc debug: translation test: code is "C"
[013bf6a8] main keystore debug: looking for keystore module matching "memory": 4 candidates
[013bf6a8] main keystore debug: using keystore module "memory"
[013436d8] main libvlc debug: CPU has capabilities ARM_NEON FPU 
[0128ad80] main playlist debug: creating audio output
[013b45d8] main audio output debug: looking for audio output module matching "any": 6 candidates
[013b45d8] vlcpulse audio output debug: using library version 12.2.0
[013b45d8] vlcpulse audio output debug:  (compiled with version 12.2.0, protocol 32)
[013b45d8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[013b45d8] alsa audio output debug: Available ALSA PCM devices:
[013b45d8] alsa audio output debug: Discard all samples (playback) or generate zero samples (capture) (null)
[013b45d8] alsa audio output debug: bcm2835 ALSA, bcm2835 ALSA Default Audio Device (default:CARD=ALSA)
[013b45d8] alsa audio output debug: bcm2835 ALSA, bcm2835 ALSA Default Audio Device (sysdefault:CARD=ALSA)
[013b45d8] alsa audio output debug: bcm2835 ALSA, bcm2835 ALSA Direct sample mixing device (dmix:CARD=ALSA,DEV=0)
[013b45d8] alsa audio output debug: bcm2835 ALSA, bcm2835 IEC958/HDMI Direct sample mixing device (dmix:CARD=ALSA,DEV=1)
[013b45d8] alsa audio output debug: bcm2835 ALSA, bcm2835 IEC958/HDMI1 Direct sample mixing device (dmix:CARD=ALSA,DEV=2)
[... cut ...]
[6d438240] mmal_xsplitter vout display debug: Change size: 1824, 984: mmal_vout=0x6d4013f0, want_mmal=1, fs=1
[74a141e8] main video output warning: picture is too late to be displayed (missing 110 ms)
[74a141e8] main video output warning: picture is too late to be displayed (missing 70 ms)
[74a141e8] main video output warning: picture is too late to be displayed (missing 30 ms)
[6d4013f0] mmal_vout generic debug: 1920x1080 -> 1920x1079 @ 0,0
[6d438240] main vout display debug: auto hiding mouse cursor
```

Jow2meeM, 2019-09-22:
I filled a ticket on the VLC bug tracker, which was just closed. They think the problem is the 'third party "mmal_xsplitter" plugin' (https://trac.videolan.org/vlc/ticket/22811).

XECDesign, 2019-09-23:
If I use VLC through that python script, it doesn't use MMAL X11 splitter for me. But I'll pass the issue on, now that you've narrowed it down. Thank you.

XECDesign, 2019-09-26:
Response:

> There are a very
limited number of cases where the video output isn't closed & reopened
(video format & size needs to nearly identical between the streams) and
if we hit one of them then all is normally well but if not then we are
torn down & restarted.  In some cases (on a fairly fast m/c where the
decode starts up very quickly) this is imperceptible, but if we were
fullscreen then flicker is likely - I can't really refuse to shutdown.
It is plausible that I've missed something here but it doesn't look like
a straightforward bug.

Jow2meeM, 2019-09-28:
I see. Thank you both for having taken the time to tend to my problem.

dronus, 2020-06-23:
> There are a very
 > limited number of cases where the video output isn't closed & reopened
 > (video format & size needs to nearly identical between the streams) and
 > if we hit one of them then all is normally well but if not then we are
 > torn down & restarted.  

Doesn't quite fit my experience. 

It happens almost always for H.264 video it seems, even if the same file is played multiple times by the playlist. In contrast, it reliable does not happen between some H.265 videos of same resolution. 

So I would look into the differences between the H.264 and H.265 setup.

ShaoMingya, 2020-11-05:
pi4B，python3.7，第一遍播放结束和第二遍播放前，也会出现1秒左右停顿。

`#!/usr/bin/env python3
import time, vlc

instance = vlc.Instance("--verbose=2")
media_list = instance.media_list_new(['/home/pi/Videos/Gallium_drops.mp4', '/home/pi/Videos/Gallium_drops.mp4'])
list_player = instance.media_list_player_new()
list_player.set_media_list(media_list)

list_player.play()
time.sleep(5)
list_player.next()

while True:
    time.sleep(300)
    
`
out lines：

`pi@raspberrypi:~ $ /home/pi/python_codes/py_vlc/vlc_test.py
[020ab008] main libvlc debug: VLC media player - 3.0.11 Vetinari
[020ab008] main libvlc debug: Copyright © 1996-2020 the VideoLAN team
[020ab008] main libvlc debug: revision 3.0.11-0-gdc0c5ced72
[020ab008] main libvlc debug: configured with ./configure  '--build=arm-linux-gnueabihf' '--prefix=/usr' '--includedir=${prefix}/include' '--mandir=${prefix}/share/man' '--infodir=${prefix}/share/info' '--sysconfdir=/etc' '--localstatedir=/var' '--disable-silent-rules' '--libdir=${prefix}/lib/arm-linux-gnueabihf' '--runstatedir=/run' '--disable-maintainer-mode' '--disable-dependency-tracking' '--disable-debug' '--config-cache' '--disable-update-check' '--enable-fast-install' '--docdir=/usr/share/doc/vlc' '--with-binary-version=3.0.11-0+deb10u1+rpt3' '--enable-a52' '--enable-aa' '--enable-aribsub' '--enable-avahi' '--enable-bluray' '--enable-caca' '--enable-chromaprint' '--enable-chromecast' '--enable-dbus' '--enable-dca' '--enable-dvbpsi' '--enable-dvdnav' '--enable-faad' '--enable-flac' '--enable-fluidsynth' '--enable-freetype' '--enable-fribidi' '--enable-gles2' '--enable-gnutls' '--enable-harfbuzz' '--enable-jack' '--enable-kate' '--enable-libass' '--enable-libmpeg2' '--enable-libxml2' '--enable-lirc' '--enable-live555' '--enable-mad' '--enable-matroska' '--enable-mod' '--enable-mpc' '--enable-mpg123' '--enable-mtp' '--enable-ncurses' '--enable-notify' '--enable-ogg' '--enable-opus' '--enable-pulse' '--enable-qt' '--enable-realrtsp' '--enable-samplerate' '--enable-sdl-image' '--enable-sftp' '--enable-shine' '--enable-shout' '--enable-skins2' '--enable-sndio' '--enable-soxr' '--enable-spatialaudio' '--enable-speex' '--enable-svg' '--enable-svgdec' '--enable-taglib' '--enable-theora' '--enable-twolame' '--enable-upnp' '--enable-vnc' '--enable-vorbis' '--enable-x264' '--enable-x265' '--enable-zvbi' '--with-kde-solid=/usr/share/solid/actions/' '--disable-d3d11va' '--disable-decklink' '--disable-directx' '--disable-dsm' '--disable-dxva2' '--disable-fdkaac' '--disable-fluidlite' '--disable-freerdp' '--disable-goom' '--disable-gst-decode' '--disable-libtar' '--disable-macosx' '--disable-macosx-avfoundation' '--disable-macosx-qtkit' '--disable-microdns' '--disable-mfx' '--disable-opencv' '--disable-projectm' '--disable-schroedinger' '--disable-sparkle' '--disable-srt' '--disable-telx' '--disable-vpx' '--disable-vsxu' '--disable-wasapi' '--enable-alsa' '--enable-dc1394' '--enable-dv1394' '--enable-libplacebo' '--enable-linsys' '--enable-nfs' '--enable-omxil' '--enable-udev' '--enable-v4l2' '--enable-wayland' '--enable-libva' '--enable-vcd' '--enable-smbclient' '--disable-oss' '--disable-crystalhd' '--disable-mmx' '--disable-sse' '--enable-neon' '--enable-mmal' '--enable-mmal-avcodec' '--disable-vdpau' '--disable-libva' '--enable-dav1d' '--disable-aom' '--disable-altivec' 'build_alias=arm-linux-gnueabihf' 'CFLAGS=-g -O2 -fdebug-prefix-map=/build/vlc-hrnPHl/vlc-3.0.11=. -fstack-protector-strong -Wformat -Werror=format-security ' 'LDFLAGS=-Wl,-z,relro -Wl,-z,now -Wl,--as-needed' 'CPPFLAGS=-Wdate-time -D_FORTIFY_SOURCE=2' 'CXXFLAGS=-g -O2 -fdebug-prefix-map=/build/vlc-hrnPHl/vlc-3.0.11=. -fstack-protector-strong -Wformat -Werror=format-security ' 'OBJCFLAGS=-g -O2 -fdebug-prefix-map=/build/vlc-hrnPHl/vlc-3.0.11=. -fstack-protector-strong -Wformat -Werror=format-security'
[020ab008] main libvlc debug: searching plug-in modules
[020ab008] main libvlc debug: loading plugins cache file /usr/lib/arm-linux-gnueabihf/vlc/plugins/plugins.dat
[020ab008] main libvlc debug: recursively browsing `/usr/lib/arm-linux-gnueabihf/vlc/plugins'
[020ab008] main libvlc debug: plug-ins loaded: 504 modules
[020b94f8] main logger debug: looking for logger module matching "any": 4 candidates
[020b94f8] main logger debug: using logger module "console"
[020ab008] main libvlc debug: translation test: code is "C"
[02123c28] main keystore debug: looking for keystore module matching "memory": 4 candidates
[02123c28] main keystore debug: using keystore module "memory"
[020ab008] main libvlc debug: CPU has capabilities ARM_NEON FPU 
[01fecd70] main playlist debug: creating audio output
[02118dc8] main audio output debug: looking for audio output module matching "any": 6 candidates
[02118dc8] vlcpulse audio output debug: using library version 12.2.0
[02118dc8] vlcpulse audio output debug:  (compiled with version 12.2.0, protocol 32)
[02118dc8] vlcpulse audio output debug: connected locally to unix:/run/user/1000/pulse/native as client #8
[02118dc8] vlcpulse audio output debug: using protocol 32, server protocol 32
[02118dc8] pulse audio output debug: adding sink 0: auto_null (伪输出)
[02118dc8] main audio output debug: using audio output module "pulse"
[01fecd70] main playlist debug: keeping audio output
[a4b00638] main input debug: Creating an input for preparsing 'vlc'
[02131bf8] main meta fetcher debug: looking for meta fetcher module matching "any": 1 candidates
[02131758] main generic debug: creating audio output
[... cut ...]
[02141d68] main input debug: Program doesn't contain anymore ES
[91c3ab38] main stream debug: removing module "record"
[91c13590] main stream debug: removing module "cache_read"
[91c00d20] main stream debug: removing module "filesystem"
[02134778] pulse audio output debug: changing sink 0: auto_null (伪输出)
[02118dc8] pulse audio output debug: changing sink 0: auto_null (伪输出)
[02118dc8] pulse audio output debug: changing sink 0: auto_null (伪输出)
[02134778] pulse audio output debug: changing sink 0: auto_null (伪输出)
`

ShaoMingya, 2020-11-05:
When "--verbose= 0":

```
pi@raspberrypi:~ $ /home/pi/python_codes/py_vlc/vlc_test.py
[91a14330] mmal_codec decoder: VCSM init succeeded: Legacy
[91a14330] main decoder error: buffer deadlock prevented
libEGL warning: DRI2: failed to authenticate
[b4d06fd8] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
[91a40aa0] mmal_codec decoder: VCSM init succeeded: Legacy
[91a40aa0] main decoder error: buffer deadlock prevented
libEGL warning: DRI2: failed to authenticate
[9170bd60] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
```

burdin-an, 2021-10-04:
Hello.
I have the same problem on RP4.
When playing H.264 or AV1 files.
From H.265 the window does not close.
 /usr/bin/vlc -I http --http-password 123 --no-playlist-autostart --playlist-tree --playlist-enqueue  --no-metadata-network-access --recursive=expand --daemon  --no-video-title-show  --one-instance --quiet   --fullscreen --repeat --aspect-ratio=16:9  --mmal-display hdmi-2 /home/pi/Videos/

josejairobaltazargallegos-afk, 2026-10-10:
Hola