RPi-Distro/vlc #6: Problem with VLC raspberry 4 and TV HDMI input

rkl099, 2019-11-14:
Fresh install with all updates on a Pi4 using a HDMI to VGA adapter connected to the TV-set.
Resolution 1360*768 autoselect. VLC starts OK.
Adjust to 1920*1080. VLC starts OK

Connect Pi4 direct to HDMI input on TV. Integrity notice and VLC logo are displayed magnified appr. 30 times. The same result if the TV is connected to HDMI2 output.

Starting “vlc” from command line with no file or a video file shows the same result as above.
Note: VLC does not crash, it is just busy. It reacts to <ctr> Q if you move the integrity window inside the screen and delete it (or just type <return>).

Starting “cvlc” from the command line with a video file works ok.

For me it seems that VLC does not sense the same screen resolution as Rasbian desktop. It may also be a problem with the Qt implementation or the VLC Qt interface.

ghost123uk, 2021-03-06:
I too have this issue.
It is repeated on every install of the latest Raspberry OS (full version)
It can be fixed by adding hdmi_ignore_edid=0xa5000080
into the /boot/config.txt file (I use my Linux PC to edit the config.txt file).
The issue is present on any of the 4 screens I use = 3 x TV and 1 x "proper" PC monitor, all using HDMI.
I am a relative newcomer to all of this, so cannot add much more info.

rkl099, 2021-03-07:
I lost the menu bar on HDMI1, but on HDMI0 it works!
Solves the problem until VLC is updated.
Thank you.

popcornmix, 2021-03-09:
Have a read of [this thread](https://www.raspberrypi.org/forums/viewtopic.php?f=35&t=245682) and see if you have dodgy display dimensions coming from your display's edid.