RPi-Distro/vlc #8: MJPEG in AVI supposedly doesn't work

6by9, 2020-05-12:
Reported at https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=263875

jc-kynesim, 2020-05-12:
I'll look into it

Botspot, 2020-05-12:
It looks like VLC throws an error:

```pi@raspberrypi:~ $ vlc /home/pi/Downloads/MOVI0006.avi
VLC media player 3.0.10 Vetinari (revision 3.0.10-0-g7f145afa84)
[01c31810] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[01bb6b58] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
qt5ct: using qt5ct plugin
qt5ct: D-Bus global menu: no
qt5ct: D-Bus system tray: no
[aa115968] mmal_codec decoder: VCSM init succeeded: CMA
[aa115968] main decoder error: buffer deadlock prevented
QObject::~QObject: Timers cannot be stopped from another thread
pi@raspberrypi:~ $ ```

jc-kynesim, 2020-06-03:
Sorry no.  I've been (and still am) busy.  I can reproduce & I will get round to it, hopefully within the next week or so.

6by9, 2020-06-24:
Problem found. The fact it failed in omxplayer too pointed the finger at the firmware, and indeed it was a firmware issue.

Your particular stream uses multiple 0xFF padding bytes between JPEG markers, and the decoder wasn't allowing for this and threw errors. I'm about to pusha patch internally, so it should appear via rpi-update at some point.

Botspot, 2020-06-24:
Thank you, 6by9!
Any idea if this firmware fix will also solve issue #9 ?

6by9, 2020-06-24:
> Any idea if this firmware fix will also solve issue #9 ?

Totally unrelated. HEVC doesn't go through the firmware at all

Botspot, 2020-09-04:
Works perfectly!
Please close this issue.

closed 2020-09-04 by XECDesign