RPi-Distro/vlc #9: Seeking in video freezes it

Botspot, 2020-06-08:
I think this only happens in HVEC videos.
Sample video [here](https://www.dropbox.com/s/42tqvsmfy5hxrbt/Apollo%2013%20%281995%29%20%281%29-001.mkv?dl=0).
## Procedure:
Download, then play with vlc.
## Expected behavior:
 - Video plays fine.
 - Clicking somewhere on the progress bar to seek the video works fine.
## Actual behavior:
 - Video plays fine.
 - Clicking the progress bar to seek freezes the video on whatever frame it was playing last.
 - Pressing Play/Pause has no effect.
 - Only solution is to restart VLC (and not seek the video)

popcornmix, 2020-06-25:
@jc-kynesim I can reliably reproduce this one, so it's worth a look.

jc-kynesim, 2020-06-25:
Yup - will do.  I'm working through VL:C issues currently so should be with it shortly.

jc-kynesim, 2020-06-29:
Will be fixed in the next release.  When seeking some frames get flushed causing decode errors. It appears it is a bad idea to report these to VLC - much better to just go "Nothing to see here - pass along" and pretend nothing has occurred.

Botspot, 2020-09-04:
Works perfectly!

closed 2020-09-04 by Botspot