RPi-Distro/vlc #10: Hardware accelerated playback on framebuffer

SolidTux, 2020-06-12:
Using VLC to play media files with the framebuffer output module works as long as VLC doesn't try to use the mmal decoder. omxplayer works fine in this case. I get the following output from VLC

```
[01a17498] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[01a33310] main interface error: no suitable interface module
[0199bb58] main libvlc error: interface "globalhotkeys,none" initialization failed
[01a33310] dummy interface: using the dummy interface module...
[6d3496e8] mmal_codec decoder: VCSM init succeeded: Legacy
[6d3496e8] main decoder error: buffer deadlock prevented
```

Is there a way to get hardware accelerated decoding with VLC while using the framebuffer?

jc-kynesim, 2020-06-15:
What evidence do you have that VLC isn't using the h/w decoder?  The "main decoder error: buffer deadlock prevented" error only appears to indicate that the codec is capable of sinking a lot of input frames before generating any output (there are deep input queues on the h/w decoder) and doesn't indicate that the codec isn't working.
If you can run VLC from the command line with the "-vv" option added you will get a _lot_ more debug which will give me more clues as to whether VLC is using the h/w codec and if it isn't then why not.

SolidTux, 2020-06-15:
Sorry, should have explained it a bit better! The problem isn't, that VLC isn't using the hardware decoder (I assume it does because of the `mmal_codec` line in the log), but that there is no output.

Here is a log for `cvlc -vv -V fb video.mp4`: [vlc.log](https://github.com/RPi-Distro/vlc/files/4779827/vlc.log)

jc-kynesim, 2020-06-29:
Sorry about the delay in replying - I've looked at your log and it looks like one of the cases where the display is too slow & VLC, instead of showing the odd frame every now and again, opts to show nothing.  Does it work better with a lower resolution video?  Any particular reason for wanting framebuffer output rather than mmal_vout?