RPi-Distro/vlc #11: SRT input support

its27, 2020-06-29:
Hi,

I have been testing the latest VLC release on the RPI3 and it works great, the only problem is that it can't open SRT:// URLs while the same version in my desktop PC can.

Are you able to add the SRT plugin?

Many Thanks,

jc-kynesim, 2020-07-01:
Can you give an example of something that doesn't work? At least in local playback .srt files give subtitles to a similarly named video file.

its27, 2020-07-01:
Sure.

The issue is with the secure reliable transport protocol rather than SRT subtitles, please see the forum link below.

https://forum.videolan.org/viewtopic.php?t=143500

its27, 2020-07-01:
https://www.srtalliance.org/srt-streams-vlc-3-0/

jc-kynesim, 2020-07-03:
Ta for the links - the issue is that the VLC SRT module relies on an external SRT package that doesn't appear to exist for our version of Debian.

scharette88, 2020-07-14:
Hey guys,

+1 here for SRT support in this VLC. Performance and latency of SRT on unstable network is really impressive, it would be a great feature to have it supported in this build,

Thanks all for all your hard work, it make the RPI a very powerful media player !

closed 2021-01-03 by its27

XECDesign, 2021-01-04:
@jc-kynesim I can take a look at adding the package. Would you happen to have a link for it?

llamafilm, 2021-01-15:
The code to support this is already in here. You just need to install libsrt and add —enable-srt when you build. But there are some issues, see #34.