RPi-Distro/vlc #12: Get cvlc to have omxplayer functionality of --live --layer --crop and --win.

KenT2, 2020-07-01:
Continued from 

https://github.com/popcornmix/omxplayer/issues/778

which is an omxplayer issue.

popcornmix, 2020-07-01:
> what functionality are you expecting from --crop & and --live (and in the case of crop how do you expect it to interact with scaling & rotation (both of source & screen))? Oh! and where do you hope subtitles turn up on a cropped image (this may be outside my control but I'd like a guess as to what you are expecting)?

I'd imagine --crop applies to the untransformed/scaled source video. 
--win is in screen coordinates after transform/scaling
I believe subtitles should use --win coordinates as the canvas and should ignore --crop

KenT2, 2020-07-02:
I am trying to rotate the video by using the VLC filter option 
         ' --video-filter=rotate --rotate-angle=90.0 '
         ' --video-filter=transform --transform-type=90 ' 

both produce lots of error messages with mmal-vout and no video on display. Is rotate/transform something you will need to implement in your code?

jc-kynesim, 2020-07-02:
Almost any filter is going to be bad news. They involve converting the frame into something that software likes and back again & VLCs filter construction gets confused very readily.  Also - if you try to do any frame manipulation on a Pi in s/w that isn't on a trivial sized frame you are going to run out of processing power.  The current vlc does support rotation coded in the bitstream & will attempt to match the current X rotation (set via xrandr) - you can also use "--mmal-vout-transform auto, 0, 90, 180, 270, hflip, vflip, transpose, antitranspose" to override the X rotation setting.  This does the rotation in h/w.

xAgWA, 2020-07-16:
@jc-kynesim where do you set the --mmal-vout-transform option? I'm not able to find anything about it

KenT2, 2020-07-16:
vlc -H 
The output is very long but its there somewhere

jc-kynesim, 2020-07-16:
I was about to say that!  I suggest you pipe the output (of vlc -H) to a file and search for it.

xAgWA, 2020-07-16:
I only see --transform-type which doesn't have any effect

jc-kynesim, 2020-07-16:
Are you sure you typed "-H" (capital H)? Are there any mmal options in the list at all "vlc -H | grep mmal"?

xAgWA, 2020-07-16:
Yes i ran vlc -H

Here is the output of vlc -H | grep mmal

VLC media player 3.0.11 Vetinari (revision 1.0.6-1611-gbf4bc6b99)
 MMAL-based decoder plugin for Raspberry Pi (mmal_codec)
      --mmal-opaque, --no-mmal-opaque 
 MMAL-based deinterlace filter plugin (mmal_deinterlace)
      --mmal-deinterlace-adv-qpu, --no-mmal-deinterlace-adv-qpu 
 MMAL-based vout plugin for Raspberry Pi (mmal_vout)
      --mmal-layer <integer>     VideoCore layer where the video is displayed.
      --mmal-blank-background, --no-mmal-blank-background 
      --mmal-adjust-refreshrate, --no-mmal-adjust-refreshrate 
      --mmal-native-interlaced, --no-mmal-native-interlaced 
  -V, --vout {any,gl,xcb_x11,mmal_vout,fb,caca,vdpau_display,flaschen,vmem,yuv,vdummy,vdummy,none}

jc-kynesim, 2020-07-16:
OK - that isn't the list I was expecting to see.  Is that the default VLC shipped with raspbian?

xAgWA, 2020-07-16:
No i compiled it from this repo

KenT2, 2020-07-16:
I updated Raspbian a few weeks ago, until then the transform did not appear in vlc -H

Here it is:

      --mmal-vout-transform <string> 
                                 Video transform for Rpi fullscreen.
          Video transform for Rpi fullscreen.Transforms availible: auto, 0, 90,
          180, 270, hflip, vflip, transpose, antitranspose

jc-kynesim, 2020-07-16:
Did you build it as a debian package (i.e. with all the patches in the debian subdir) or just the base tree?  Most of the mmal stuff is in the patches.

xAgWA, 2020-07-16:
I did't build it as a debian package. How do i do that?

jc-kynesim, 2020-07-16:
That is a question with a long answer. https://www.debian.org/doc/manuals/maint-guide/build.en.html gives the official answer.
Some of the problem is that this repo is for generating the .debs for distribution rather than general purpose builds.  Is there some reason why you want to build it yourself rather than use the prebuilt binaries?

popcornmix, 2020-07-16:
You can also use apt to install the (patched) source tree the package was built from.

DougShuffield, 2020-07-16:
Should I expect to see any mmal options on Stretch?  'cvlc -H | grep mmal' produces this:

pi@cam1_rx:~ $ cvlc -H | grep mmal
VLC media player 3.0.11 Vetinari (revision 3.0.11-0-gdc0c5ced72)
pi@cam1_rx:~ $

xAgWA, 2020-07-17:
I'm getting the same output for the latest version installed via apt-get on stretch. This is the reason i'm building it myself so i can run it on stretch

DougShuffield, 2020-07-17:
@jc-kynesim It is the rasbian binary updated yesterday.  Did the full apt-get update && apt-get upgrade && apt-get dist-upgrade.

popcornmix, 2020-07-18:
stretch? Did you mean that? We've been on buster for over a year. Stretch isn't supported by us and won't have the latest vlc.

DougShuffield, 2020-07-18:
Yes.  This is an embedded system with a fairly large fielded install base all over the country, so we are somewhat hesitant to upgrade in place.  Not that we won't, just would rather not to keep variables to a minimum.  If required I will begin looking into it. I was hoping to be able to use stretch. Let me know if it is out of the question or somehow impossible.

popcornmix, 2020-07-18:
Installing vlc on stretch isn't something we can help you with. All things are possible with enough effort but it's not trivial (you can try installing a buster deb file which will no doubt have many dependencies that are not satisfied and you can try installing other buster packages to fix it up). Or you can try building from source (which may or may not have dependency issues).

Probably best to start with a buster image and test if vlc does everything you need. If so, you can investigate whether installing it on stretch or upgrading the stretch systems is easier.

DougShuffield, 2020-07-20:
@popcornmix I have spun up a new pi 3B+ with latest buster and vlc.  What would be the equivalent commands for clvc playing on mmal layers for the following omxplayer commands:

Camera TCP source (h264 video 1640x922,, no audio):
omxplayer -n -1 --crop 547,0,1094,922 --win 0,0,640,1080 --layer 20 tcp://192.168.42.21:3333

Gameplay UDP source: (h264 video 1920x1080, mp2 audio):
omxplayer --crop 320,0,1600,1080 --win 640,0,1920,1080 --layer 10 udp://@239.255.42.49:5004?overrun_nonfatal=1

Seems I need some combination of --vout mmal_vout, --mmal-layer, --width, --height, --video-x, --video-y, and then possible some input codec parameters or audio output parameters, but the magic incantation is currently escaping me.  Any help is much appreciated.

popcornmix, 2020-07-21:
> I've got windowing & transparency working. I'm working on crop. If "live" doesn't just work with the base vlc then I'm unlikely to find a fix for it in this release.

I don't think there's been a release with these features in, so you'll have to wait for that.
@jc-kynesim is on holiday this week.

DougShuffield, 2020-08-05:
@jc-kynesim Is there any ETA on the release that will include these features?

jc-kynesim, 2020-08-05:
I'll check - I'm not quite sure what has happened (I don't generate the releases)

DougShuffield, 2020-08-07:
Performed an 'apt-get update && apt install vlc -y' and got the newest version on buster which looks to have the updates.  I am half way there and was able to pretty quickly determine how to get the camera working.  This is streaming from another Pi 3b+ running raspivid listening on port 3333:

original omxplayer command:  
omxplayer -n -1 --threshold 0.01 --fps 30 --crop 547,0,1094,922 --win 0,0,640,1080 --layer 20 tcp://192.168.42.21:3333

new cvlc command:  
cvlc -f --crop 547:922 --vout mmal_vout --mmal-layer 20 --mmal-vout-window 640x1080+0+0 tcp/h264://192.168.42.21:3333

[image]

It does seem that the cvlc --crop option may be more limited than the equivalent omxplayer option as the cvlc command will only take the center of the image using the x:y ratio instead of being able to specify x1,y1,x2,y2 coordinates with the omxplayer option, but I only need the center anyway so it will work for me.  I may be missing some other option with cvlc that would allow the more complete and generic crop function of omxplayer.

Also, I am having issues getting the second udp stream to work so I can't test the other half of the split screen solution just yet.  I believe there is some issue with vlc playing my udp stream at all, so I will take some more time to figure that out before finalizing the split screen solution with two instances of cvlc.

jc-kynesim, 2020-08-07:
The --crop option is more versatile than documented - you can have:
<aspect_num>:<aspect_den>
<width>x<height>+<x>+<y>
<left>+<top>+<right>+<bottom>

KenT2, 2020-08-14:
VLC media player 3.0.11 Vetinari (revision 3.0.11-0-gdc0c5ced72)

Trying out the windowing and transparency options. Windowing works but I cannot get transparency to work 

`cvlc -f --vout mmal_vout --mmal-layer 20 --mmal-vout-window 100x100+0+0 --mmal-vout-transparent 5sec.mp4`

resizes the window but everything behind is black. Any thoughts

jc-kynesim, 2020-08-18:
Transparency is a pig(tm).  It does work but the problem is that we still have an X window and that X-window thinks it is fullscreen too and it is full of black and on top of everything else.  If you want to stop VLC generating a window in X the easiest way is to zap the environment var that gives it its display so "DISPLAY= vlc my_stream.avi" (N.B. there is a space between "DISPLAY=" and "vlc").  This is something of a kludge but it does work.  Beware that it also looses the trad shortcut keys too in favour of a somewhat odd command line interface, "help" should give the commands.

KenT2, 2020-08-28:
I have been struggling with this for a few days. I was hoping to use cvlc either with dbus controlling it or via the python-vlc API. Both work without the background suppression.

The python-vlc method of suppression will probably not work, even if I could find a way to do it with the API, because the Tkinter/PIL part of Pi Presents will get suppressed along with the black background as they are in the same process.

Using python's Popen to start a new process to run VLC would solve that problem and with DISPLAY= does suppress the VLC background. However including DISPLAY= seems to stop DBUS functioning, maybe because DBUS is an X thing.

Am I doing something wrong or have I hit a show stopper unless VLC is modified in some other way.

jc-kynesim, 2020-09-01:
I don't think you are doing anything wrong.  The only remaining things I can think of are -V mmal_vout --no-qt-video-autoresize, which makes mmal the only output device (so no need for -f) and stops qt resizing its control window, but that doesn't stop vlc from picking a stupid size / position anyway.  After that maybe custom skins (which I know nothing about)? but I'm out of useful ideas.

KenT2, 2020-09-04:
It's even worse than you say because --width, --height, --video-x and --video-y do not work so the window is stuck where vlc puts it. I also suspect the vlc window will steal focus.
DISPLAY= cvlc -Irc does get rid of the vlc window and gives some control commands that I hope I can pipe into cvlc using pexpect so that's the next experiment. Failing than vlc as a replacement for omxplayer in RPi is going to be fullscreen only with no in track user controls :-(
Thanks for your help

KenT2, 2020-10-25:
Success
DISPLAY= cvlc -I rc removes the window and provides all the commands I require, with some minor limitations.

I have published my experiments here [https://github.com/KenT2/pp-vlc](url)

closed 2020-10-25 by KenT2