RPi-Distro/vlc #13: Remove the X window related black background when playing fullscreen

KenT2, 2020-07-01:
Continued from 

https://github.com/popcornmix/omxplayer/issues/778

which concerns omxplayer

Controlling vlc with python-vlc is proving very successful. However I have found that, in addition to the black 'mmal' background on the monitor showing the video there is also a x/framebuffer related black background on the other display when a video is playing (suspect it covers the whole desktop). I wonder if it possible to remove this when --fullscreen is selected.

jc-kynesim, 2020-07-01:
If playing from X then the the VLC X-window goes fullscreen and the intention was to put the mmal overlay on top of it so you didn't see it - which obviously isn't working here. Given the separation of display & windows in VLC it is not easy to get rid of.  However if you are controlling VLC from python you maybe don't need the X controls at all and could run cvlc (or set DISPLAY= so vlc can't find its X-server)?

KenT2, 2020-07-01:
For Pi Presents I use Tkinter and X to display images and text so I cannot turn off X. I suspect other people have built multi-media apps using X and omxplayer.

I was wondering if you could de-couple the mmal display from --fullscreen as after your --win mods its not fullscreen any more .  So if --vout is mmal-xsplitter you always display the mmal layer. Because its not --fullscreen VLC will not produce a black background, it may produce a black video window but I might be able to cover this up with the mmal layer. 

Further to this if the -no-video option is used VLC does not produce any video window. Would it be possible for your mmal code to ignore this and produce the mmal layer.

jc-kynesim, 2020-07-01:
Does life work OK for you with "-V mmal_vout"?  It has the disadvantage for normal use that you don't have any controls but you might be OK if you are doing that elsewhere.

KenT2, 2020-07-02:
mmal_vout works fine with the the 3 videos I have tried. Will it have the --win etc. options that you are implementing for mmal-xsplitter.

I assume you mean by 'controls' the ability to do a mouse click on the video. I do not need that as mouse clicks are serviced by the Tkinter window below the video, this is how omxplayer works. I have tried this and it works fine.

What is the difference between vout and xsplitter, I did see a post from Pi Towers somehere that said vout should not be used and could be deleted from the options.

KenT2, 2020-07-02:
Did you mean me to use mmal-vout and then set fullscreen to False. I did this for another reason and found that the video displayed on the correct display and that I got a black x-window on HDMI-1. Are you trying to get rid of this x-window or shall I try and park it offscreen or something to nullify it.

jc-kynesim, 2020-07-02:
Xsplitter is a fake video output device that opens both x/opengles2 and mmal_vout and switches between them depending on fullscreen.  If you are missing one or the other then it should just be a straight pipe to the one that does exist.
I'm somewhat surprised by the extra black window on HDMI-1 if you are using mmal_vout directly - at least in my test version here if I do that with transparency on then I don't seen an X-window (I don't have 2 monitors on my Pi currently).  But given that I can achieve your desired result in my test code then I wouldn't worry about it until you find it doesn't work in the next version.

KenT2, 2020-10-25:
This Issue was continued here   [https://github.com/RPi-Distro/vlc/issues/12](url) with a successful outcome

DISPLAY= cvlc -I rc removes the window and provides all the commands I require, with some minor limitations.

I have published my experiments here https://github.com/KenT2/pp-vlc

closed 2020-10-25 by KenT2