RPi-Distro/vlc #14: Native deinterlacing

scharette88, 2020-07-04:
Hello all,

In OMXplayer, I've been using the --nativedeinterlace flag, so I could set the HDMI on the PI to being interlace and playback the video files natively in interlace. It was the only way to get smooth playback when using interlaced files, when I was using the --refresh flag, something was not right with the clock, and the playback was jittery. I was wondering if the --nativedeinterlace option could be ported to VLC ?

Thanks !

scharette88, 2020-07-04:
Hello again,

After looking at this discussion : 
https://github.com/raspberrypi/firmware/issues/292

It seem like it is, or was in VLC but I can't seem to find the option or the flag to enable this option,

Thanks !

jc-kynesim, 2020-07-06:
The option you are looking for is --mmal-native-interlaced.  It will (hopefully unsurprisingly) only work in fullscreen.  However beware that whilst the option exists it relies on code that was written a long time ago, I haven't ever tested, and I don't think I currently have any way of testing so I'm not going to be surprised if you say that it dies horribly..

closed 2020-07-10 by scharette88