RPi-Distro/vlc #16: using subtitles causes choppy motion

najoshi, 2020-08-07:
I am running vlc on a Pi 4 and I'm using a subtitles file (srt) for the subtitles. Every time the subtitle changes there is a split-second pause and then it continues. So it looks very jerky/choppy. Is there a fix for this?

jc-kynesim, 2020-08-07:
Sorry - no.  Subtitle rendering seems to happen in the main display thread and is unlikely to be fixable without a major rewrite of the core.  I guess that on x86 it is just that much faster that it fits in a frame time so no-one has needed to care about it. (I have done the test where I nop out all of my subtitle display code and you still get the glitches.)
I'll look at it again when we move to VLC4

popcornmix, 2020-08-12:
> the issue concerns every text subtitles, except maybe ssa, as it's inherent to the text renderer in VLC which output pictures with a non-unit aspect ratio. It means that subtitles needs to be resized, which is currently done in CPU using swscale. I have patches to remove the resize altogether, but it's not ready for mailing list yet.

https://trac.videolan.org/vlc/ticket/22317 and
https://forum.videolan.org/viewtopic.php?f=32&t=144325&sid=fca605b7c9abadbc5478fbb18b0bf78b&start=25

najoshi, 2020-08-14:
So do you think it might be possible to SSA style subtitles and not have this problem?

najoshi, 2020-08-24:
So just for anyone who is looking, I tried SSA style subtitles and it seemed to eliminate the choppiness.

mirkosamir, 2021-08-09:
Same Problem here, the only solution at this time i found is convert to ass type subtitles, or use the old yuvp renderer

in my raspberry pi 4 w/last Raspberry Pi OS.

~/.config/vlc/vlcrc 

[code]
# Use YUVP renderer (boolean)
freetype-yuvp=1 
[/code]

this render avoid laggy o choppy video playback, but i cant find the way to change the size of the font, vlc refuse to take the vlcrc config for the fonts size and type.