RPi-Distro/vlc #17: How to specify windows coordinates to cvlc

SvenVD, 2020-08-08:
omxplayer has the --win option that can be used like this:
--win "0 720 1920 1080"

rpisurv.net is heavily relying on this functionality, how can this be achieved with vlc or vlc python libraries?

Thanks

jc-kynesim, 2020-08-10:
--mmal-vout-window fullscreen|<width>x<height>+<x>+<y> will set the fullscreen output window position

popcornmix, 2020-08-10:
I've done a few tests with latest apt vlc and it seems to be working. Making a note here, so I can refer back to what was tested.
Running from console (without x):
This displays video in a smaller window with framebuffer console hidden:
```
cvlc --mmal-vout-window 1280x720+64+128 <file>
```
This displays video in a smaller window with framebuffer console visible:
```
cvlc --mmal-vout-transparent --mmal-vout-window 1280x720+64+128 <file>
```
This displays video in a smaller window with framebuffer console visible and video behind framebuffer console:
```
cvlc  --mmal-layer -200 --mmal-vout-transparent  --mmal-vout-window 1280x720+64+128 <file>
```
This flips the video left-to-right. It also flips the dest coordinate (I'm not sure that is expected behaviour)
```
cvlc  --mmal-vout-transform hflip --mmal-vout-transparent  --mmal-vout-window 1280x720+64+128 <file>
```
This rotates video clockwise by 90'. It also transforms the dest coordinate (I'm not sure that is expected behaviour)
```
cvlc  --mmal-vout-transform 90 --mmal-vout-transparent  --mmal-vout-window 1280x720+64+128 <file>
```

jc-kynesim, 2020-08-10:
Transform affecting win coords is expected as the "normal" use of it is to deal with rotated screens.  I think VLC has a way of forcing source transform (i.e. overriding landscape/portrait encoded in the source) but I've forgotten how to do it (N.B. it isn't the transform module!)

popcornmix, 2020-08-10:
omxplayer transforms source video and then renders to given dest coordinates, so the behaviour is different (although it's straightforward to adapt).
Also vlc is preserving aspect ratio. e.g. this doesn't render to a square - is there a way of forcing that? omxplayer does output square video here.
```
cvlc  --mmal-vout-transparent  --mmal-vout-window 1280x1280+64+128 <file>
```

jc-kynesim, 2020-08-10:
I suspect there may be a zoom option that does that - not sure

SvenVD, 2020-10-18:
> omxplayer transforms source video and then renders to given dest coordinates, so the behaviour is different (although it's straightforward to adapt).
> Also vlc is preserving aspect ratio. e.g. this [...]

I have the same issue. I do not want vlc to preserve the aspect ratio. How to force that.

Also I can't seem to make anything logical from:
<width>x<height>+<x>+<y>

It seems if x = 0 then video is displayed in the middle of the screen? If I enter a negative value for x then the video moves from the middle to the left of the screen and a postive values moves to the right, is that how it is supposed to work.

While <y> = 0 does seem to start on top of  the screen with positive values moving the video down.

jc-kynesim, 2020-10-19:
You can force the source display aspect ratio with "--aspect-ratio a:b" so "--aspect-ratio 1:1" will give you a square pici.  There is no explicit option for zoom to fill window but if you dup the window size into the aspect ratio you will get what you want.

SvenVD, 2020-12-30:
Thanks for the help, I just release [Rpisurv 3](https://github.com/SvenVD/rpisurv/tree/v3_latest) Using cvlc player at its core, replacing omxplayer completely

popcornmix, 2020-12-31:
Cool. Okay to close @SvenVD ?

SvenVD, 2021-01-01:
Yes

closed 2021-01-01 by SvenVD

KenT2, 2021-01-20:
A few additional hints discovered by trial and error:
Using --crop or --aspect-ratio allows the --mmal-vout-window to work correctly provided the aspect ratio of the processed source matches that of the window
Using --crop and --aspect-ratio in the same command does not work. Use only one.
If using libvlc or python-libvlc the --crop and --aspect-ratio command options are ignored. They can be set using MediaPlayer methods