RPi-Distro/vlc #19: getting the elapsed video time in the marquee

najoshi, 2020-08-12:
Is there a way to get the current elapsed time in seconds of the video to be part of the marquee? In previous iterations of my project I have been using the subtitles file to create a video counter in the subtitles, however, subtitles don't work very well in vlc at the moment and there isn't an easy solution, as you mentioned in [this issue](https://github.com/RPi-Distro/vlc/issues/16). So I was hoping to use the marquee option to do a similar thing, but I can't seem to figure out if it is possible to do so. I can, of course, use the time strings to get the system time values, but that is not what I want. According to [this documentation page](https://wiki.videolan.org/Documentation:Advanced_Use_of_VLC/) it seems to be possible to access video metadata for the marquee string, but I cannot figure out how to do it. Do you have any suggestions? Or maybe another way to do what I want?

najoshi, 2020-08-13:
So I did a little more digging and it seems like one used to be able to add the current video position to the marquee, but then it seems like they took out that feature in v2.0 and added in the ability to have the system time in the marquee instead. Which doesn't make any sense to me...

jc-kynesim, 2020-08-13:
I'm sorry - I know a lot about the specific areas I've modified but not a whole lot about the wider operation of VLC.