RPi-Distro/vlc #20: mmap failure, too many files open, and seg faults

najoshi, 2020-08-25:
So I am coding a simple digital picture frame (that also does video) using tkinter and python-vlc on a Pi 4. I am taking some of the code from PiPresents. [Here is the repo](https://github.com/najoshi/nik-presents). I am having various problems with vlc when I am testing using my full set of pictures and videos, which is around 75,000. The pictures work just fine, but the videos are having issues. The first issue is that the number of file descriptors just keep increasing as more videos get played and eventually I get this error:

`The per-process limit on the number of open file descriptors has been reached.
`

Then, I decided to test with with just videos to see what happens. After some (variable) number of videos play, I start getting this error even thought there is plenty of memory:

`mmap() failed: Cannot allocate memory
`

but the videos continue to play for a while, and then finally I get a seg fault. Can anyone shed any light on any of these? Why don't the file descriptors get released? Why does the memory allocation fail?

najoshi, 2020-08-25:
Also, getting these errors:

```
mmap() failed: Cannot allocate memory
Failed to create permanent mapping for memfd region with ID = 2923567703
Ignoring received block reference with non-registered memfd ID = 2923567703
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Protocol error
```

jc-kynesim, 2020-08-25:
Once we have run out of fds then most bets are off.  I'll see if I can track what is leaking but this is always hard.

mqnc, 2021-12-04:
Having the same issues on raspberry pi 3 (not the file descriptor thing but mmap and file descriptor stuff), also plenty of memory left. I just play audio files tho.