RPi-Distro/vlc #21: Package needs install to play CSS-protected DVDs

Botspot, 2020-09-04:
I've noticed that many DVDs don't play at all in RaspiOS's VLC. The files are visible, but their contents is unreadable due to CSS protection.
See [this forum thread](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=284450).
Anyway, the problem was solved when I installed the `libdvdcss` package. Now these DVDs play flawlessly in VLC, just like any other ordinary DVD.

Could that package be included by default for new RaspiOS images going forward?

XECDesign, 2020-09-04:
Probably not

https://wiki.videolan.org/Frequently_Asked_Questions/#Is_libdvdcss_legal.3F

Botspot, 2020-09-04:
Is it possible to do it legally?
Windows does it out of the box, so there must be a way.

XECDesign, 2020-09-04:
I'm guessing MS pays a license fee for DVD decryption that gets built into the cost of Windows.

Botspot, 2020-09-04:
Hmm, a license fee certainly wouldn't be practical for Raspbian. I wonder how ChromebookOS did it then.

Maybe VLC could detect the problem and notify users with a popup window saying "to play this dvd, you need to install libdvdcss. This is at your own risk, bla bla bla."

popcornmix, 2020-09-04:
I don't believe Windows supports it since Windows 10.
https://www.pcworld.com/article/2953724/how-to-play-dvds-in-windows-10-for-free.html

There is a $15 dollar option from the Windows store.

XECDesign, 2020-09-04:
The way Ubuntu do it offer to install it during installation. The first wizard could in theory have a page asking if people who like to install things like livdvdcss, widevine and so on.

@spl237 is that something worth considering at some point?

Botspot, 2020-09-04:
> I don't believe Windows supports it since Windows 10.
> https://www.pcworld.com/article/2953724/how-to-play-dvds-in-windows-10-for-free.html
> 
> There is a $15 dollar option from the Windows store.

Well that's interesting... All CSS-protected DVDs have played fine on all our win10 computers & laptops. Though they all came with built-in dvd drives so maybe that $15 was bundled with the cost of the PC.

spl237, 2020-09-04:
I don't think it is wise for us to suggest the installation of libraries which do anything to circumvent DRM. If users want to install them themselves unprompted, that's fine, but I don't think we as an organisation should be offering them - as far as I am aware, under the DMCA, it is still nominally against the law to bypass CSS. Even though everyone does it...

XECDesign, 2020-09-04:
Based on my vast lack of legal understanding and experience suggests that it's fine to use, but illegal to distribute tools which circumvent DRM. I saw something about Canonical hosting those repos outside of the US and just adding a tickbox to install them during install.

Also libdvdcss hasn't had any legal challenges against it yet, but I wouldn't want us to be the ones to set precedents.

spl237, 2020-09-04:
The information about adding DVD playback and libdvdcss to VLC is easily locatable online - just google "vlc won't play dvd" and it's in the first link. I don't think we need to do anything to make it easier; in fact I think it would be unwise if we did. There's a small chance of potential legal hassle, which I don't think is worthwhile. Besides - who on earth is still playing DVDs these days... ;)

Botspot, 2020-09-04:
@spl237 then perhaps VLC could detect the issue and be more specific about the problem, like "Failure to decode css protected disk" in the log.
Currently, vlc would just say "could not open media device", which is really unhelpful.

spl237, 2020-09-04:
That would be a suggestion to make to the authors of VLC rather than to us!