RPi-Distro/vlc #23: Sync vedios

pramodmadhav, 2020-09-23:
Just like omxplayer sync ,2,3
Can one do same thing with vlc

jc-kynesim, 2020-09-23:
What are you expecting that to do?

pramodmadhav, 2020-09-24:
Sync different video files playing on separate video players

jc-kynesim, 2020-09-24:
I thought that might be it. I'm afraid that I don't see that making it into VLC from us - it would require substantial changes to the core vlc code.

pramodmadhav, 2020-10-08:
Ok then can it (VLC) sync to external time code like MTC or SMPTE over network

SebDominguez, 2020-11-07:
[omxplayer-sync](https://github.com/turingmachine/omxplayer-sync) use dbus commands to sync slave player to the current time code from a master player using simple seek commands. I didn't dig deep into the vlc api but it seems that vlc support control over dubs so technically it should be possible to create an equivalent to Omxplayer-sync using vlc...

pramodmadhav, 2022-02-23:
can one set position in vlc (hr:min:sec:frames)

popcornmix, 2022-02-23:
What about https://wiki.videolan.org/Documentation:Modules/netsync/

jakobereksen, 2024-03-04:
Hey, I just wanted to open a new issue on this, but then stumbled over this. What is the current state of the issue? AFAIK netsync feature on vlc is still broken. 
To me personally this is the most important feature that is not yet available on vlc (video art, context). I can still use omx for that but that sadly only for 1080p. I do understand of course that the importance this has to me does not equate in high priority for everyone. Is there a feature bounty system? I could imagine a handful of people me included would be willing to donate for this specifically. 
Cheers :)