RPi-Distro/vlc #24: Dbus interaction of running cvlc stream

SvenVD, 2020-10-03:
Omxplayer can connect to dbus after which you can programmatically control the coordinates of a stream/video without stopping and restarting the stream/video. How to do this with cvlc?

Thanks

popcornmix, 2020-10-05:
There's documentation in the vlc wiki:
https://wiki.videolan.org/DBus-usage/
https://wiki.videolan.org/DBus-spec/

SvenVD, 2020-10-19:
In omxplayer you have the VideoPos dbus method. It doesnt seem to be present for cvlc? How to move the window with dbus then?

`gdbus introspect -r --session --dest org.mpris.MediaPlayer2.vlc --object-path /
node / {
  node /org {
    node /org/mpris {
      node /org/mpris/MediaPlayer2 {
        interface org.freedesktop.DBus.Introspectable {
          methods:
            Introspect(out s data);
          signals:
          properties:
        };
        interface org.freedesktop.DBus.Properties {
          methods:
            Get(in  s arg_0,
                in  s arg_1,
                out v arg_2);
            Set(in  s arg_0,
                in  s arg_1,
                in  v arg_2);
            GetAll(in  s arg_0,
                   out a{sv} arg_1);
          signals:
            PropertiesChanged(s arg_0,
                              a{sv} arg_1,
                              as arg_2);
          properties:
        };
        interface org.mpris.MediaPlayer2 {
          methods:
            Quit();
            Raise();
          signals:
          properties:
            readonly s Identity = 'VLC media player';
            readonly s DesktopEntry = 'vlc';
            readonly as SupportedMimeTypes = ['audio/mpeg', 'audio/x-mpeg', 'video/mpeg', 'video/x-mpeg', 'video/mpeg-system', 'video/x-mpeg-system', 'video/mp4', 'audio/mp4', 'video/x-msvideo', 'video/quicktime', 'application/ogg', 'application/x-ogg', 'video/x-ms-asf', 'video/x-ms-asf-plugin', 'application/x-mplayer2', 'video/x-ms-wmv', 'video/x-google-vlc-plugin', 'audio/wav', 'audio/x-wav', 'audio/3gpp', 'video/3gpp', 'audio/3gpp2', 'video/3gpp2', 'video/divx', 'video/flv', 'video/x-flv', 'video/x-matroska', 'audio/x-matroska', 'application/xspf+xml'];
            readonly as SupportedUriSchemes = ['file', 'http', 'https', 'rtsp', 'realrtsp', 'pnm', 'ftp', 'mtp', 'smb', 'mms', 'mmsu', 'mmst', 'mmsh', 'unsv', 'itpc', 'icyx', 'rtmp', 'rtp', 'dccp', 'dvd', 'vcd'];
            readonly b HasTrackList = false;
            readonly b CanQuit = true;
            readonly b CanSetFullscreen = true;
            readwrite b Fullscreen = false;
            readonly b CanRaise = false;
        };
        interface org.mpris.MediaPlayer2.Player {
          methods:
            Previous();
            Next();
            Stop();
            Play();
            Pause();
            PlayPause();
            Seek(in  x arg_0);
            OpenUri(in  s arg_0);
            SetPosition(in  o arg_0,
                        in  x arg_1);
          signals:
          properties:
            readonly a{sv} Metadata = {'mpris:trackid': <objectpath '/org/videolan/vlc/playlist/3'>, 'xesam:url': <'rtsp://admin2:adminCamRoot12@192.168.2.11:554/cam/realmonitor?channel=1&subtype=1'>, 'vlc:publisher': <5>};
            readonly s PlaybackStatus = 'Playing';
            readwrite s LoopStatus = 'None';
            readwrite d Volume = 1.0;
            readwrite d Shuffle = false;
            readonly i Position = 271472874;
            readwrite d Rate = 1.0;
            readwrite d MinimumRate = 0.032000000000000001;
            readwrite d MaximumRate = 32.0;
            readonly b CanControl = true;
            readonly b CanPlay = true;
            readonly b CanPause = true;
            readonly b CanSeek = false;
        };
        interface org.mpris.MediaPlayer2.TrackList {
          methods:
            GetTracksMetadata(in  ao arg_0,
                              out aa{sv} arg_1);
            AddTrack(in  s arg_0,
                     in  o arg_1,
                     in  b arg_2);
            RemoveTrack(in  o arg_0);
            GoTo(in  o arg_0);
          signals:
            TrackListReplaced(ao arg_0,
                              o arg_1);
            TrackAdded(a{sv} arg_0,
                       o arg_1);
            TrackRemoved(o arg_0);
            TrackMetadataChanged(o arg_0,
                                 a{sv} arg_1);
          properties:
            readonly ao Tracks = ['/org/videolan/vlc/playlist/3'];
            readonly b CanEditTracks = true;
        };
      };
    };
  };
};`

Thanks