RPi-Distro/vlc #25: libvlc_video_set_output_callbacks() not found

oomek, 2020-10-07:
I've been trying to get the following example working on PI4 to get zero copy rendering on an OpenGL texture.
https://github.com/videolan/vlc/blob/master/doc/libvlc/sdl_opengl_player.cpp
But it seems the `libvlc_video_set_output_callbacks()` function is missing in libVLC 3.x 
Is there any way to go around it? Will libVLC 4.0 work on PI4 ?

popcornmix, 2020-10-07:
VLC 4 is being worked on but it is some way off.

oomek, 2020-10-07:
Super fast reply as usual, thanks @popcornmix 
I've been trying to get my way around OMX deprecation, but I'm getting nowhere. What's the rough timeframe until it's ready to consume by the public? Are there any other unexplored options that would allow for zero copy rendering that I could transform with OpenGL?

jc-kynesim, 2020-10-07:
I've had a VERY rough VLC4 running to prove some V4L2 work, but one that is fit for any sort of public consumption is (as popcornmix said) a way off - not less than a month, probably noticably more than that.  The structural differences between VLC3 & VLC4 are non-trivial (and their Qt doesn't like me so the UI is broken too).