RPi-Distro/vlc #27: libvlc mmal errors on PI4

oomek, 2020-10-16:
HI, I'm having some troubles with libvlc.
When I play h264 yuv420 videos using the current libvlc api I get the following errors on RPI4:

```
[ae01ec78] mmal_codec decoder: VCSM init succeeded: Legacy
mmal: mmal_vc_component_create: failed to create component 'vc.ril.hvs' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.hvs' (1)
```

Additionally x264 yuv444 videos does not show any video and I get this instead
```
[a923ca40] mmal_codec decoder: VCSM init succeeded: Legacy
[a923ca40] main decoder error: buffer deadlock prevented
```

Linux retropie4 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux
VLC media player 3.0.11 Vetinari (revision 3.0.11-0-gdc0c5ced72)
dtoverlay=vc4-kms-v3d-pi4

oomek, 2020-10-16:
Oh and btw, is there any way to force alsa audio instead of pulseaudio?
I set alsa with `libvlc_audio_output_device_set(m_media_player.get(), "ALSA", "hw:0,0");` after `libvlc_media_player_new()` call
but I'm still getting the following errors:
```
[b03148c8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
```

jc-kynesim, 2020-10-19:
OK
1) The VCSM message is only an "Info" message so I would have expected to be killable by "--verbose 0" but VLCs log level selection doesn't seem to work the way I would expect.  I'll reduce it to a debug grade message.

2) The mmal errors are caused by the code seeing if it can set up HVS for fullscreen output (which it can't under KMS) and come from the userland libraries.  It is arguable that VLC should know it is doomed before trying - I'll look into that.

3) If you point me at a sample yuv444 video I'll see what is going on.  The one thing I'm sure about is that it isn't going to be successfully decoded by our h/w block.

4) Unsure about alsa/pulse as that isn't an area that I've poked at all.  I'll have a look to see what can be done there but I don't want to change module priorities if I don't have to in case someone else is somehow relying on the default behaviour.