RPi-Distro/vlc #28: [Question] What's the equivalent of omxplayer -n N (to select, e.g., HDMI output, -n 2/7)

bjohas, 2020-10-17:
Thanks for the work on omxplayer and vlc.

What is the equivalent of the '-n' option in omxplayer? 

tldr: I've got a pi attached to a screen (but no keyboard) and want to play videos to that screen by sshing into the Pi, i.e., without launching the desktop environment, and without connecting a keyboard or mouse to the pi.

I tried this
```
vlc --vout fb --fbdev=/dev/fb0 file.mp4
```
but I get
```
fb vout display error: cannot get terminal mode (Inappropriate ioctl for device)
```
(I'm on Pi4, with Ubuntu - I realise the vlc distribution may be for Raspbian.)

Many thanks!
Björn

jc-kynesim, 2020-10-18:
If you were running Raspbian then it should all just work - if there is no X then our VLC will default to mmal-based fullscreen playback so a simple "vlc myvideo.mp4" should work.  On a base Ubuntu distribution I have much less idea.

bjohas, 2020-10-18:
Thanks - on Raspbian, with vlc, how would you chose which HDMI port to use for playback?

jc-kynesim, 2020-10-18:
By default (with no X) it'll use hdmi-1, otherwise "--mmal-display hdmi-1" or "--mmal-display hdmi-2". There are a number of related options which you can find with "vlc -H", this creates a LOT of output so pipe that to a file and search the file for options that start --mmal-.

jc-kynesim, 2020-10-18:
Just to check - you are running the Raspbian VLC?  Most of the stuff I've done there is not upstreamed so I wouldn't expect it to turn up in the standard distribution.

bjohas, 2020-10-18:
Ah - no. That's a very good point! No, I just installed from repo!

I just tried to ./configre - but there's a lot of missing packages. How do I install all the dependencies?

jc-kynesim, 2020-10-18:
In general the answer to that is "sudo apt-get build-dep vlc" BUT if you don't have the Pi userland libs (which you should find in /opt/vc if you have them) you aren't going to get my code enabled and I wouldn't expect then to exist (or work) in a standard Ubuntu dist.

bjohas, 2020-10-18:
Yeah, I don't have those.... OK, I'll get another card and install Raspbian :)

bjohas, 2020-10-18:
Thank yo so much for the tips!