RPi-Distro/vlc #30: mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module

ShaoMingya, 2020-11-05:
PI4B，use python-vlc，

```
#!/usr/bin/env python3
import time, vlc

instance = vlc.Instance("--verbose=0","--no-xlib")
media_list = instance.media_list_new(['/home/pi/Videos/test.mp4', '/home/pi/Videos/Gallium_drops.mp4'])
list_player = instance.media_list_player_new()
list_player.set_media_list(media_list)

list_player.set_playback_mode(vlc.PlaybackMode.loop)
list_player.play()

while True:
    time.sleep(300)
```

My requirement is to be able to switch between each video seamlessly when using a playlist.
But every time the video is switched, the playback window will be black for about one second.

```
pi@raspberrypi:~ $ /home/pi/python_codes/py_vlc/vlc_test.py
[91825a98] mmal_codec decoder: VCSM init succeeded: Legacy
[a4a01ca8] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
[91808358] mmal_codec decoder: VCSM init succeeded: Legacy
[91808358] main decoder error: buffer deadlock prevented
[a4a05ae8] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
[91808ae0] mmal_codec decoder: VCSM init succeeded: Legacy
[91808ae0] main decoder error: buffer deadlock prevented
[a4a43840] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module

```

masseelch, 2021-05-31:
I have the exact same problem but with [libvlc-go](https://github.com/adrg/libvlc-go), which are `libVLC`s bindings for go.

KenT2, 2021-05-31:
You may need to add  --vout mmal_vout  to the instance options

masseelch, 2021-06-01:
> You may need to add --vout mmal_vout to the instance options

Dang! You are a hero. Now some of the videos do work without a gap. For some of them I still get the `main decoder error: buffer deadlock prevented` error though.

KenT2, 2021-06-01:
I get that error as well. I just ignore it.

masseelch, 2021-06-01:
That error does occur when the black gap happens, that was descibed by @ShaoMingya.

AVollane, 2021-10-30:
Hello. How about it now? Did you find something? I have the same problem with LibVLCSharp.

jc-kynesim, 2021-11-01:
"Buffer deadlock prevented" isn't a real error in the sense that something has gone wrong.  What it means is that the mmal h/w decode can buffer more frames than vlc is expecting before outputting anything.  It would be very nice to reduce the size of that Q but the firmware makes it very difficult (impossible)

jc-kynesim, 2021-11-01:
VLC doesn't, in general, do anything about starting up the next stream until the previous one has finished and tends to tear down everything (including the video out) in between.  On a PC the stop/restart can be fast enough that you don't see it, on a pi it is more visible.  The large input buffering doesn't help as VLC spends more of its time stuffing the input than dealing with anything else.

zocker-160, 2022-09-02:
> You may need to add --vout mmal_vout to the instance options

I am getting `vlc: unknown option or missing mandatory argument "--vout mmal_vout"`

What am I doing wrong?

VLC  `3.0.17.4-0+deb11u1+rpi1+rpt3`
PI3B on Debian 11