RPi-Distro/vlc #31: VLC hang and need a reboot to play video

SebDominguez, 2020-11-16:
I was brought here because I thought that [my problems with OMXplayer](https://github.com/popcornmix/omxplayer/issues/789) were due to the fact that this player was depreciated for the new raspberry pi architecture. But now vlc is crashing similarly.

I'm running a python script using [python-vlc](https://pypi.org/project/python-vlc/) and sometimes vlc just crash without anyway to kill it (no output on `ps -aux | grep vlc`).

I have to reboot the Pi4 and I can't reproduce this crash consistently, it just happened randomly. I don't know where would be located any useful logs to troubleshoot this.

here is a simplified version of my script

```python3
#!/usr/bin/env python3

import logging
import time, signal
import os
import vlc
from enum import Enum, auto
import paho.mqtt.client as mqtt

DEBUG_LEVEL = logging.DEBUG
VLCARGS='--quiet --alsa-audio-device sysdefault:CARD=Headphones -I dummy --no-video-title'
LOGPATH='/home/pi/player/'
VIDEO_PATH = '/home/pi/video/'

class Status(Enum):
	STOPPED = auto()
	PAUSED = auto()
	PLAYING = auto()
	ENDED = auto()

class MyPlayer:
	status = None
	filePath = None

	def __init__(self, args):
		self.client = mqtt.Client('player')
		self.client.on_connect = self.on_connect
		self.client.on_publish = self.on_publish
		self.client.message_callback_add("player/PLAY", self.on_mqtt_play)
		self.client.message_callback_add("player/STOP", self.on_mqtt_stop)
		self.client.connect("master.local", 1883, 60)
		self.client.loop_start()
		self.instance = vlc.Instance(args)
		self.player = self.instance.media_player_new()
		self.events = self.player.event_manager()
		self.events.event_attach(vlc.EventType.MediaPlayerPlaying, self.callback_playing)
		self.events.event_attach(vlc.EventType.MediaPlayerPaused, self.callback_paused)
		self.events.event_attach(vlc.EventType.MediaListPlayerStopped, self.callback_stopped)
		self.events.event_attach(vlc.EventType.MediaPlayerEndReached, self.callback_endReached)

	def callback_playing(self, event):
		self.status = Status.ENDED
		self.client.publish("player/STAT/PLAYING", payload=self.filePath, qos=0, retain=False)
		logger.debug("Playing")

	def callback_paused(self, event):
		self.status = Status.PAUSED
		self.client.publish("player/STAT/PAUSED", payload=None, qos=0, retain=False)
		logger.debug("Pause")

	def callback_stopped(self, event):
		self.status = Status.STOPPED
		self.client.publish("player/STAT/STOPPED", payload=None, qos=0, retain=False)
		logger.debug("Stop")

	def callback_endReached(self, event):
		self.status = Status.ENDED
		self.client.publish("player/STAT/END", payload=None, qos=0, retain=False)
		logger.debug("End")

	def play(self):
		self.player.play()

	def playFile(self, path):
		if self.status == Status.PLAYING:
			logger.debug("A file is currently playing: stopping first...")
			self.stop()
		try:
			location = self.instance.media_new(path)
			self.filePath = path
			self.player.set_media(location)
			self.play()
		except Exception as e:
			logger.critical("Error while playing the file: " + path)
			logger.critical(e)

	def pause(self):
		self.player.pause()

	def stop(self):
		self.player.stop()

	def on_mqtt_stop(self, client, userdata, message):
		print("Recived on: " + message.topic + " payload: " + message.payload.decode())
		self.stop()

	def on_mqtt_play(self, client, userdata, message):
		print("Recived on: " + message.topic + " payload: " + message.payload.decode())
		self.playFile(message.payload.decode())

	def on_connect(self, client, userdata, flags, rc):
		logger.debug("MQTT Connected with result code "+str(rc))
		self.client.subscribe("player/#")

	def on_publish(self, client, userdata, mid):
		logger.debug("mid = " + str(mid))

class MyFilter(object):
	def __init__(self, level):
		self.__level = level

	def filter(self, logRecord):
		return logRecord.levelno >= self.__level

class GracefulExit:
	kill_now = False
	def __init__(self):
		signal.signal(signal.SIGINT, self.exit_gracefully)
		signal.signal(signal.SIGTERM, self.exit_gracefully)
	def exit_gracefully(self,signum, frame):
		self.kill_now = True

def main():
	app_killer = GracefulExit()
	while not app_killer.kill_now:
		try:
			time.sleep(0.1)
		except BaseException:
			logger.critical("Encountered an axeption.")
			break
	logger.info("End of the program.")

if __name__ == '__main__':

	#create a logger
	logger = logging.getLogger('player')
	logger.setLevel(logging.DEBUG)
	handler = logging.FileHandler(LOGPATH + 'player.log')
	formatter = logging.Formatter('%(asctime)s - %(name)s - %(levelname)s - %(message)s')
	handler.setFormatter(formatter)
	#set filter to log only DEBUG_LEVEL
	handler.addFilter(MyFilter(DEBUG_LEVEL))
	logger.addHandler(handler)

	#write a debug line to log file
	logger.debug('========== Starting player ==========')

	logger.debug("Playing intro using VLC")
	p = MyPlayer(VLCARGS)
	main()
```

SebDominguez, 2020-11-18:
This is the output that I normally have when playing a video vile using cvlc:

```
$ cvlc --play-and-exit -v intro.mp4
VLC media player 3.0.11 Vetinari (revision 3.0.11-0-gdc0c5ced72)
[014696c0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[01485c10] main interface error: no suitable interface module
[013eece8] main libvlc error: interface "globalhotkeys,none" initialization failed
[01485c10] dummy interface: using the dummy interface module...
[b25012b8] mp4 stream warning: unknown box type elng (incompletely loaded)
[b25012b8] mp4 stream warning: unknown box type clfn (incompletely loaded)
[b25012b8] mp4 stream warning: unknown box type dscp (incompletely loaded)
[b25012b8] mp4 stream warning: unknown box type kywd (incompletely loaded)
[b25012b8] mp4 stream warning: unknown box type thmb (incompletely loaded)
[b25012b8] mp4 stream warning: unknown box type auth (incompletely loaded)
[b25012b8] mp4 stream warning: unknown box type titl (incompletely loaded)
[b25084e8] mp4 demux warning: elst box found
[b25084e8] mp4 demux warning: STTS table of 1 entries
[b25084e8] mp4 demux warning: elst box found
[b25084e8] mp4 demux warning: STTS table of 1 entries
[b250c748] mmal_codec decoder: VCSM init succeeded: CMA
[b250d300] faad decoder warning: decoded zero sample
[014696c0] alsa audio output warning: device cannot be paused
[b250c748] main decoder error: buffer deadlock prevented
Fontconfig warning: ignoring UTF-8: not a valid region tag
[b1f502d0] gles2 generic error: parent window not available
[b1f4fbc0] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
[b1f502d0] xcb generic error: window not available
[b1f4fbc0] mmal_xsplitter vout display error: Failed to open Xsplitter:xcb_x11 module
[b1f02cd0] main video output warning: picture is too late to be displayed (missing 224 ms)
[b1f02cd0] main video output warning: picture is too late to be displayed (missing 191 ms)
[b1f02cd0] main video output warning: picture is too late to be displayed (missing 157 ms)
[b1f02cd0] main video output warning: picture is too late to be displayed (missing 124 ms)
[b1f02cd0] main video output warning: picture is too late to be displayed (missing 91 ms)
[b1f02cd0] main video output warning: picture is too late to be displayed (missing 57 ms)
[b1f02cd0] main video output warning: picture is too late to be displayed (missing 24 ms)
[0144ace0] main playlist: end of playlist, exiting
```

this is the output I have once vlc randomly crash:

```
 cvlc --play-and-exit -v intro.mp4
VLC media player 3.0.11 Vetinari (revision 3.0.11-0-gdc0c5ced72)
[002d86c0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[002f49a8] main interface error: no suitable interface module
[0025dce8] main libvlc error: interface "globalhotkeys,none" initialization failed
[002f49a8] dummy interface: using the dummy interface module...
[b25012b8] mp4 stream warning: unknown box type elng (incompletely loaded)
[b25012b8] mp4 stream warning: unknown box type clfn (incompletely loaded)
[b25012b8] mp4 stream warning: unknown box type dscp (incompletely loaded)
[b25012b8] mp4 stream warning: unknown box type kywd (incompletely loaded)
[b25012b8] mp4 stream warning: unknown box type thmb (incompletely loaded)
[b25012b8] mp4 stream warning: unknown box type auth (incompletely loaded)
[b25012b8] mp4 stream warning: unknown box type titl (incompletely loaded)
[b250b7e0] mp4 demux warning: elst box found
[b250b7e0] mp4 demux warning: STTS table of 1 entries
[b250b7e0] mp4 demux warning: elst box found
[b250b7e0] mp4 demux warning: STTS table of 1 entries
[b250e648] mmal_codec decoder: VCSM init succeeded: CMA
```

SebDominguez, 2020-11-18:
I've managed to considerably make my script more stable by adding `time.sleep(1.5)` after each api calls to give some times to vlc to do his things... Maybe those craches are due to the fact that vlc doesn't have time to hide the dust under the carpet between 2 calls...

dasl-, 2022-02-25:
Hi @SebDominguez , you may want to try `over_voltage=2` in `/boot/config.txt`. See: https://github.com/Hexxeh/rpi-firmware/issues/249#issuecomment-1050767968

Curious if this helps.

SebDominguez, 2022-02-25:
Thank you I'll try this ASAP.
I was starting to loose any hope to get this fixed