RPi-Distro/vlc #37: Can CVLC display on Composite Video or Official Touchscreen when using --mmal_vout

KenT2, 2021-01-23:
Can CVLC display on Composite Video or Official Touchscreen

If so what are the names for --mmal-display

SvenVD, 2021-02-05:
+1
 [we](https://www.tapatalk.com/groups/rpisurv/viewtopic.php?p=554#p554) have the same question.

KenT2, 2022-07-28:
I have determined by experiment that cvlc can display on the official touchscreen. However --mmal-display has no effect.

- If only the touchscreen is connected the video displays on it even if --mmal-display= HDMI-1

- If HDMI0 and the touchscreen is attached then the video displays on HDMI0 even if --mmal-display=DSI-1

So it appears the display used is determined by what is connected with priority given to HDMI

The listing provided by  cvlc -H indicates that the only valid values are HDMI-1 and HDMI-2, could DSI-1, and also A/V if it works, be added to the cvlc command?