RPi-Distro/vlc #43: Unable to play HVEC H265 file in RPI noobs image

gurtajs4, 2021-05-02:
Hi , 

I just bought new RPI4 and  installed Noobs image.
The image has VLC and ffplay build in but i am not able to play any hevc files, looks like HW accel not working

What am i missing??

Below are my config files
[config.txt](https://github.com/RPi-Distro/vlc/files/6412119/config.txt)

my cmdline.txt 
**console=serial0,115200 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles**

popcornmix, 2021-05-03:
Are you up to date? (accept the option to update software on first boot wizard, or run `sudo apt update; sudo apt full-upgrade`

gurtajs4, 2021-05-04:
Hi @popcornmix ,

I have update the RPI , but still no success.
2 sample video are available here, Could you please help
 
https://www.dropbox.com/sh/vebftvy05lo4keb/AAAu0JhHRr06gX7QVbNoyCjHa/From%20Client/Video%20Files/4K?dl=0&subfolder_nav_tracking=1 

VLC media player 3.0.12 Vetinari (revision 3.0.12-1-0-gd147bb5e7e)
xcb_connection_has_error() returned true
[01120428] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[0111e2a8] main interface error: no suitable interface module
[010a4b58] main libvlc error: interface "globalhotkeys,none" initialization failed
[0111e2a8] dummy interface: using the dummy interface module...
[b233a9c8] mmal_avcodec decoder: Looking for HEVC decoder 'hevc'
[b2390fd0] gles2 generic error: parent window not available
[b2393230] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
[b2390fd0] xcb generic error: window not available
[b2393230] mmal_xsplitter vout display error: Failed to open Xsplitter:xcb_x11 module
[hevc @ 0xb2343280] Out of pu + coeff intermediate memory: pus=768
[hevc @ 0xb2343280] hardware accelerator failed to decode picture
[hevc @ 0xb233ed70] Unable to resolve output frame
[b233a9c8] mmal_avcodec decoder error: avcodec_send_packet critical error
[hevc @ 0xb237bd80] Out of pu + coeff intermediate memory: pus=768
[hevc @ 0xb237bd80] hardware accelerator failed to decode picture
[hevc @ 0xb2343280] Out of pu + coeff intermediate memory: pus=768
[hevc @ 0xb2343280] hardware accelerator failed to decode picture
[hevc @ 0xb233ed70] Unable to resolve output frame
[b233a9c8] mmal_avcodec decoder error: avcodec_send_packet critical error
[hevc @ 0xb237bd80] Out of pu + coeff intermediate memory: pus=768
[hevc @ 0xb237bd80] hardware accelerator failed to decode picture
[hevc @ 0xb233ed70] Unable to resolve output frame
[b233a9c8] mmal_avcodec decoder error: avcodec_send_packet critical error
[hevc @ 0xb234f750] Out of pu + coeff intermediate memory: pus=768
[hevc @ 0xb234f750] hardware accelerator failed to decode picture
[hevc @ 0xb233ed70] Unable to resolve output frame
[b233a9c8] mmal_avcodec decoder error: avcodec_send_packet critical error
[hevc @ 0xb2343280] Out of pu + coeff intermediate memory: pus=768
[hevc @ 0xb2343280] hardware accelerator failed to decode picture
[hevc @ 0xb233ed70] Unable to resolve output frame
[b233a9c8] mmal_avcodec decoder error: avcodec_send_packet critical error
[hevc @ 0xb237bd80] Out of pu + coeff intermediate memory: pus=768
[hevc @ 0xb237bd80] hardware accelerator failed to decode picture
[hevc @ 0xb233ed70] Unable to resolve output frame
[b233a9c8] mmal_avcodec decoder error: avcodec_send_packet critical error
[hevc @ 0xb234f750] Out of pu + coeff intermediate memory: pus=768
[hevc @ 0xb234f750] hardware accelerator failed to decode picture
[hevc @ 0xb233ed70] Unable to resolve output frame
[b233a9c8] mmal_avcodec decoder error: avcodec_send_packet critical error
[hevc @ 0xb233ed70] Unable to resolve output frame
[b233a9c8] mmal_avcodec decoder error: avcodec_send_packet critical error
[hevc @ 0xb2343280] Out of pu + coeff intermediate memory: pus=768
[hevc @ 0xb2343280] hardware accelerator failed to decode picture
[hevc @ 0xb234f750] Out of pu + coeff intermediate memory: pus=768
[hevc @ 0xb234f750] hardware accelerator failed to decode picture
[hevc @ 0xb237bd80] Could not find ref with POC 158
[hevc @ 0xb233ed70] Unable to resolve output frame
[b233a9c8] mmal_avcodec decoder error: avcodec_send_packet critical error
[hevc @ 0xb2343280] Out of pu + coeff intermediate memory: pus=768
[hevc @ 0xb2343280] hardware accelerator failed to decode picture
[hevc @ 0xb237bd80] Out of pu + coeff intermediate memory: pus=768
[hevc @ 0xb237bd80] hardware accelerator failed to decode picture
[hevc @ 0xb233ed70] Unable to resolve output frame
[b233a9c8] mmal_avcodec decoder error: avcodec_send_packet critical error
[hevc @ 0xb234f750] Out of pu + coeff intermediate memory: pus=768
[hevc @ 0xb234f750] hardware accelerator failed to decode picture
[hevc @ 0xb233ed70] Unable to resolve output frame
[b233a9c8] mmal_avcodec decoder error: avcodec_send_packet critical error
[hevc @ 0xb2343280] Out of pu + coeff intermediate memory: pus=768
[hevc @ 0xb2343280] hardware accelerator failed to decode picture
[hevc @ 0xb233ed70] Unable to resolve output frame
[... 44 lines cut ...]
[hevc @ 0xb234f750] hardware accelerator failed to decode picture
[hevc @ 0xb233ed70] Unable to resolve output frame
[b233a9c8] mmal_avcodec decoder error: avcodec_send_packet critical error
[hevc @ 0xb2343280] Out of pu + coeff intermediate memory: pus=768
[hevc @ 0xb2343280] hardware accelerator failed to decode picture
[hevc @ 0xb233ed70] Unable to resolve output frame
[b233a9c8] mmal_avcodec decoder error: avcodec_send_packet critical error
[hevc @ 0xb237bd80] Out of pu + coeff intermediate memory: pus=768
[hevc @ 0xb237bd80] hardware accelerator failed to decode picture
[hevc @ 0xb233ed70] Unable to resolve output frame
[b233a9c8] mmal_avcodec decoder error: avcodec_send_packet critical error
[hevc @ 0xb234f750] Out of pu + coeff intermediate memory: pus=768
[hevc @ 0xb234f750] hardware accelerator failed to decode picture
[hevc @ 0xb233ed70] Unable to resolve output frame
[b233a9c8] mmal_avcodec decoder error: avcodec_send_packet critical error
[hevc @ 0xb233ed70] Unable to resolve output frame
[b233a9c8] mmal_avcodec decoder error: avcodec_send_packet critical error
^C[b233a9c8] mmal_avcodec decoder error: CMA buf pool alloc buf failed
rpi_get_display_buffer: Failed to get buffer from pool
[hevc @ 0xb2343280] Failed to allocate output frame
[hevc @ 0xb2343280] hardware accelerator failed to decode picture
[b233a9c8] mmal_avcodec decoder error: CMA buf pool alloc buf failed
rpi_get_display_buffer: Failed to get buffer from pool
[hevc @ 0xb234f750] Failed to allocate output frame
[hevc @ 0xb234f750] hardware accelerator failed to decode picture

popcornmix, 2021-05-05:
I think you need to increase the CMA allocated.
In `/boot/config.txt` remove any `gpu_mem=` line.
Change the line with `dtoverlay=vc4-fkms-v3d` to `dtoverlay=vc4-kms-v3d,cma-512`

gurtajs4, 2021-05-05:
I did that too buddy, Still same issue
```
removed any gpu_mem= line.
Change the line with dtoverlay=vc4-fkms-v3d to dtoverlay=vc4-kms-v3d,cma-512
```

Could you confirm if you are able to load the latest image 
https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-03-25/2021-03-04-raspios-buster-armhf-lite.zip

install VLC and could play those sample Video smoothly?

jc-kynesim, 2021-05-05:
No - that debug shows we are running out of pu/coeff intermediate storage - I'm going to guess that those streams are complex by our standards. A new version of ffmpeg may be required - there is a patch for V4L2 in the pipeline, I'll check those streams against it.

gurtajs4, 2021-05-05:
OK I have one RPI 4 image from some vendor that do seems to play these video perfectly fine.
I think the VLC was build manually here's the version dump. Let me know if you need any other info from that sdcard to debug faster.

pi:~ $ vlc --version
VLC media player 3.0.11 Vetinari (revision 1.0.6-1613-gc2ebfd0cb)
VLC version 3.0.11 Vetinari (1.0.6-1613-gc2ebfd0cb)
Compiled by pi on XXX (Nov 20 2020 01:00:07)
Compiler: gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)
This program comes with NO WARRANTY, to the extent permitted by law.
You may redistribute it under the terms of the GNU General Public Licence;
see the file named COPYING for details.
Written by the VideoLAN team; see the AUTHORS file.

jc-kynesim, 2021-05-05:
A little investigation later:
Indeed '1B - 4k Video Sync Right - 15mins.mp4' doesn't play properly due to an intermediate buffer that was sized for what was the worst case seen when the code was written. The next ffmpeg (which will probably be out in a week or two) should resize it it required.
On the other hand I have no idea why 'Lupa Sync Test - H265 4k.mp4' doesn't play for you - it played fine on a clean RpiOS for me.

gurtajs4, 2021-05-05:
thanks @jc-kynesim  for verification.
Do you have patches available for fixing the issue? I could give it try today.

If so, do I need to build VLC only or both VLC and ffmpeg?

Are these correct repository for picking the latest code?

from https://github.com/RPi-Distro/vlc.git
https://github.com/RPi-Distro/ffmpeg.git

**Also do you have instruction to configure and build them? I have really struggled to build these with hvec support , correct configure command would help a lot**

jc-kynesim, 2021-05-05:
Is there a good reason why you don't want to use "apt install vlc"? Its what I did.

jc-kynesim, 2021-05-05:
And just to be clear does "Lupa" play for you - the original complaint suggests not?

gurtajs4, 2021-05-05:
I actually want to achieve sync between 2 raspberry pi players playing same 4k videos .
I haven't decided on how to do it yet but was hoping if i need to tweak few things in code to achieve this.  Hence the requirement to compile it

jc-kynesim, 2021-05-05:
Ho boy!  That's going to be exciting... Just FYI I do have a plausible idea of the complexity involved in what you are trying to do and "a few tweaks" seems really quite hopeful.  But - good luck!
Often the easiest way to get a working compile is to grab the sources with "apt-get source", the dependancies with "apt-get build-dep" and run the debian package builder.

gurtajs4, 2021-05-05:
No none are capable of playing 4K yet as we see its published. Let see how far it gets. Thanks.
Could you provide the patch for  **"running out of pu/coeff intermediate storage"**

jc-kynesim, 2021-05-05:
OK - If you really want it.
[0001-Allow-resize-of-pu-coeff-buffer-in-hwaccel-rpi.patch.gz](https://github.com/RPi-Distro/vlc/files/6427599/0001-Allow-resize-of-pu-coeff-buffer-in-hwaccel-rpi.patch.gz)
This is an ffmpeg patch - vlc uses ffmpeg shared libs so you don't need to rebuild vlc to use it

gurtajs4, 2021-05-05:
Ok let me give it a try :)

jc-kynesim, 2021-05-05:
If you can't play Lupa now this won't fix that...

gurtajs4, 2021-05-05:
No thats why i am creating my own image with VLC ffmpeg that can play 4K videos first.

gurtajs4, 2021-05-05:
Hey @jc-kynesim ,

I did as you asked  

sudo apt-get source ffmpeg
sudo apt-get build-dep ffmpeg
**debuild -b -uc -us**

But this does not build with HW Acceleration support. 
Do i need to use some other command or 

manually add in file debian/rules
add "–enable-mmal at the end of CONFIG :
??
Is there better procedure script to build ffmpeg/vlc

Sorry for asking lots of basic questions

gurtajs4, 2021-05-05:
Finally I was able to build and problem seems to be solved now. Thanks a lot @jc-kynesim  and @popcornmix  for help.

gurtajs4, 2021-05-05:
But now playing a H264 MP4 is an issue

VLC media player 3.0.12 Vetinari (revision 3.0.12-1-0-gd147bb5e7e)
[00591550] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[00515b58] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
[b397f9a0] mmal_codec decoder: VCSM init succeeded: CMA
[b397f9a0] mmal_codec decoder error: MMAL error 2 "ENOSPC"
[b397f9a0] mmal_codec decoder error: MMAL error reported by ctrl
[b397f9a0] mmal_codec decoder error: MMAL error 2 "ENOSPC"
[b397f9a0] mmal_codec decoder error: MMAL error 2 "ENOSPC"
[b397f9a0] mmal_codec decoder error: MMAL error 2 "ENOSPC"
[b397f9a0] mmal_codec decoder error: MMAL error 2 "ENOSPC"
[b397f9a0] mmal_codec decoder error: MMAL error 2 "ENOSPC"
[b397f9a0] main decoder error: buffer deadlock prevented

jc-kynesim, 2021-05-06:
I'm glad you've fixed it - what is your gpu_mem= in /boot (if any)?  Most of the buffers alloced come from CMA but some end up coming from gpu_mem for H264 (it is a bit confused at the moment).  I have "gpu_mem=256" - it almost certainly doesn't need to be that big but for dev work I just set it "big enough".

gurtajs4, 2021-05-06:
Hey @jc-kynesim ,

I tried setting "gpu_mem=256"  in config.txt,
Still see below issue

VLC media player 3.0.12 Vetinari
Command Line Interface initialized. Type `help' for help.
**[b247f7a0] mmal_codec decoder: VCSM init succeeded: CMA**
**[b247f7a0] main decoder error: buffer deadlock prevented**
[b0421168] gles2 generic error: parent window not available
[b0420c00] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
[b0421168] xcb generic error: window not available
[b0420c00] mmal_xsplitter vout display error: Failed to open Xsplitter:xcb_x11 module

Sample File - https://www.dropbox.com/s/o58z2i2pd0chse5/Test_h264.mp4?dl=0 

Are you able to play this on fresh RPI 4 image along with 4K video?

jc-kynesim, 2021-05-06:
Nope - I can't play that - it is a 4k stream - pi H264 h/w only deals with HD  (its the same h/w that was in a Pi1)- I'll grant that fallback/giveup could be cleaner but it isn't expected to work.

The errors in your log all all come down to "No X".

I'm guessing buffering or something like that for your glitch - decode gets behind for some reason (common enough on startup) before getting its act back together - if you are going to try syncing multiple Pis you are going to learn a LOT about this sort of thing.

jc-kynesim, 2021-05-06:
To be accurate re. errors the "deadlock" line isn't a "no X" error - it means that the decoder has a large input buffer and can accept a lot of frames before emitting one - so not an error really.

gurtajs4, 2021-05-06:
and how about the shuttering issue?

```
Also with the 4K that i got it working after compiling ffmpeg with your patch,
I see an issue, where when the video just start to play, it display the frames for 2 seconds and then the Monitor
goes blank/black and then the playback resume normal and continue? Any hints?
```

jc-kynesim, 2021-05-06:
I don't see that here with the stream in local storage on the Pi - how are you fetching the stream? If it is already local I don't have any good ideas.

gurtajs4, 2021-05-06:
Yes locally, is there a ways to enable more verbose logs to check what is the exact issue?

jc-kynesim, 2021-05-06:
Oh yes: add -vv to your command line and you'll get plenty of logging, but just to warn you you are about to be on your own here - I'm sorry but this is a rabbit hole that will take too much of my time but if you want to learn about vlc avsync for your project then this is a good place to start...

gurtajs4, 2021-05-06:
Understand, thanks

gurtajs4, 2021-05-06:
Looks like the initial shuttering is because of title being displayed, if i play it without title then it work fine.
Is there a way in code to disable title?

vlc --no-video-title

jc-kynesim, 2021-05-07:
grep through the source for the option string and set its default to the value you want.

gurtajs4, 2021-05-07:
Sure i will chase down that part. 
Few more clarification. Previously I tried building FFMpeg from RPI apt-get source and debain  pkg builder. But that version did not      worked for me may be it did not had "libavcodec/rpivid_hevc.c" at all... Not completly sure, so I ended up using ffmpeg from repository "https://github.com/RPi-Distro/ffmpeg.git" and then applied your patch and got things working

I tried build VLC from RPI source, facing some compilation issue related to MMAL 
https://github.com/RPi-Distro/ffmpeg had issues too and your repository https://github.com/jc-kynesim/vlc does not have debian pkg configuration too... Which one is the right source and correct way to build VLC of PI4. Perhaps if you have ./configure command, you could help with

graysky2, 2022-08-23:
I hit this on a search issue on a search.  I too would like to build VLC with HW decoding ability like RPiOS currently has but for Arch ARM 64bit.  I have thus far applied (I think) the right configure options and deps, but the resulting binary does not use HW decoding.

https://github.com/graysky2/PKGBUILDs/commits/vlc-try

Can someone confirm these step:
1. Build ffmpeg4.4 patched with @jc-kynesim's https://github.com/jc-kynesim/rpi-ffmpeg/tree/dev/4.4/rpi_import_1
2. Build VLC with the added options linked above but using source from https://github.com/RPi-Distro/vlc not from vanilla upstream

graysky2, 2022-08-24:
I was able to get the resulting thing to build, adding ` -I/usr/include/ffmpeg4.4` to the C/CXXFLAGS but it does not play:

```
% cvlc /mnt/media/jellyfish-90-mbps-hd-hevc.mkv                                                      
VLC media player 3.0.17.4 Vetinari (revision 3.0.13-8-g41878ff4f2)
[000000558aaea7a0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[000000558ab51c60] main interface error: no suitable interface module
[000000558aa29590] main libvlc error: interface "globalhotkeys,none" initialization failed
[000000558ab51c60] dummy interface: using the dummy interface module...
[0000007fa4c01f30] main decoder error: Codec `hevc' (MPEG-H Part2/HEVC (H.265)) is not supported.
[0000007fa4c01f30] main decoder error: Codec not supported
[0000007fa4c01f30] main decoder error: VLC could not decode the format "hevc" (MPEG-H Part2/HEVC (H.265))
```

[Here is a verbose log](https://gist.github.com/graysky2/c9bafb85b2515760179ceeceb9d0e45a) generated by `cvlc --file-logging --logfile archarm-custom.log -vv /mnt/media/jellyfish-90-mbps-hd-hevc-10bit.mkv`

graysky2, 2022-08-24:
@XECDesign - I noticed you commited to the RPi-Distro VLC repo.  Any thoughts?

jc-kynesim, 2022-08-25:
@graysky I've got a few recent comments from you in my email that claim
to be this issue but I can't see them on the web page. Any ideas what is
going on?

graysky2, 2022-08-25:
@jc-kynesim - Yes, I deleted my comments having successfully built VLC with hardware decoding for Arch ARM.  See: https://github.com/graysky2/PKGBUILDs/commits/vlc-rpi

In short, a modified patched ffmpeg is needed (your code) as well as some changes to vlc.  I got the patches from the RPi-Distro links above and studied the `rules` files to see which configure options where needed.  Beyond that, a tweak to my C/CXXFLAGS and a udev rule was all it took.

graysky2, 2022-08-25:
Thanks for the innovative work to make HW decoding possible and thanks to the RPiOS folks who put the packages together :+1: