RPi-Distro/vlc #45: Looping VLC Freezes after any amount of time

nixternal, 2021-06-04:
VLC command line: DISPLAY=:0 cvlc --vout gles2 -fq --no-osd --one-instance BigBuckBunny_2000hevc.mp4 -L

File used to reproduce this: https://s3.amazonaws.com/x265.org/video/BigBuckBunny_2000hevc.mp4

Latest VLC & System Updates as of June 3, 2021. Run VLC from python script & systemd. I can still get VLC to crash when not firing it from a script & systemd & just from the command line.

System info:
```
Hardware        : BCM2711
Revision        : c03112
Model           : Raspberry Pi 4 Model B Rev 1.2
Kernel          : 5.10.17-v7l+
```
```
BOOTLOADER: up to date
   CURRENT: Thu 29 Apr 2021 04:11:25 PM UTC (1619712685)
    LATEST: Thu 29 Apr 2021 04:11:25 PM UTC (1619712685)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
```

/boot/cmdline.txt:
`console=serial0,115200 console=tty1 root=PARTUUID=738a4d67-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait logo.nologo consoleblank=0 loglevel=0 quiet splash cma=512M`

/boot/config.txt:
```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# Disable Pi Rainbow Color Test
disable_splash=1

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
[... 147 lines cut ...]
Jun  3 23:01:35 foobar kernel: [44641.543154]  r5:00000000 r4:c3d26198
Jun  3 23:01:35 foobar kernel: [44641.543173] [<c04563dc>] (__dentry_kill) from [<c04575a0>] (dput.part.8+0x218/0x3d0)
Jun  3 23:01:35 foobar kernel: [44641.543188]  r7:00080040 r6:c0e3deac r5:c3d261e8 r4:c3d26198
Jun  3 23:01:35 foobar kernel: [44641.543206] [<c0457388>] (dput.part.8) from [<c0457770>] (dput+0x18/0x1c)
Jun  3 23:01:35 foobar kernel: [44641.543223]  r10:ffffe000 r9:00000000 r8:c3d26198 r7:c18cd790 r6:00080007 r5:c4b737e8
Jun  3 23:01:35 foobar kernel: [44641.543236]  r4:c4692900
Jun  3 23:01:35 foobar kernel: [44641.543254] [<c0457758>] (dput) from [<c043cda0>] (__fput+0xd8/0x25c)
Jun  3 23:01:35 foobar kernel: [44641.543272] [<c043ccc8>] (__fput) from [<c043cf3c>] (____fput+0x18/0x1c)
Jun  3 23:01:35 foobar kernel: [44641.543289]  r9:fffffe30 r8:c1354d6c r7:c2848000 r6:c2848554 r5:00000000 r4:c4692900
Jun  3 23:01:35 foobar kernel: [44641.543310] [<c043cf24>] (____fput) from [<c02429e8>] (task_work_run+0xb4/0xe4)
Jun  3 23:01:35 foobar kernel: [44641.543332] [<c0242934>] (task_work_run) from [<c0233dd0>] (get_signal+0x654/0xae0)
Jun  3 23:01:35 foobar kernel: [44641.543349]  r9:fffffe30 r8:5ac3c35a r7:c4d09a40 r6:b6ddf518 r5:c4692900 r4:ffffe000
Jun  3 23:01:35 foobar kernel: [44641.543369] [<c023377c>] (get_signal) from [<c020b87c>] (do_work_pending+0x30c/0x5a8)
Jun  3 23:01:35 foobar kernel: [44641.543386]  r10:c1205048 r9:fffffe30 r8:5ac3c35a r7:c426ff44 r6:b6ddf518 r5:c426ffb0
Jun  3 23:01:35 foobar kernel: [44641.543399]  r4:ffffe000
Jun  3 23:01:35 foobar kernel: [44641.543416] [<c020b570>] (do_work_pending) from [<c0200074>] (slow_work_pending+0xc/0x20)
Jun  3 23:01:35 foobar kernel: [44641.543430] Exception stack(0xc426ffb0 to 0xc426fff8)
Jun  3 23:01:35 foobar kernel: [44641.543445] ffa0:                                     00000007 80404a5a aa4adaa8 20636f6c
Jun  3 23:01:35 foobar kernel: [44641.543462] ffc0: b127870c 00878000 00000007 00000036 aa4adaa8 00000000 00000001 0000000b
Jun  3 23:01:35 foobar kernel: [44641.543478] ffe0: b02a9154 aa4ada8c b0295224 b6ddf518 a0000010 00000007
Jun  3 23:01:35 foobar kernel: [44641.543495]  r10:00000036 r9:c426e000 r8:c0200204 r7:00000036 r6:00000007 r5:00878000
Jun  3 23:01:35 foobar kernel: [44641.543508]  r4:b127870c
Jun  3 23:01:35 foobar kernel: [44641.543526] Code: e1c420d0 e5950000 e3a0cc01 e3001122 (e5823004) 
Jun  3 23:01:35 foobar kernel: [44641.543547] ---[ end trace f0ab2836e5589be3 ]---
```

jc-kynesim, 2021-06-04:
Thanks for that. I'm looking at VLC currently - I'll look into it. I'd like to blame the issue on a kernel driver problem (and I'm not ruling that out) but HEVC decode does have the ability to break things too.

nixternal, 2021-06-04:
Thanks @jc-kynesim! I just found it interesting that the dump looks similar to that Chrome one in the bug report for firmware. To be honest, I don't know if this has been happening all of the time, but I swear pre-February 26, 2021 it wasn't bombing like this, however I can't be positive. I do have a machine that is from that time frame & it ran for a week and didn't crash.

KenT2, 2021-06-08:
Similar reported here
[https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=312054](url)

dividuum, 2021-06-20:
I'd like to add another observation: My software info-beamer currently still uses a heavily modified [ffmpeg branch](https://github.com/info-beamer/FFmpeg) originally based on [this patch](https://github.com/popcornmix/FFmpeg/tree/2711). So while this is probably not immediately helpful with regards to VLC, I also observe the occasional and seemingly random playback lockup that I've not yet been able to reliably reproduce. Maybe these issues are related and the problem isn't VLC?

`dmesg` output follows. Similar to the original post, note the difference between the very first log line and all other lines. The first line is what caused my program to get stuck. The following lines get added once I forcefully terminate my program.
```
[ 4039.365072] vc_sm_cma_vchi_rx_ack: received response 1818324333, throw away...
[ 5023.137220] [vc_sm_cma_ioctl_alloc]: something failed - cleanup. ret -512
[ 5023.137338] ------------[ cut here ]------------
[ 5023.137361] WARNING: CPU: 3 PID: 6927 at mm/memory.c:2519 __apply_to_page_range+0x56c/0x570
[ 5023.137368] Modules linked in: rpivid_mem v3d gpu_sched vc4 cec drm_kms_helper drm fb_sys_fops sysfillrect sysimgblt syscopyarea drm_panel_orientation_quirks vc_sm_cma(C) enc28j60 pwm_bcm2835 rtc_ds1307 rtc_pcf8523 i2c_bcm2708 i2c_dev regmap_i2c ftdi_sio pl2303 cp210x usbserial rpi_poe_fan raspberrypi_hwmon dm_crypt dm_mod rpi_backlight backlight hid_multitouch goodix snd_soc_core snd_pcm_dmaengine snd_compress snd_bcm2
[ 5023.137846] CPU: 3 PID: 6927 Comm: video buffer Tainted: G         C        5.10.17-v7l+ #1403
[ 5023.137850] Hardware name: BCM2711
[ 5023.137854] Backtrace:
[ 5023.137868] [<c0b63050>] (dump_backtrace) from [<c0b633e4>] (show_stack+0x20/0x24)
[ 5023.137875]  r7:ffffffff r6:00000000 r5:60000113 r4:c12e69fc
[ 5023.137883] [<c0b633c4>] (show_stack) from [<c0b67774>] (dump_stack+0xcc/0xf8)
[ 5023.137892] [<c0b676a8>] (dump_stack) from [<c0220bac>] (__warn+0xfc/0x114)
[ 5023.137899]  r10:00000001 r9:00000009 r8:c03f06ec r7:000009d7 r6:00000009 r5:c03f06ec
[ 5023.137903]  r4:c0e3a78c r3:c1205094
[ 5023.137910] [<c0220ab0>] (__warn) from [<c0b63b54>] (warn_slowpath_fmt+0x70/0xd8)
[ 5023.137916]  r7:000009d7 r6:c0e3a78c r5:c1205048 r4:00000000
[ 5023.137924] [<c0b63ae8>] (warn_slowpath_fmt) from [<c03f06ec>] (__apply_to_page_range+0x56c/0x570)
[ 5023.137929]  r9:ddd00000 r8:ddd00000 r7:ddd00000 r6:00000000 r5:cd997000 r4:ffffffea
[ 5023.137937] [<c03f0180>] (__apply_to_page_range) from [<c03f0718>] (apply_to_page_range+0x28/0x30)
[ 5023.137943]  r10:00000001 r9:ddd00000 r8:ddd00000 r7:c22f3010 r6:00000000 r5:cd997000
[ 5023.137946]  r4:c1205048
[ 5023.137955] [<c03f06f0>] (apply_to_page_range) from [<c0215f80>] (cma_allocator_free+0xd4/0xe8)
[ 5023.137961] [<c0215eac>] (cma_allocator_free) from [<c0216a38>] (__arm_dma_free.constprop.3+0x190/0x218)
[ 5023.137967]  r9:00000000 r8:ddd00000 r7:00000000 r6:c3d10d40 r5:ddd00000 r4:c1205048
[ 5023.137973] [<c02168a8>] (__arm_dma_free.constprop.3) from [<c0216aec>] (arm_dma_free+0x2c/0x34)
[ 5023.137979]  r10:00000000 r9:00000001 r8:00080060 r7:ddd00000 r6:c22f3010 r5:00000000
[ 5023.137983]  r4:ddd00000
[ 5023.137989] [<c0216ac0>] (arm_dma_free) from [<c0217084>] (arch_dma_free+0x2c/0x34)
[ 5023.137993]  r5:00000000 r4:ddd00000
[ 5023.138003] [<c0217058>] (arch_dma_free) from [<c02a2938>] (dma_direct_free+0x12c/0x174)
[ 5023.138007]  r5:ddd00000 r4:ddd00000
[ 5023.138014] [<c02a280c>] (dma_direct_free) from [<c02a1158>] (dma_free_attrs+0x80/0xd0)
[ 5023.138020]  r10:c3b57cb0 r9:00000001 r8:00080060 r7:00000000 r6:00000000 r5:ddd00000
[ 5023.138024]  r4:c22f3010
[ 5023.138043] [<c02a10d8>] (dma_free_attrs) from [<bf502d04>] (vc_sm_release_resource+0xe4/0x104 [vc_sm_cma])
[ 5023.138048]  r7:00000000 r6:ddd00000 r5:bf5072c0 r4:c3751380
[ 5023.138062] [<bf502c20>] (vc_sm_release_resource [vc_sm_cma]) from [<bf502dc0>] (vc_sm_dma_buf_release+0x54/0xa8 [vc_sm_cma])
[ 5023.138067]  r7:00080040 r6:c3b87270 r5:c3751390 r4:c3751380
[ 5023.138078] [<bf502d6c>] (vc_sm_dma_buf_release [vc_sm_cma]) from [<c0871990>] (dma_buf_release+0x54/0x90)
[ 5023.138082]  r5:00000000 r4:c3ca0b40
[ 5023.138092] [<c087193c>] (dma_buf_release) from [<c045651c>] (__dentry_kill+0x140/0x1d0)
[ 5023.138096]  r5:00000000 r4:c3b87220
[ 5023.138103] [<c04563dc>] (__dentry_kill) from [<c04575a0>] (dput.part.8+0x218/0x3d0)
[ 5023.138108]  r7:00080040 r6:c0e3deac r5:c3b87270 r4:c3b87220
[ 5023.138113] [<c0457388>] (dput.part.8) from [<c0457770>] (dput+0x18/0x1c)
[ 5023.138119]  r10:ffffe000 r9:00000000 r8:c3b87220 r7:c18cd790 r6:00080007 r5:c3b57cb0
[ 5023.138122]  r4:c31a8e40
[ 5023.138129] [<c0457758>] (dput) from [<c043cda0>] (__fput+0xd8/0x25c)
[ 5023.138134] [<c043ccc8>] (__fput) from [<c043cf3c>] (____fput+0x18/0x1c)
[ 5023.138140]  r9:fffffe30 r8:c1354d6c r7:c3790f80 r6:c37914d4 r5:00000000 r4:c31a8e40
[ 5023.138150] [<c043cf24>] (____fput) from [<c02429e8>] (task_work_run+0xb4/0xe4)
[ 5023.138159] [<c0242934>] (task_work_run) from [<c0233dd0>] (get_signal+0x654/0xae0)
[ 5023.138165]  r9:fffffe30 r8:5ac3c35a r7:c2d24ec0 r6:b6541518 r5:c31a8e40 r4:ffffe000
[ 5023.138172] [<c023377c>] (get_signal) from [<c020b87c>] (do_work_pending+0x30c/0x5a8)
[ 5023.138178]  r10:c1205048 r9:fffffe30 r8:5ac3c35a r7:c3df5f44 r6:b6541518 r5:c3df5fb0
[ 5023.138182]  r4:ffffe000
[ 5023.138188] [<c020b570>] (do_work_pending) from [<c0200074>] (slow_work_pending+0xc/0x20)
[ 5023.138192] Exception stack(0xc3df5fb0 to 0xc3df5ff8)
[... cut ...]
info-beamer-3878290628 /space # uname -a
Linux info-beamer-3878290628 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux

info-beamer-3878290628 /space # vcgencmd version
Feb 25 2021 12:11:39 
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start_x)

info-beamer-3878290628 /space # vcgencmd bootloader_version
Oct  2 2020 08:44:17
version bfe1c7ddc094b735134e96f2338c0ab2c9b10a5b (release)
timestamp 1601624657
update-time 0
capabilities 0x00000000
```

dividuum, 2021-06-20:
And another one. Pi4 (d03114):

```
[23252.649143] vc_sm_cma_vchi_rx_ack: received response 0, throw away...
[33300.648109] [vc_sm_cma_ioctl_alloc]: something failed - cleanup. ret -512
[33300.648201] ------------[ cut here ]------------
[33300.648221] WARNING: CPU: 2 PID: 3593 at mm/memory.c:2519 __apply_to_page_range+0x56c/0x570
[33300.648228] Modules linked in: rpivid_mem v3d gpu_sched vc4 cec drm_kms_helper drm fb_sys_fops sysfillrect sysimgblt syscopyarea drm_panel_orientation_quirks vc_sm_cma(C) snd_soc_bcm2835_i2s snd_soc_pcm512x_i2c snd_soc_pcm512x snd_soc_pcm5102a snd_soc_rpi_simple_soundcard snd_soc_hifiberry_dacplus clk_hifiberry_dacpro enc28j60 pwm_bcm2835 rtc_ds1307 rtc_pcf8523 i2c_bcm2708 i2c_dev regmap_i2c ftdi_sio pl2303 cp210x usbserial rpi_poe_fan raspberrypi_hwmon dm_crypt dm_mod rpi_backlight backlight hid_multitouch goodix snd_soc_core snd_pcm_dmaengine snd_compress snd_bcm2835(C) snd_pcm snd_timer snd rt2800usb rt2800lib rt2x00usb rt2x00lib brcmfmac brcmutil rtl8192cu rtl8192c_common rtl_usb rtlwifi xpad ff_memless tun i2c_bcm2835 overlay lib80211_crypt_tkip lib80211_crypt_ccmp lib80211 mac80211 cfg80211 rfkill xt_owner xt_REDIRECT iptable_nat xt_MASQUERADE nf_nat xt_state nf_conntrack ipt_REJECT nf_reject_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 xt_tcpudp iptable_filter ip_tables x_tables crc_ccitt
[33300.648695]  squashfs libarc4 ctr ccm crypto_simd cryptd sha256_generic
[33300.648744] CPU: 2 PID: 3593 Comm: video buffer Tainted: G         C        5.10.17-v7l+ #1403
[33300.648748] Hardware name: BCM2711
[33300.648753] Backtrace: 
[33300.648767] [<c0b63050>] (dump_backtrace) from [<c0b633e4>] (show_stack+0x20/0x24)
[33300.648774]  r7:ffffffff r6:00000000 r5:60000013 r4:c12e69fc
[33300.648782] [<c0b633c4>] (show_stack) from [<c0b67774>] (dump_stack+0xcc/0xf8)
[33300.648790] [<c0b676a8>] (dump_stack) from [<c0220bac>] (__warn+0xfc/0x114)
[33300.648796]  r10:00000001 r9:00000009 r8:c03f06ec r7:000009d7 r6:00000009 r5:c03f06ec
[33300.648800]  r4:c0e3a78c r3:c1205094
[33300.648807] [<c0220ab0>] (__warn) from [<c0b63b54>] (warn_slowpath_fmt+0x70/0xd8)
[33300.648812]  r7:000009d7 r6:c0e3a78c r5:c1205048 r4:00000000
[33300.648820] [<c0b63ae8>] (warn_slowpath_fmt) from [<c03f06ec>] (__apply_to_page_range+0x56c/0x570)
[33300.648826]  r9:d3100000 r8:d3100000 r7:d3100000 r6:00000000 r5:c8f93000 r4:ffffffea
[33300.648834] [<c03f0180>] (__apply_to_page_range) from [<c03f0718>] (apply_to_page_range+0x28/0x30)
[33300.648839]  r10:00000001 r9:d3100000 r8:d3100000 r7:c235c410 r6:00000000 r5:c8f93000
[33300.648843]  r4:c1205048
[33300.648851] [<c03f06f0>] (apply_to_page_range) from [<c0215f80>] (cma_allocator_free+0xd4/0xe8)
[33300.648858] [<c0215eac>] (cma_allocator_free) from [<c0216a38>] (__arm_dma_free.constprop.3+0x190/0x218)
[33300.648863]  r9:00000000 r8:d3100000 r7:00000000 r6:c42a7ac0 r5:d3100000 r4:c1205048
[33300.648870] [<c02168a8>] (__arm_dma_free.constprop.3) from [<c0216aec>] (arm_dma_free+0x2c/0x34)
[33300.648875]  r10:00000000 r9:00000001 r8:00080060 r7:d3100000 r6:c235c410 r5:00000000
[33300.648879]  r4:d3100000
[33300.648885] [<c0216ac0>] (arm_dma_free) from [<c0217084>] (arch_dma_free+0x2c/0x34)
[33300.648890]  r5:00000000 r4:d3100000
[33300.648898] [<c0217058>] (arch_dma_free) from [<c02a2938>] (dma_direct_free+0x12c/0x174)
[33300.648903]  r5:d3100000 r4:d3100000
[33300.648910] [<c02a280c>] (dma_direct_free) from [<c02a1158>] (dma_free_attrs+0x80/0xd0)
[33300.648916]  r10:c3fa6330 r9:00000001 r8:00080060 r7:00000000 r6:00000000 r5:d3100000
[33300.648920]  r4:c235c410
[33300.648938] [<c02a10d8>] (dma_free_attrs) from [<bf503d04>] (vc_sm_release_resource+0xe4/0x104 [vc_sm_cma])
[33300.648943]  r7:00000000 r6:d3100000 r5:bf5082c0 r4:c4107c80
[33300.648957] [<bf503c20>] (vc_sm_release_resource [vc_sm_cma]) from [<bf503dc0>] (vc_sm_dma_buf_release+0x54/0xa8 [vc_sm_cma])
[33300.648962]  r7:00080040 r6:c3f88490 r5:c4107c90 r4:c4107c80
[33300.648974] [<bf503d6c>] (vc_sm_dma_buf_release [vc_sm_cma]) from [<c0871990>] (dma_buf_release+0x54/0x90)
[33300.648979]  r5:00000000 r4:c311e540
[33300.648987] [<c087193c>] (dma_buf_release) from [<c045651c>] (__dentry_kill+0x140/0x1d0)
[33300.648991]  r5:00000000 r4:c3f88440
[33300.648998] [<c04563dc>] (__dentry_kill) from [<c04575a0>] (dput.part.8+0x218/0x3d0)
[33300.649002]  r7:00080040 r6:c0e3deac r5:c3f88490 r4:c3f88440
[33300.649008] [<c0457388>] (dput.part.8) from [<c0457770>] (dput+0x18/0x1c)
[33300.649013]  r10:ffffe000 r9:00000000 r8:c3f88440 r7:c18cd790 r6:00080007 r5:c3fa6330
[33300.649017]  r4:c2aec0c0
[33300.649024] [<c0457758>] (dput) from [<c043cda0>] (__fput+0xd8/0x25c)
[33300.649030] [<c043ccc8>] (__fput) from [<c043cf3c>] (____fput+0x18/0x1c)
[33300.649035]  r9:fffffe30 r8:c1354d6c r7:c2ffdd00 r6:c2ffe254 r5:00000000 r4:c2aec0c0
[33300.649043] [<c043cf24>] (____fput) from [<c02429e8>] (task_work_run+0xb4/0xe4)
[33300.649052] [<c0242934>] (task_work_run) from [<c0233dd0>] (get_signal+0x654/0xae0)
[33300.649057]  r9:fffffe30 r8:5ac3c35a r7:c3518a80 r6:b64a9518 r5:c2aec0c0 r4:ffffe000
[33300.649064] [<c023377c>] (get_signal) from [<c020b87c>] (do_work_pending+0x30c/0x5a8)
[33300.649070]  r10:c1205048 r9:fffffe30 r8:5ac3c35a r7:c437ff44 r6:b64a9518 r5:c437ffb0
[33300.649074]  r4:ffffe000
[33300.649080] [<c020b570>] (do_work_pending) from [<c0200074>] (slow_work_pending+0xc/0x20)
[33300.649084] Exception stack(0xc437ffb0 to 0xc437fff8)
[33300.649089] ffa0:                                     00000014 80404a5a 9ae04a38 763e2d70
[... cut ...]
info-beamer-2776321857 /space/log # uname -a
Linux info-beamer-2776321857 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux

info-beamer-2776321857 /space/log # vcgencmd version
Feb 25 2021 12:11:39 
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start_x)

info-beamer-2776321857 /space/log # vcgencmd bootloader_version
Apr 29 2021 17:11:25
version c2f8c388c4ee37ad709ace403467d163e8dd91ce (release)
timestamp 1619712685
update-time 1619712685
capabilities 0x0000001f
```

nixternal, 2022-01-12:
Here to confirm the issue persists in Raspbian 11 (bullseye) unfortunately. Not using FKMS with the new Raspbian version, or really anything out of ordinary. Took Raspbian Lite, installed X, Nodm, Matchbox Window Manager, and VLC.

`cvlc -f --no-osd --one-instance media/BigBuckBunny_2000hevc-1.mp4 -L`

nixternal, 2022-01-24:
OK, something I have noticed. https://www.libde265.org/hevc-bitstreams/bbb-3840x2160-cfg02.mkv

mediainfo command says CODEC ID: V_MPEGH/ISO/HEVC

If I clip out just 5 seconds of that video in Handbrake & keep it HEVC/H.265, it still has that for a CODEC ID. However, if you clip it with Adobe or whatever Windows uses, it shows CODEC ID: hvc1.

hvc1 crashes, V_MPEGH/ISO/HEVC does not crash.

nixternal, 2022-01-27:
Also noticed that H.264 is still running just fine as well.