RPi-Distro/vlc #46: 1080p videos causes the screen to blank

KenT2, 2021-06-05:
I have had lots of problems with 1080p videos blanking the screen intermittently. Have finally pinned it down to window size (hopefully).

When playing 1080p videos with the attached bash script the screen blanks at the start of each video.
if the window is changed to 1600x900 or 400x225 then blanking does not occur.
If HDMI-2 is used blanking of HDMI-1 is less prevalent
With Jellyfish blanking is less prevalent,
Blanking does not occur on 720p versions of the files produced with Handbrake

GPUMEM is 512

Videos are here:
http://www.museumoftechnology.org.uk/ken/Big_Buck_Bunny_1080_10s_30MB.mp4
http://www.museumoftechnology.org.uk/ken/Jellyfish_1080_10s_5MB.mp4

[disp.sh.txt](https://github.com/RPi-Distro/vlc/files/6602779/disp.sh.txt)

KenT2, 2021-06-19:
Another similar problem, this time when using 2 screens.
Video window size is 400x225 which displays 1080p videos OK on a single screen.
Display a 720p video on HDMI-1 then display a 1080p video (BBB) on HDMI-2. HDMI-1 blanks intermittently.
If two  720p videos are used then all is OK
GPUMEM is 512 and I have just done a full-upgrade and reboot
Script as in previous post

popcornmix, 2021-06-21:
Does force_turbo=1 in config,txt help?

KenT2, 2021-06-22:
Yes this solves all the problems.

Should I recommend this to Pi Presents users or is it something that can be fixed in the code? The use in Pi Presents is to reduce the gap between videos by loading one while the other is paused at the end.

Interestingly processor occupation is only a few percent Maybe there is a short peak load requirements.

popcornmix, 2021-06-23:
It sounds like an issue with core_freq being too low.
Try with `core_freq_min=500` rather than `force_turbo=1` (only boosts core, rather than core, arm, v3d etc).

Just to confirm, this is on a Pi4 using default fkms driver and happens with a single display (as well as with two displays), and occurs with that single display running at 1920x1080@60 resolution? The video files are H.264.

KenT2, 2021-06-23:
core_freq_min=500 makes everything work OK
The original test was done with 2 monitors (second one VGA). I have just repeated the original test with one monitor with the conditions as in your post above. (video is 30 fps, monitor is 60)
The blanking occurred with a mmal-vout-window of 800*450 but not with some other sizes. I have just found out that if I add --no-video-title it works OK at 800*450

popcornmix, 2021-06-23:
@timg236 sounds like an issue with firmware core clock calculation for hdmi.