RPi-Distro/vlc #48: VLC on Raspian

akoryuerek, 2021-10-17:
Hi Guys,

vlc doesn’t work since the last kernel update on raspberry. When I’m downgrading the kernel with rpi-update 1e494f150046bd04331ab0b19e4e5322eb7b7773 It´s all ok again. I assume it depends on the new video core firmware which is included with the latest update. Will be there a fix or is this issue already known? Please feedback.

popcornmix, 2021-10-17:
Is 1e494f150046bd04331ab0b19e4e5322eb7b7773 the most recent firmware/kernel that still works?
i.e. is 15471b6e99b6db2f3206f0aed67d0aebbfc9f9cf the first broken update?

akoryuerek, 2021-10-17:
1e494f150046bd04331ab0b19e4e5322eb7b7773 is ok, but 15471b6e99b6db2f3206f0aed67d0aebbfc9f9cf fails.

popcornmix, 2021-10-17:
@6by9  Looks like "video_decode: Ensure all buffers are flushed before port disable completes".

@akoryuerek can you describe what you are playing that fails? e.g. does:
https://download.blender.org/peach/bigbuckbunny_movies/big_buck_bunny_1080p_h264.mov
fail?

akoryuerek, 2021-10-17:
Local  files are working without any problem, but when im trying to stream it then it doesn’t work and it also doesn’t matter what kind of streaming protocol I am using.  I tried with http an rtsp.

popcornmix, 2021-10-18:
We are looking for a way of reproducing this.
Can you link to a publicly accessible streaming url that works on older firmware/kernel and fails on newer?

JRDavisUF, 2021-10-19:
rpi-update successfully updated to ffdfc6e625eb38086b8c15eead2be90acab825c2 and problem is fixed.  thx!