RPi-Distro/vlc #50: Bullseye // KMS // VLC

akoryuerek, 2021-11-17:
Hi Guys, 

is it currently possible to play a stream with Bullseye or should we better keep Buster? Following some logs:

Nov 16 18:45:07 rasp-bullseye rpisurv[592]: [hevc @ 0xaa800b10] hardware accelerator failed to decode picture
Nov 16 18:45:07 rasp-bullseye rpisurv[592]: [b340add8] mmal_avcodec decoder error: avcodec_send_packet critical error
Nov 16 18:45:07 rasp-bullseye rpisurv[592]: [hevc @ 0xaa800b10] Could not find ref with POC 4
Nov 16 18:45:07 rasp-bullseye rpisurv[592]: usrvcsm: [vcsm_malloc_cache]: [592] [VLC frame]: ioctl mem-alloc FAILED [-1] (hdl: 0)
Nov 16 18:45:07 rasp-bullseye rpisurv[592]: [b340add8] mmal_avcodec decoder error: CMA buf pool alloc buf failed
Nov 16 18:45:07 rasp-bullseye rpisurv[592]: rpi_get_display_buffer: Failed to get buffer from pool
Nov 16 18:45:07 rasp-bullseye rpisurv[592]: [hevc @ 0xaa800b10] Failed to allocate output frame
Nov 16 18:45:07 rasp-bullseye rpisurv[592]: [hevc @ 0xaa800b10] hardware accelerator failed to decode picture
Nov 16 18:45:07 rasp-bullseye rpisurv[592]: [b340add8] mmal_avcodec decoder error: avcodec_send_packet critical erroR

jc-kynesim, 2021-11-17:
On the whole buster is going to be easier for a while. I have a V4L2 / DRM implementation for VLC4 but it isn't ready for prime time yet, and in all honesty, due to the limitations imposed by linux vs mmal the buster version is likely to work "better" for many purposes for at least some time.

SvenVD, 2022-01-09:
Is there an ETA, when VLC on bullseye is on par with VLC on buster?
Thanks

redneckmatt, 2022-07-26:
I got around this bullseye-induced issue for my project by setting the GL driver to "Legacy" in raspi-config.

My VLC cli-only onvif viewer project required the following feature that doesn't seem to have a KMS alternative:
--mmal-vout-window