RPi-Distro/vlc #51: Is Pi Zero 2 W supported for VLC? Is it known to work with Raspian OS Lite "Bullseye"?

MikeyWetzel, 2021-11-19:
I have a Pi Zero 2 W, and installed Raspian OS Lite (32-bit, which is the only choice) using the Raspberry Pi Imager. This version is dated 10-30-2021 and is called "Bullseye". I also did the obligatory update. Then installed VLC with "sudo apt install vlc". All is well so far.

Problem is that ANY video I play...and I've tried many...and tried many options....only shows a (fullscreen) black screen. Audio plays through my monitor connected to HDMI, but there is no video. I'm headless, so there's no GUI, and the system stays fullscreen black and is unresponsive. I've tried vlc and cvlc, I've tried using --mmal-display (even those there's only one HDMI.) I can't tell if the system is hung, or if I'm just no pressing the magic key combo to get back to my terminal. Reboot is my only option.

Catzou, 2021-11-20:
Hi,

I'm having the same issue here on my Raspi Zero 2W, i'm using the full version of the OS here with the desktop, and all i got from VLC is audio with black screen.

When starting VLC with the terminal, this is what i get : 

`VLC media player 3.0.16 Vetinari (revision 1.0.6-1682-g88158c836)
[005fcb98] main libvlc: Lancement de vlc avec l'interface par défaut. Utiliser « cvlc » pour démarrer VLC sans interface.
libEGL warning: DRI2: failed to authenticate
[61919800] mmal_codec decoder: VCSM init succeeded: CMA
mmal: mmal_component_create_core: could not find component 'vc.ril.video_decode'
[61919800] mmal_codec decoder error: Failed to create MMAL component vc.ril.video_decode (status=4 ENOSYS)
libEGL warning: DRI2: failed to authenticate
[6250b088] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
[6250b088] mmal_xsplitter vout display error: Failed to open Xsplitter:mmal_vout module
mmal: mmal_vc_component_create: failed to create component 'vc.ril.hvs' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.hvs' (1)
mmal: mmal_vc_component_create: failed to create component 'vc.ril.hvs' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.hvs' (1)
`

The only i manage to play video is to specify --codec="ffmpeg" but it's all choppy and doesn't work properly (and i'm not sure what i'm doing here ...)

pbx33, 2022-03-07:
> Hi,
>
> I'm having the same issue here on my Raspi Zero 2W, i'm using the full version of the OS here with the desktop, and all i got from VLC is audio with black screen.
>
> When starting VLC with the [...]

..same hw/sw.. same issue/errors

pbx33, 2022-03-08:
Here are two solutions which are working for my Rasp zero 2 w after struggling around a few days:
1. Buy MPG2 hw-decode license, add it to boot/config.txt ...reboot...then my mpg2 videos are playing nice with rasp os 32 bit and vlc bullseye 'out of the box'
(related to mpeg2 decoding in my case, but maybe same if you activate other codecs for hw-acceleration)
...
or
...
2. make vlc to prefer “ffmpeg“ video-decoder in advanced settings (maybe u have to disable hw acceleration in the "ffmpeg" decoder preferences)

Good luck

hamedtaha, 2022-05-10:
Hello 
I have the same issue with Pi Zero W and no way VLC can play any video. this was super annoying 
 
I've tried the below option, purchased a  MPG2 hw-decode license and added it to config,txt but it didn't work 
VLC still can't play

> 1. Buy MPG2 hw-decode license, add it to boot/config.txt ...reboot...then my mpg2 videos are playing nice with rasp os 32 bit and vlc bullseye 'out of the box'
>    (related to mpeg2 decoding in my case, but maybe same if you activate other codecs for hw-acceleration)

decoder preferences
[image]
[image]
[image]
)

bcink, 2022-10-31:
Change video codec output to X11 video output XBC

lmitcham, 2023-05-31:
I have the same issue if I use the GUI or call the python script from the local command line:

      import vlc
 
      # creating vlc media player object
      media = vlc.MediaPlayer("1.mp4")
 
      # start playing video
      media.play()

HOWEVER, if I SSH into the Pi and run the same command line script from there ... I see the video fullscreen on the Pi HDMI display.

This tells me vlc on the Zero 2W is possible, with the correct command/setup.  Is there anyone with an idea of how to approach this issue?

mowens3, 2025-03-26:
on Raspberry Zero 2W you need to set the default audio device in alsamixer so it's not set to hdmi by default