RPi-Distro/vlc #54: VLC Won't Play ffmpeg Timelapse Video Files Properly

ghalfacree, 2021-12-18:
While working on the book chapter which revealed [this bug](https://github.com/RPi-Distro/vlc/issues/53), I've come across what appears to be another: files created from JPG images using ffmpeg don't play properly in VLC.

## Reproduction
```
libcamera-still --width 1920 --height 1080 -t 100000 --timelapse 10000 -o %5d.jpg
ffmpeg -r 0.5 -i %05d.jpg animation.mp4
vlc animation.mp4
```

## Expected result
Video file plays back normally.

## Actual result
On the Raspberry Pi, the video file is entirely black for the first ~4 seconds; the remainder begins to play, starting at a frame created from the third image file, but begins to skip frames and stops before reaching frames created from the last image files.

On an Ubuntu 20.04.3 desktop, the video file opens at the beginning and appears normal at first; however, it exhibits the same skipped-frame glitching as on the Raspberry Pi and stops before reaching frames created from the last image file.

The same video loaded into the Videos (Totem) application on the desktop plays fine from the start, at the correct frame rate, with no glitching, ending on the correct frame.

## Versions
Raspberry Pi OS "Bullseye"
Linux raspberrypi 5.10.63-v7l+ #1488 SMP Thu Nov 18 16:15:28 GMT 2021 armv7l GNU/Linux

ffmpeg version 4.3.3-0+rpt2+deb11u1 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 10 (Raspbian 10.2.1-6+rpi1)
  configuration: --prefix=/usr --extra-version=0+rpt2+deb11u1 --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-mmal --enable-neon --enable-rpi --enable-v4l2-request --enable-libudev --enable-epoxy --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared --libdir=/usr/lib/arm-linux-gnueabihf --cpu=arm1176jzf-s --arch=arm

VLC media player 3.0.16 Vetinari (revision 1.0.6-1682-g88158c836)
VLC version 3.0.16 Vetinari (1.0.6-1682-g88158c836)
Compiled by pi on serge-testpi (Nov 10 2021 15:40:45)
Compiler: gcc version 10.2.1 20210110 (Raspbian 10.2.1-6+rpi1)

## Supporting information
The problem appears to be VLC, rather than ffmpeg: the files play fine in non-VLC playback software, and I have confirmed that the same occurs if I use the version of ffmeg - 4.2.4 - installed on my desktop rather than making the file on the Raspberry Pi.

I have attached a sample video file: it should show my (upside-down, as I forgot to flip it) fingers counting from a closed hand up to five extended fingers and back down to just two extended fingers played at a framerate of 0.5 FPS (two seconds per image.) I note that it plays properly in GitHub's built-in video player.

https://user-images.githubusercontent.com/1130041/146644257-cd38b770-cc0a-4ca0-9217-97d721952180.mp4

XECDesign, 2022-01-06:
@popcornmix @jc-kynesim

jc-kynesim, 2022-01-06:
I'll have a look

jc-kynesim, 2022-01-06:
Just for the record I'll note that, whilst I might expect 2Hz stuff to play back plausibly, once you push the frame interval out to a few secs then a lot of stuff is going to get confused. Video players have to be able to deal with discontinuities in timestamp and recover from that, rather than waiting around for a long time, and somewhere in the 2-10secs range is the normal discontinuity threshold.