RPi-Distro/vlc #60: "no matching alpha blending routine" when adding logo overlay

ErikApption, 2022-02-20:
I'm trying to overlay a logo in vlc on my RPi (Bullseye) but when I add the logo using:
`cvlc -v --random --loop --fullscreen $xid --video-filter "logo{file=weather.jpg,position=1}" playlist.xspf`

but this gives me the following error
``
VLC media player 3.0.16 Vetinari (revision 1.0.6-1682-g88158c836)
Remote control interface initialized. Type `help' for help.
[00da3b68] dummy interface: using the dummy interface module...
[e35012a0] playlist stream warning: skipping unexpected element <vlc:item>
[e35012a0] playlist stream warning: skipping unexpected element <vlc:item>
[e3901650] mp4 stream warning: unknown box type cTIM (incompletely loaded)
[e3901650] mp4 stream warning: unknown box type cTSC (incompletely loaded)
[e3901650] mp4 stream warning: unknown box type cTSZ (incompletely loaded)
[e3901650] mp4 stream warning: Unknown uuid type box
[e390bab0] mp4 demux warning: elst box found
[e390bab0] mp4 demux warning: STTS table of 1 entries
[e390bab0] mp4 demux warning: elst box found
[e390bab0] mp4 demux warning: STTS table of 1 entries
[dce73e20] mmal_codec decoder: VCSM init succeeded: CMA
[dce73e20] main decoder error: buffer deadlock prevented
[e3501fe0] gles2 gl: Initialized libplacebo v2.72.0 (API v72)
[e3502ac0] mmal_xsplitter vout display error: Failed to open Xsplitter:mmal_vout module
[d1515ea8] blend blend error: no matching alpha blending routine (chroma: YUVA -> MMAL)
[d151ec88] logo filter error: failed to blend a picture
[d1515ea8] blend blend error: no matching alpha blending routine (chroma: YUVA -> MMAL)
[d151ec88] logo filter error: failed to blend a picture
[d1515ea8] blend blend error: no matching alpha blending routine (chroma: YUVA -> MMAL)
``

pettedemon, 2022-02-25:
Hi, I have the same problem!
`cvlc --fullscreen --no-osd --loop --file-caching=10000 --sub-source logo --logo-position 10 --logo-file /home/pi/Desktop/logo/logo-tv.png /home/pi/Desktop/php/scala.m3u`

ErikApption, 2022-02-25:
This issue was addressed in the vlc forums
https://forum.videolan.org/viewtopic.php?p=525059#p525059

pettedemon, 2022-02-25:
@ErikApption so what is the solution?
thanks

ErikApption, 2022-02-25:
haven't figured out the solution yet but I assume it is something with --sub-source logo which is a different type of filter.

pettedemon, 2022-02-25:
I try different options but I have no result in RP 3b+. Can I ask you if you any problems dropped frame? 
In Omxplayer I haven't any problem, now some dropped frames. I think to try Kodi but I don't know if it can use as vlc like a playlist player with a logo.
thanks

ErikApption, 2022-02-25:
honestly many of these options are obscure with vlc... one thing I would do is test the same overlay on a windows/linux box to see if it is a rpi or vlc bug. And yes I have constantly dropped frames but on bullseye, vlc is one of the few players with hardware accelleration.

pettedemon, 2022-02-25:
I try with a virtual machine in virtual box and I have no problem with the logo... so I try in RP3 and it doesn't work.

ErikApption, 2022-02-25:
@pettedemon which option did work on a virtual machine?

pettedemon, 2022-02-25:
Sorry it is the desktop version
`PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"`
is not raspbian

with this command no problem with logo
`cvlc --sub-source logo --logo-position 10 --logo-file /home/pi/Desktop/video/tv.png FSHN-00059.mp4`

jc-kynesim, 2022-03-03:
The problem is YUVA.  By default you get h/w blending but our h/w doesn't support YUVA - the only formats with A are the various permutations of RGBA - so you want a logo in a format that decodes to RGBA.  Really VLC should be able to do that but it is probable that the pipeline creation code is just not up to the job of doing that automatically.

pettedemon, 2022-03-03:
So, what you suggest?

ErikApption, 2022-03-03:
@jc-kynesim - I assume this means that Vlc needs a feature to convert logos to RGBA and blend them in RGBA without going to YUVA? Might be worth wording this clearly in their tracker, I think VLC 4 is around the corner

pettedemon, 2022-03-10:
Hi,
I try to set the logo in the GUI interface and set bottom-right. So next I try to launch vlc from command line...
a simple 

`cvlc -f video.mp4`

now if I do ALT+C and ALT+F the logo appear. I try to search what I do so 
ALT + C = Crop one pixel from the bottom of the video
ALT + F = Crop one pixel from the right of the video
and the logo appear on the bottom right.

So how can do this in command line? There is no problem with the logo but only the right command. Thanks

pettedemon, 2022-03-11:
Hi, I try to set the crop function
`cvlc -f --crop=16:9 --video-filter=croppadd{cropbottom=0,cropright=0,croptop=0,cropleft=0} /home/pi/Desktop/video/MSC-00001.mp4 --sub-source logo --logo-file /home/pi/Desktop/logo/plogo-tv.png --logo-position 10 --no-osd`

but it doesn't appear.
but if I click ALT + C and then ALT + F it works...
why?

thanks

pettedemon, 2022-03-15:
Maybe can be helpful, I try to deactivate KMS in raspberry, and the "trick" ALT+c and ALT+f doesn't work

pettedemon, 2022-03-18:
Hi, I install Raspberry Pi OS (Legacy) with desktop, I enable FKMS and all works without any problem on my Raspberry 3b+. So the problem is the FKMS is not possible to enable in Debian version: 11 (bullseye). At now there are no problem with the logo and there are not dropped frames.