RPi-Distro/vlc #62: RaspberryPiOS x86 VLC player menu bar and settings coruption

b1ack0p, 2022-04-14:
Build : Linux  4.19.0-20-686-pae #1 SMP Debian 4.19.235-1 (2022-03-17) i686 GNU/Linux
VLC version: 3.0.12-0+deb10u1

screenshots:

[image]
[image]

XECDesign, 2022-07-06:
A new, bullseye-based version of the .iso was released a few days ago. I'm not able to reproduce the problem on that release. Are you still seeing it?

b1ack0p, 2022-07-16:
> A new, bullseye-based version of the .iso was released a few days ago. I'm not able to reproduce the problem on that release. Are you still seeing it?

hi. thank you for response. i tested RaspiOS x86/bullseye on virtualbox and menu bar on VLC player is visible. seems no more problem with VLC player. I will install it on real machine later.

closed 2022-07-16 by b1ack0p