RPi-Distro/vlc #66: No MMAL Vout on Bullseye

NickCis, 2022-06-24:
It seems that in Raspberry PI OS 11 (Bullseye) the [raspberry pi vlc package of this repository  `3.0.16`](https://github.com/RPi-Distro/vlc/tree/bullseye-rpt) is superseded by the `3.0.17` of the Raspian repository:

```
# apt-cache madison vlc
       vlc | 3.0.17.4-0+deb11u1+rpi1 | http://raspbian.raspberrypi.org/raspbian bullseye/main armhf Packages
       vlc | 3.0.16-1+rpi1+rpt2 | http://archive.raspberrypi.org/debian bullseye/main armhf Packages
```

I usually think that having an updated version is a positive thing, the problem with this is that the Raspian package doesn't have raspberry pi patches in order to provide the `mmal_vout`.

Are there any instructions in order how to bring changes from the upstream repository ([debian](https://salsa.debian.org/multimedia-team/vlc/-/commits/bullseye/) or raspian) in order to have the 3.0.17.4 with raspberry pi patches applied?.

XECDesign, 2022-06-24:
There’s a build that’s waiting for a little bit of internal testing. It should be public some time next week.

NickCis, 2022-06-24:
@XECDesign Thanks for the quick reply, is there a way to access that source code or build?. I don't mind building it from source and testing it out.

Anyway, I will try to apply the mmal patches of this repo to the source of the raspbian (vía the `apt source get`).

Thanks again :)

jc-kynesim, 2022-06-27:
It is worth pointing out here that Bullseye by default runs KMS for display which grabs the HVS output hardware and that prevents mmal_vout from working. If you have your setup loading FKMS (fake)KMS then you do have mmal_vout  i.e. if you have a bullseye that mmal_vout works on currently then it should continue to do so but if it doesn't then upgrading won't make it work. The new version gives you drm_vout which is intended to do much the same job as mmal_vout but has limitations imposed by KMS/DRM so some things like rotation and compositing over X don't work (but it does give you HDR if you like that sort of thing)

KenT2, 2022-06-27:
That's great
are the command line options for drm_vout in the same place that the mmal ones were?

Will compositing over X ever work?

popcornmix, 2022-06-27:
> One other question, will fake KMS always be in Bullseye and any future RPiOS version.

Always in bullseye. But it is deprecated so may be dropped in a future major release update.

NickCis, 2022-06-27:
> The new version gives you drm_vout which is intended to do much the same job as mmal_vout but has limitations imposed by KMS/DRM so some things like rotation and compositing over X don't work (but it does give you HDR if you like that sort of thing)

I'm currently using the mmal output with the fkms overlay. I don't run X server, just a cvlc service that plays a Fullscreen video (I'm using the raspberry pi in order to loop videos connected to a tv screen). I thought the recommended way to do it was with the mmal output, should I switch to the drm+kms? Will this allow me to select hdmi outputs as the mmal does? If I use the kms overlay I won't be able to set resolutions using the `tvservice` command or am I mistaken? In that case, how the video output should be configured?

Thanks for the answers :)

XECDesign, 2022-07-01:
A new build has just gone out. There should be another build in the not too distant future with some bug fixes.

parheliamm, 2022-07-29:
> A new build has just gone out. There should be another build in the not too distant future with some bug fixes.

When can I receive the new release build ?