RPi-Distro/vlc #69: new release 3.0.17.4-0 playback still worse than kodi

parheliamm, 2022-07-30:
The new release didn't has impressive improvements on play back perspective. The new release has same level with the older release on 64bit RPI-OS. The CPU usage is a little bit lower than before, but the playback quality didn't has improvements as well.

On 64 bit OS, the kodi player seems like the one and the only one solution at this moment.

popcornmix, 2022-07-30:
Are you talking about vlc playback in a desktop window, or running fullscreen.
Playback in a desktop window will always be worse than fullscreen.

parheliamm, 2022-07-30:
> Are you talking about vlc playback in a desktop window, or running fullscreen. Playback in a desktop window will always be worse than fullscreen.

I meant to say playback in a desktop window. Is that a known issue? Is there a plan to improve this?
On the other hand, if I want to playback via fullscreen, can I get the same level with kodi by default? Do I need to change the video out parameter for better performance?

parheliamm, 2022-07-31:
Is there any HW limitation from BCM chipset?

popcornmix, 2022-08-01:
> I meant to say playback in a desktop window. Is that a known issue? Is there a plan to improve this?

It's known but there is likely no solution under X. There just are a lot of layers or format conversions and memory copies involved in getting pixels into a window. Things could improve with moves to Wayland/Weston which could properly use hardware overlays for video in a window, but I wouldn't hold your breath.

> On the other hand, if I want to playback via fullscreen, can I get the same level with kodi by default? Do I need to change the video out parameter for better performance?

No, the default parameters will be the ones we consider best. Changing these is likely to make things worse.