RPi-Distro/vlc #71: RPI4 + VLC: drm_vout generic debug: drmu_env_polltask_cb: Timeout

beta-tester, 2022-08-04:
hi,
from time to time i have the VLC issue, that in full screen mode i can not return to normal windowed mode. no double click nor ESC nor other keyboard shortcuts are working anymore.
even the GPIO - Shutdown button isn't working properly.

first i thought it is fixed with VLC update 3.0.17.4-0+deb11u1+rpi1+rpt2, but it isn't.

i started vlc from the command line `vlc -vvv &> vlc.txt` and opened a mp4 file.
after a long while VLC wasn't reacting to any user input.
even worse, the picture freezed but the autio was playing normal.
normally when vlc isn't reacting anymore to user input, i get back control as soon the video reached its end, but i gave up, because the length of that video i tried is about 8 hours.

here the `-vvv` output:
[vlc.xz.gz](https://github.com/RPi-Distro/vlc/files/9261378/vlc.xz.gz)

at line 515595 it starts to see `drm_vout generic debug: drmu_env_polltask_cb: Timeout` in the output.
maybe this is the cause of the issue,
```log
...
[h264_v4l2m2m @ 0x7914da90] capture: Buffer requeue
[h264_v4l2m2m @ 0x7914da90] --- capture VIDIOC_QBUF: index 16, ts=0.110342 count=1
[h264_v4l2m2m @ 0x7914da90] --- output pre VIDIOC_QBUF: index 0, ts=0.110363 count=0
[h264_v4l2m2m @ 0x7914da90] --- output VIDIOC_QBUF: index 0, ts=0.110363 count=1
[786381d8] drm_vout generic debug: drmu_env_polltask_cb: Timeout
[h264_v4l2m2m @ 0x7914da90] capture: Buffer requeue
[h264_v4l2m2m @ 0x7914da90] --- capture VIDIOC_QBUF: index 18, ts=0.110343 count=1
[h264_v4l2m2m @ 0x7914da90] pts_stats_add: decoder: New interval: 16667->16666/1=16666
[h264_v4l2m2m @ 0x7914da90] --- output pre VIDIOC_QBUF: index 0, ts=0.110364 count=0
[h264_v4l2m2m @ 0x7914da90] --- output VIDIOC_QBUF: index 0, ts=0.110364 count=1
...
```

when i pressed the GPIO-shutdown button as last chance i tried, the audio stopped and it looks like parts of the VLC application ends. but the RPi4 did not shut down visually. i still could see the freezed picture of the video. fortunately it looks like the VLC.txt file isn't gone lost.

beta-tester, 2022-08-04:
here my config file

[config.xz.gz](https://github.com/RPi-Distro/vlc/files/9261609/config.xz.gz)

jc-kynesim, 2022-08-04:
There's a new VLC coming soon that fixes (at least some) fullscreen switch bugs. Thanks for the report.

jc-kynesim, 2022-08-04:
The polltask timeout isn't the cause of the issue - all it means is that the display hasn't had any pictures recently.

beta-tester, 2022-08-04:
thank you for the info.
so i have to change the headline of my issue report when the timeout is not the cause.

KoxSosen, 2022-08-06:
Hello there!

I`m experiencing the same exact issue. 

If you need help with anything in terms of testing/providing debug information, please let me know.

beta-tester, 2022-08-09:
i figured out a very reproducable case, where i loose control over VLC in fullscreen mode.

1. open VLC in windowed mode
2. open a video file
3. start the video (still in windowed mode)
4. move the VLC window to the center of the screen (make sure there is a big gap between VLC window and the left top of the desktop screen)
5. move the mouse cursor to the center of the VLC window
6. make a double click in the center of the thc window to switch to fullscreen mode, but do not move the mouse cursor
7. make a right click (in non-fullscreen mode the context menu would appear but it is invisible in fullscreen mode)
8. move the invisible mouse cursor to the left top (in non-fullscreen mode you would leave the context menu area, in fullscreen mode you do not see this)
9. move the invisible mouse cursor further to the left top (to leave the invisible/overlayed vlc window)
10. mke a left click
11. now i have no control over VLC anymore.
    - no ESC, no double click, no keyboard shortcut are working anymore
    - GPIO-shutdown push-button is working (previous i mention that is isn't, but at my last tests it was working always)

i don't know if all steps are required or if you would get back control if you find the vlc window area with the invisible mouse cursor.

looks to me, that part of the issue is related where the mouse cursor "sits" in fullscreen mode "relative" to the non-fullscreen position to the VLC window position.

and i am using a keyboard touch pad combo (logitech K400+) and i guess that i make sometimes a right click by accident while vlc is in fullscreen mode.

beta-tester, 2022-08-09:
here a log file of `vlc -vv &> >(tee vlc.txt)`:
[vlc.gz](https://github.com/RPi-Distro/vlc/files/9288129/vlc.gz)

beta-tester, 2022-08-09:
@jc-kynesim ,
i just received and installed the announced vlc update `3.0.17.4-0+deb11u1+rpi1+rpt3`. i rebooted my RPi4,
but i can still reproduce the issue to 100% in the same way as described before.

jc-kynesim, 2022-08-16:
Yup - not surprised I'm afraid. Fullscreen is a bit of a kludge in order gain performance.  It simply grabs the screen away from X in order to use DRM directly - the underlying X windows are still there and keyboard control etc if still handled by X, so if you use the mouse on the invisible X desktop it still does stuff. There isn't an easy fix - I wish there was.

beta-tester, 2022-08-17:
> It simply grabs the screen away from X in order to use DRM directly - the underlying X windows are still there and keyboard control etc if still handled by X, so if you use the mouse on the invisible X desktop it still does stuff. There isn't an easy fix - I wish there was.

for fullscreen mode, wouldn't it be possible to maximize the vlc 'X' window and make it without "decorations" (frame/bordertitle/menu-area) so the vlc 'X' window takes over the full X screen area of the desctop, so the mouse isn't able to leave the vlc 'X' window area anymore - in hope i does not lose the mouse and keyboard control...
and then put the VLC/DRM overlay on top ot that?

hrbekma, 2022-09-23:
i have the same prob too. Still waiting for fix comming with update

atimmernans, 2022-12-12:
> for fullscreen mode, wouldn't it be possible to maximize the vlc 'X' window and make it without "decorations" (frame/bordertitle/menu-area) and make it topmost so the vlc 'X' window takes over the full X screen area of the desctop, so the mouse isn't able to leave the vlc 'X' window area anymore - in hope i does not lose the mouse and keyboard control... and then put the VLC/DRM overlay on top ot that?
> 
> ...this is an ugly hack, but the issue is (kind of) solved.

If you can put such an X window covering the whole screen, no need to hack anything, just use vlc in that window normally. This will avoid all the other side effects that the current full screen mode seems to suffer from.

atimmernans, 2022-12-12:
Note that on mi PI4, MMAL or DRM output suffer from this problem but OpenGL and OpenGL2 works fine (if you disregard tearing issues).