RPi-Distro/vlc #72: VLC not taking snapshots

Botspot, 2022-09-11:
Most of my prior issues with VLC have been solved with a reboot. But not this one. Today I tried to take a snapshot from a video using VLC and I could not find the saved picture anywhere. It turns out VLC did not save a picture at all.

The result appears to be the same no matter the video codec or file extension.

Full output from running `vlc`:
```
VLC media player 3.0.17.4 Vetinari (revision 3.0.13-8-g41878ff4f2)
[01dd4b98] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
[01e34fa0] main playlist: playlist is empty
[ee004880] gles2 gl: Initialized libplacebo v2.72.0 (API v72)
[ee006780] mmal_xsplitter vout display error: Failed to open Xsplitter:mmal_vout module
[01e9b7f0] avcodec decoder: Using DRM Video Accel for hardware decoding
[01e531f0] alsa audio output error: cannot estimate delay: Input/output error

#Right here I clicked Video -> Take Snapshot.

[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed179090] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed179090] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed179090] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed179090] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed179090] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed179090] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed167050] chain filter error: Too high level of recursion (3)
[f03c7f90] main filter error: Failed to create video converter
[ed179090] main filter error: Failed to create video converter
[01ec0c30] main video output error: Failed to convert image for snapshot
```

The `mmal_xsplitter vout display error: Failed to open Xsplitter:mmal_vout module` error may indicate a problem on my end if others cannot reproduce it, but I'm not sure what I would have done to cause it. The Full KMS GL driver is enabled in raspi-config, and on this install I've tried to avoid making any gpu- or video-related changes to the system.

6by9, 2022-10-28:
FWIW Duplicated on forum thread https://forums.raspberrypi.com/viewtopic.php?t=340860

jc-kynesim, 2022-12-16:
This should be fixed

Botspot, 2022-12-16:
Yes it is. Also I noticed that going into fullscreen is a lot faster now too.

Great work everyone.

closed 2022-12-16 by Botspot