RPi-Distro/vlc #73: Highly unreliable with freezes when using Pi 3B and Debian 11

zocker-160, 2022-09-20:
I hope someone can help me with this, because I am starting to get a bit desperate with this problem.

I am running an application for displaying a 2x2 or 3x3 grid displaying CCTV cameras which uses libvlc using `python-vlc`. When running on Pi 3 hardware and Debian 10, everything works, but since moved to Debian 11, I am struggling with major issues.

VLC is initialized like this
```py
self.vlcInstance = vlc.Instance(
    "--vout=mmal_vout",
    "--no-xlib"
)
```
Pi3 is connected to a 720p TV and I have allocated 256MB of memory to the GPU.
Everything either freezes just after a few seconds with `sudo killall` being the only option to close it or I get a segmentation fault, here are my logs:

no KMS, legacy
```
libEGL warning: DRI2: failed to authenticate
Created new TCP socket 43 for connection
Created new TCP socket 44 for connection
Created new TCP socket 45 for connection
Created new TCP socket 46 for connection
[6a30d770] mmal_codec decoder: VCSM init succeeded: Legacy
[01776af0] mmal_codec decoder: VCSM init succeeded: Legacy
[6a10df80] mmal_codec decoder: VCSM init succeeded: Legacy
[6a30d770] mmal_codec decoder: VCSM init succeeded: Legacy
[6a10df80] mmal_codec decoder: VCSM init succeeded: Legacy
[01776af0] mmal_codec decoder: VCSM init succeeded: Legacy
[75051a08] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
[017b9888] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
[750564e0] mmal_codec decoder: VCSM init succeeded: Legacy
[67d44b68] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
[750564e0] mmal_codec decoder: VCSM init succeeded: Legacy
[6a20b130] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
/usr/bin/dashboard: line 3:  1166 Segmentation fault      python3 /opt/tvservice.py --nomouse "$@"
```

no KMS, legacy **without** `--no-xlib`
```
libEGL warning: DRI2: failed to authenticate
Created new TCP socket 45 for connection
Created new TCP socket 46 for connection
Created new TCP socket 43 for connection
Created new TCP socket 44 for connection
[0284a2f0] mmal_codec decoder: VCSM init succeeded: Legacy
[6a20d7f8] mmal_codec decoder: VCSM init succeeded: Legacy
[6a00df80] mmal_codec decoder: VCSM init succeeded: Legacy
[0284a2f0] mmal_codec decoder: VCSM init succeeded: Legacy
[6a00df80] mmal_codec decoder: VCSM init succeeded: Legacy
[6a20d7f8] mmal_codec decoder: VCSM init succeeded: Legacy
[74f13600] mmal_codec decoder: VCSM init succeeded: Legacy
[74f13600] mmal_codec decoder: VCSM init succeeded: Legacy
libEGL warning: DRI2: failed to authenticate
libEGL warning: DRI2: failed to authenticate
libEGL warning: DRI2: failed to authenticate
[67c463c0] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
[592c0ae8] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
[6835bc88] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
libEGL warning: DRI2: failed to authenticate
[6a109e28] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
[6a11bc98] xcb_x11 generic error: shared memory attachment error: Cannot allocate memory
```

KMS
```
Created new TCP socket 47 for connection
Created new TCP socket 48 for connection
Created new TCP socket 49 for connection
Created new TCP socket 50 for connection
[67519a30] mmal_codec decoder: VCSM init succeeded: CMA
[5ec14338] mmal_codec decoder: VCSM init succeeded: CMA
[68f025b0] mmal_codec decoder: VCSM init succeeded: CMA
[5ec14338] mmal_codec decoder: VCSM init succeeded: CMA
[67519a30] mmal_codec decoder: VCSM init succeeded: CMA
[68f025b0] mmal_codec decoder: VCSM init succeeded: CMA
[67b12060] mmal_codec decoder: VCSM init succeeded: CMA
[68f08d38] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
[6916dc00] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
[68f08d38] mmal_xsplitter vout display error: Failed to open Xsplitter:mmal_vout module
mmal: mmal_vc_component_create: failed to create component 'vc.ril.hvs' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.hvs' (1)
[6916dc00] mmal_xsplitter vout display error: Failed to open Xsplitter:mmal_vout module
[6a349838] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
mmal: mmal_vc_component_create: failed to create component 'vc.ril.hvs' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.hvs' (1)
[6a349838] mmal_xsplitter vout display error: Failed to open Xsplitter:mmal_vout module
mmal: mmal_vc_component_create: failed to create component 'vc.ril.hvs' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.hvs' (1)
mmal: mmal_vc_component_create: failed to create component 'vc.ril.hvs' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.hvs' (1)
mmal: mmal_vc_component_create: failed to create component 'vc.ril.hvs' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.hvs' (1)
mmal: mmal_vc_component_create: failed to create component 'vc.ril.hvs' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.hvs' (1)
[67b12060] mmal_codec decoder: VCSM init succeeded: CMA
[009b64b0] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
[009b64b0] mmal_xsplitter vout display error: Failed to open Xsplitter:mmal_vout module
mmal: mmal_vc_component_create: failed to create component 'vc.ril.hvs' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.hvs' (1)
mmal: mmal_vc_component_create: failed to create component 'vc.ril.hvs' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.hvs' (1)
[67519a30] main decoder error: buffer deadlock prevented
mmap() failed: Cannot allocate memory
Failed to create permanent mapping for memfd region with ID = 4091698473
Failed to regester memfd mempool. Reason: could not attach memfd SHM ID to pipe
mmap() failed: Cannot allocate memory
Failed to create permanent mapping for memfd region with ID = 2433974757
mmap() failed: Cannot allocate memory
Failed to create permanent mapping for memfd region with ID = 2823137409
Ignoring received block reference with non-registered memfd ID = 2823137409
```

Raspberry Pi 3B
Debian 11.5
python-vlc 3.0.16120
vlc 3.0.17.4-0+deb11u1+rpi1+rpt3

Any help would be highly appreciated!

popcornmix, 2022-09-20:
Can you just confirm you are running RPiOS bullseye? (I believe you are based on the kernel build number).
Note: That there are [actual debian images](https://raspi.debian.net/tested-images/) which are quite different and don't include this repo's version of vlc.

zocker-160, 2022-09-20:
I installed the system using PI Imager, here is my os-release
```
$ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
```
$ apt-cache policy vlc
vlc:
  Installed: 3.0.17.4-0+deb11u1+rpi1+rpt3
  Candidate: 3.0.17.4-0+deb11u1+rpi1+rpt3
  Version table:
 *** 3.0.17.4-0+deb11u1+rpi1+rpt3 900
        900 http://archive.raspberrypi.org/debian bullseye/main armhf Packages
        100 /var/lib/dpkg/status
     3.0.17.4-0+deb11u1+rpi1 500
        500 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf Packages
```