RPi-Distro/vlc #78: RTSP-Stream cannot be opened anymore

Blackbox-git, 2022-12-10:
Hello, since the last VLC update I can not open the RTSP stream of my network camera. I always get the following error:

VLC is unable to open the MRL 'rtsp://Username:password@192.168.111.5:554/11'. Check the log for details.

[image]

What can I do about it? How can i get to a old version?

XECDesign, 2022-12-12:
What's the full version number (the output of `apt policy vlc`)?

Blackbox-git, 2022-12-12:
Hey in the following I send you the requested information. I updated VLC again today, because a new version was available.

### Version of the Release Date September 22nd 2022 -> worked!

```
pi@DisplayPi:~ $ apt policy vlc
vlc:
  Installed: 3.0.17.4-0+deb11u1+rpi1+rpt3
  Candidate: 3.0.18-0+rpt1+deb11u1
  Version table:
     3.0.18-0+rpt1+deb11u1 500
        500 http://archive.raspberrypi.org/debian bullseye/main armhf Packages
     3.0.18-0+deb11u1+rpi1 500
        500 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf Packages
 *** 3.0.17.4-0+deb11u1+rpi1+rpt3 100
        100 /var/lib/dpkg/status 
```

### Upgraded Version 2022-12-09 -> Described Bug

```
pi@DisplayPi:~ $ apt policy vlc
vlc:
  Installed: 3.0.18-0+deb11u1+rpi1
  Candidate: 3.0.18-0+deb11u1+rpi1
  Version table:
 *** 3.0.18-0+deb11u1+rpi1 500
        500 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf Packages
        100 /var/lib/dpkg/status
     3.0.17.4-0+deb11u1+rpi1+rpt4 500
        500 http://archive.raspberrypi.org/debian bullseye/main armhf Packages
```

### Upgraded Version 2022-12-12 ->worked!

```
pi@DisplayPi:~ $ apt policy vlc
vlc:
  Installed: 3.0.18-0+rpt1+deb11u1
  Candidate: 3.0.18-0+rpt1+deb11u1
  Version table:
 *** 3.0.18-0+rpt1+deb11u1 500
        500 http://archive.raspberrypi.org/debian bullseye/main armhf Packages
        100 /var/lib/dpkg/status
     3.0.18-0+deb11u1+rpi1 500
        500 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf Packages
```

As you could see from the headlines, the now recent update of VLC is working normally again. I can not explain what I do wrong on Friday (2022-12-09). I used the following commands to upgrade like today:

```
sudo apt update && sudo -y full-upgrade
```

**Can you explain what happend?**

PS: Because you hear it far too little. Thank you for your great work!

XECDesign, 2022-12-14:
> Can you explain what happend?

Sure. Since there was a [security issue](https://security-tracker.debian.org/tracker/CVE-2022-41325), Debian released a new version, which filtered through to Raspbian.

Because we don't want to end up in a situation where there's major security issue, we generally version our packages such that Debian's updates take precedence. The downside is that if people update to that version, they lose our changes until I get a chance to rebase our changes and release a new package.

> PS: Because you hear it far too little. Thank you for your great work!

Thanks! Much appreciated.

Blackbox-git, 2022-12-14:
Thank you for the explanation. Then I will look at the policy of the package before the next update.

closed 2022-12-14 by Blackbox-git