RPi-Distro/vlc #79: VLC some subtitles are not displayed in full screen mode

atimmernans, 2022-12-12:
I was using vlc .0.17.4-0+deb11u1+rpi1+rpt3 and just now I am using 3.0.18-0+rpt1+deb11u1.
I often watch Japanese anime and at some points voice subtitles are not shown in full screen mode.
At first I thought that it was random, but I finally noticed that it occurred when translations for on-screen Japanese texts (building names, phone texts, ...) were displayed.
In windowed mode both these translations and voices subtitles are displayed.
A quick check shows that there is a single ssa subtile track embedded in the mkv file and disabling it disables both voice subtitles and textual translations.

XECDesign, 2022-12-12:
Did this only break after the update to .18 or was the issue present on .17 as well?

atimmernans, 2022-12-12:
Was present on both versions.

popcornmix, 2022-12-12:
Are you running from X? If so can you test from console.
Switch to different VT (e.g. ctrl-alt-F1) then launch VLC from command line.

My understanding is that when vlc is launched from X, in full screen mode, it can only acquire two planes (a video plane and an OSD plane). And I believe in some circumstances subtitles may require additional planes. (@jc-kynesim can confirm exact details).

Launching from console doesn't have that limitation and can use many planes.

jc-kynesim, 2022-12-12:
What he just said

popcornmix, 2022-12-12:
No need to rename file. E.g. if file is called:
`The Hobbit.mkv`
You need to run:
`vlc "The Hobbit.mkv"`
or
`vlc The\ Hobbit.mkv`
Either use quotes around name, or escape the space with `\`. Using tab as filename auto-complete can also help here.

I think the behaviour is "as designed" (i.e. there is no perfect solution given the constraints of X running on DRM impose).

If in VLC, you go to tools/preferences/video/output, you can choose a different output method from "DRM vout plugin" (or "Automatic" which will chose that) and you won't have this issue although performance will be worse.
"OpenGL video output" should work (I'm not sure if any alternatives are better).

It would also be possible to create a vlc wrapper script, that does a `sudo chvt` around launching the real vlc, so you get the performance and all subtitles, although without the usual vlc gui. You could associate this script with your video file extensions so it is run automatically by double clicking.