RPi-Distro/vlc #82: Green band in full screen of MPEG-2 TS

atimmernans, 2022-12-16:
Using MMAL or DRM in full screen, a green band appears on some MPEG-2 TS files.

Checking the media info it has a picture size of 1920x1080 but its buffer is 1920x1088.
I suspect thus that the wrong size is used by the ouptut module and the green band corresponds to these extra 8 lines.

jc-kynesim, 2022-12-16:
Almost certainly (the video is going to be 1920x1088 cropped to 1920x1080 with the remaining lines 0). The question is where is the cropping going missing. Can you point me at a sample file (you say some so I assume it isn't all)? Is this with h/w or s/w decode (Pi3 or earlier with license is likely h/w otherwise s/w)?

atimmernans, 2022-12-16:
The files are slightly large (23GB) so I check around and found this one from the MPlayer test samples which has the same problem.

Note that this does not occur with other output modules nor in windowed mode on my Pi4.

jc-kynesim, 2022-12-16:
Will be fixed in the next release - I've just fixed the problem as part of a different issue

atimmernans, 2023-02-09:
I tested the new release and the problem is indeed solved

closed 2023-02-09 by atimmernans