RPi-Distro/vlc #83: Seamless video loop

eduardomcp, 2022-12-17:
Hi,

With omxplayer we are able to have a video on loop using h264 format (silent, not mp4 with h264 + audio), with no stutter or showing any blank screen, or CLI, on each repetition. (I need to only use scripts and CLI no X). 

I'm not able to do so with VLC via CLI and since omxplayer has been deprecated, I had to return to buster and am stuck in an old OS version just due to this. Can someone shed some light on how to repeat a video single file seamlessly via CLI (preferably with audio) via script, on raspbian bullseye? 
 - is VLC not the solution for this? ( vlc loop and repeat flags aren't cutting it) 
 - maybe some special codec that I haven't tried yet?

gllmAR, 2023-01-03:
for seemless loop, you can try the input-repeat flag which is better than loop flag.

```
cvlc --input-repeat=99999999999 --fullscreen media.mp4
```

Sandvoxel, 2023-04-14:
I am running into this exact issue with vlc and was wondering if you found anything that works better the black screen when switching videos is not something I can have in the project I am working on.

maxroseGIT, 2024-09-01:
Interesting question. I was searching this and found this issue.

Has there been any progress with this question? Or any ideas how to solve it.

Thanks for any info here.

shatteredsword, 2025-07-18:
with omxplayer I was able to use a very simple script that allowed the hot-swapping of video files as a kiosk for our organizaion, using a samba share to access the directory. this allowed for endless looping of any video file dropped in the directory. This seamless functionality just doesn't exist in vlc. 
````
while true
do
        omxplayer -o hdmi /home/pi/pi-kiosk/kiosk/* 2>> /home/pi/pi-kiosk/error.log 1> /dev/null
done
````