RPi-Distro/vlc #84: VLC version  3.0.18-0+rpt1+deb11u1 - mmal issue

nightside1, 2022-12-18:
Hi, 
I've a rasperry Pi4 with latest bullseye OS.
I've updated VLC from version 3.0.17-0 to the new one, but I've issue with --mmal-vout-window position. 
There is a way to install the previous release? 
If yes, can someone help me? 
Thanks

jc-kynesim, 2022-12-18:
OK - that is odd - that code won't have changed the slightest between those two releases. What exactly does "I've issue with" entail? Can you post the output of whatever command line you are using with `-vv` added please? (Please cut it once it reaches actual playback  - 20 billion lines of "see my picture", "I've got another picture" add no value! Setup is what I'm after!)

nightside1, 2022-12-18:
Hi jc-kynesim,
with the previous version, I was able to split my TV screen into 9 or 6 different RTSP sources.
With this version mmal-vout-window is working only for a few sources, the other ones don't start.
In addition, I'm not able to play RTSP stream in full screen if dtoverlay=vc4-kms-v3d is commented in config.txt
If vc4-kms-v3d is enabled, mmal-vout-windows doesn't work (this issue was present also in version 3.0.17

Here below is the command that I've used without any issue with version 3.0.17

**For Full-screen stream (1 source):** 
`cvlc -vv --loop --no-video-title 'rtsp://x.x.x.x:xxxx/videoMain'`

**_LOGS with dtoverlay=vc4-kms-v3d commented in config.txt_**
[deinterlace_v4l2m2m @ 0x565760] VIDIOC_REQBUFS failed: Cannot allocate memory
[Parsed_deinterlace_v4l2m2m_0 @ 0x5656c0] Failed to allocate destination buffers
[00539800] drm_av_deinterlace filter error: Failed to get frame: Cannot allocate memory
[00539800] drm_av_deinterlace filter debug: >>> drmp_av_deinterlace: (nil)
[00539800] drm_av_deinterlace filter debug: <<< drmp_av_deinterlace
[deinterlace_v4l2m2m @ 0x565760] VIDIOC_REQBUFS failed: Cannot allocate memory
[Parsed_deinterlace_v4l2m2m_0 @ 0x5656c0] Failed to allocate destination buffers
[00539800] drm_av_deinterlace filter error: Failed to get frame: Cannot allocate memory
[00539800] drm_av_deinterlace filter debug: >>> drmp_av_deinterlace: (nil)
[00539800] drm_av_deinterlace filter debug: <<< drmp_av_deinterlace
[deinterlace_v4l2m2m @ 0x565760] VIDIOC_REQBUFS failed: Cannot allocate memory
[Parsed_deinterlace_v4l2m2m_0 @ 0x5656c0] Failed to allocate destination buffers
[00539800] drm_av_deinterlace filter error: Failed to get frame: Cannot allocate memory
[00539800] drm_av_deinterlace filter debug: >>> drmp_av_deinterlace: (nil)
[00539800] drm_av_deinterlace filter debug: <<< drmp_av_deinterlace

**For multiview:** 
```
vlc -vv --mmal-vout-transparent --mmal-vout-window 1280x720+00+00 --aspect-ratio 16:9 --rtsp-tcp "rtmp://x.x.x.x/bcs/channel0_ext.bcs?channel=0&stream=2"
vlc -vv --mmal-vout-transparent --mmal-vout-window 640x360+00+720 --aspect-ratio 16:9 --rtsp-tcp "rtsp://x.x.x.x:xxxx/videoMain"
vlc -vv --mmal-vout-transparent --mmal-vout-window 640x360+640+720 --aspect-ratio 16:9 --rtsp-tcp "rtsp://x.x.x.x:xxxx/videoSub"
vlc -vv --mmal-vout-transparent --mmal-vout-window 640x360+1280+720 --aspect-ratio 16:9 --rtsp-tcp "rtsp://x.x.x.x:xxxx/videoMain"
vlc -vv --mmal-vout-transparent --mmal-vout-window 640x360+1280+360 --aspect-ratio 16:9 --rtsp-tcp "rtsp://x.x.x.x:xxxx/videoMain"
vlc -vv --mmal-vout-transparent --mmal-vout-window 640x360+1280+00 --aspect-ratio 16:9 --rtsp-tcp "rtsp://x.x.x.x:xxxx/videoMain"
```

**_LOGS with dtoverlay=vc4-kms-v3d commented in config.txt_**
as already explained, some sources are properly working, but others no, here below the logs for not working sources:

  2 candidates
[b0809800] freetype spu text debug: Building font databases.
[b0809800] freetype spu text debug: Took -10508 microseconds
[b0809800] main spu text debug: using text renderer module "freetype"
[b0801130] main generic debug: looking for hw decoder module matching "any": 1 c                                                                                                                               andidates
[b0801130] drm_avcodec generic debug: Open: pix_fmt=181
[b0801130] main generic debug: using hw decoder module "drm_avcodec"
[b0b0a5d0] avcodec decoder: Using DRM Video Accel for hardware decoding
[h264_v4l2m2m @ 0xb0801500] Using device /dev/video10
[h264_v4l2m2m @ 0xb0801500] driver 'bcm2835-codec' on card 'bcm2835-codec-decode                                                                                                                               ' in mplane mode
[h264_v4l2m2m @ 0xb0801500] requesting formats: output=H264 capture=YU12
[b0b0a5d0] avcodec decoder debug: codec (h264_v4l2m2m) started
[b0b0a5d0] main decoder debug: using video decoder module "avcodec"
[h264_v4l2m2m @ 0xb0801500] capture VIDIOC_REQBUFS failed: Cannot allocate memor                                                                                                                               y
[h264_v4l2m2m @ 0xb0801500] can't request capture buffers
[h264_v4l2m2m @ 0xb0801500] capture set status 1074026002 (ON) failed: err=22
[00cc2cc8] main vout display debug: auto hiding mouse cursor
[h264_v4l2m2m @ 0xb0801500] V4L2 capture poll unexpected timeout: events=0x106
[h264_v4l2m2m @ 0xb0801500] Poll thinks src Q has space; none found
[h264_v4l2m2m @ 0xb0801500] V4L2 capture poll unexpected timeout: events=0x106
[h264_v4l2m2m @ 0xb0801500] Poll thinks src Q has space; none found
[00ccfe90] main audio output debug: resampling stopped (drift: 4610 us)
[h264_v4l2m2m @ 0xb0801500] V4L2 capture poll unexpected timeout: events=0x106
[h264_v4l2m2m @ 0xb0801500] Poll thinks src Q has space; none found
[h264_v4l2m2m @ 0xb0801500] V4L2 capture poll unexpected timeout: events=0x106
[h264_v4l2m2m @ 0xb0801500] Poll thinks src Q has space; none found
[h264_v4l2m2m @ 0xb0801500] V4L2 capture poll unexpected timeout: events=0x106
[h264_v4l2m2m @ 0xb0801500] Poll thinks src Q has space; none found
[h264_v4l2m2m @ 0xb0801500] V4L2 capture poll unexpected timeout: events=0x106
[h264_v4l2m2m @ 0xb0801500] Poll thinks src Q has space; none found
[h264_v4l2m2m @ 0xb0801500] V4L2 capture poll unexpected timeout: events=0x106
[h264_v4l2m2m @ 0xb0801500] Poll thinks src Q has space; none found

Please let me know if additional tests are needed.

Regards,