RPi-Distro/vlc #87: Runs Video when typed in CLI or run script manually with ./ but will not work with systemd

samthejam12, 2023-02-03:
I am trying to create a kiosk that has a video loop. This video gets cross checked with a md5 hash from a website so it needs to be launched from within a script after it does the hash. 

I can type "" in the CLI and it will run the video fine. 
I can run a test script manually with ./testsrcipt.sh and it runs fine. 

However when I run it with systemd, I just cannot get it working. Here are the error messages:

kiosk.service - startme
     Loaded: loaded (/etc/systemd/system/kiosk.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2023-02-03 14:00:46 GMT; 3min 54s ago
   Main PID: 404 (bash)
      Tasks: 2 (limit: 4915)
        CPU: 30.111s
     CGroup: /system.slice/kiosk.service
             ├─ 404 /bin/bash /home/pi/scripts/runvideo.sh
             └─3500 /usr/bin/vlc -I dummy --fullscreen --no-osd --loop /home/pi/pi.mp4

Feb 03 14:04:40 pi bash[3486]: [158B blob data]
Feb 03 14:04:40 pi bash[404]: 55b4ffce4bfd00336eaec019d25ff218
Feb 03 14:04:40 pi bash[404]: Up to date
Feb 03 14:04:40 pi bash[3500]: [01365530] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Feb 03 14:04:40 pi bash[3500]: [0134ae60] dbus interface error: Failed to connect to the D-Bus session daemon: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Feb 03 14:04:40 pi bash[3500]: [0134ae60] main interface error: no suitable interface module
Feb 03 14:04:40 pi bash[3500]: [012e8b98] main libvlc error: interface "dbus,none" initialization failed
Feb 03 14:04:40 pi bash[3500]: [0135f730] main interface error: no suitable interface module
Feb 03 14:04:40 pi bash[3500]: [012e8b98] main libvlc error: interface "globalhotkeys,none" initialization failed
Feb 03 14:04:40 pi bash[3500]: [0135f730] dummy interface: using the dummy interface module...
Feb 03 14:04:40 pi bash[3500]: error: XDG_RUNTIME_DIR not set in the environment.
Feb 03 14:04:40 pi bash[3500]: error: XDG_RUNTIME_DIR not set in the environment.
Feb 03 14:04:40 pi bash[3500]: [b3503188] gles2 generic error: parent window not available
Feb 03 14:04:40 pi bash[3500]: [b3507e90] mmal_xsplitter vout display error: Failed to open Xsplitter:opengles2 module
Feb 03 14:04:40 pi bash[3500]: [b3503188] xcb generic error: window not available
Feb 03 14:04:40 pi bash[3500]: [b3507e90] mmal_xsplitter vout display error: Failed to open Xsplitter:xcb_x11 module
Feb 03 14:04:40 pi bash[3500]: [b3507e90] mmal_xsplitter vout display error: Failed to open Xsplitter:mmal_vout module
Feb 03 14:04:40 pi bash[3500]: [b3507e90] mmal_xsplitter vout display: Try drm
Feb 03 14:04:40 pi bash[3500]: [b3503188] drm_vout generic: <<< OpenDrmVout: Fmt=DPV0
Feb 03 14:04:40 pi bash[3500]: [b3503188] drm_vout generic error: Failed to get xlease
Feb 03 14:04:40 pi bash[3500]: [b3507e90] mmal_xsplitter vout display error: Failed to open Xsplitter:drm_vout module
Feb 03 14:04:40 pi bash[3500]: [b3507e90] mmal_xsplitter vout display: Drm no go
Feb 03 14:04:40 pi bash[3500]: [b3507e90] mmal_xsplitter vout display: No valid output found for vout (DPV0/DPV0)
Feb 03 14:04:40 pi bash[3500]: [b3507e90] gles2 vout display error: parent window not available
Feb 03 14:04:40 pi bash[3500]: [b3507e90] gl vout display error: parent window not available
Feb 03 14:04:40 pi bash[3500]: [b3507e90] xcb vout display error: window not available
Feb 03 14:04:40 pi bash[3500]: [b3507e90] drm_vout vout display: <<< OpenDrmVout: Fmt=DPV0
Feb 03 14:04:40 pi bash[3500]: [b3507e90] drm_vout vout display error: Failed to get xlease
Feb 03 14:04:40 pi bash[3500]: Error opening terminal: unknown.

jc-kynesim, 2023-02-03:
Not really sure, my experience with systemd is quite limited. I'd guess either the user it is trying to run as doesn't have appropriate permissions (member of video group I think) or some important environment variable isn't set.

cmatheson, 2024-01-25:
I ran into this same issue and can confirm that adding the user to the `video` group solved the problem.