RPi-Distro/vlc #90: No proper video playback is possible in fullscreen

sudo-splinter-cell, 2023-10-14:
RPI 4B. 4GB. Currently on Bullseye RPIOS (with latest updates). Compositor is disabled. GPU ram allocation is 128mb.

VLC is latest version. Video playback "in fullscreen" without any issue is "impossible". I have tried different output modules. When auto is selected, returning from fullscreen crashes VLC and the video playback position is forgotten. You need to rerun VLC and find where you have left off manually. When OpenGL output module is selected, you have "screen tearing" problem in fullscreen. X11 Video output (XCB) resolves this issue but it can't display subtitles. X Video Output (XCB) resolves both issues, however, after like 20 seconds of video playback, VLC crashes the "whole" system and RPI reboots itself.

None of the other output modules can actually display any video.

So in summary, currently, there is no way to enjoy a video in full screen under Raspberry Pi (if you need subtitles). The only acceptable playback option seems to be the OpenGL module which has the screen tearing issue. Which is quite lame to be honest.

I refuse to upgrade the whole system to Bookworm just to be able to enjoy a video in fullscreen. Bookworm has a lot of issues to be solved still. So any bugfix to this issue would greatly be appreciated.
Thanks.

jon709, 2023-10-15:
Generally vlc doesn't seem usable full-screen currently on a my pi4 with the latest (x64/bullseye) raspberry pi os and I think that's just where we are at present. Using "OpenGL for Embedded Systems 2" as the output module seems the most reliable, provided you stick to a screen resolution no more than 1920x1080, but you will get tearing from time-to-time.

With other output modules you can lose control in full-screen as the keyboard and mouse button inputs seem to get redirected into the ether somewhere (the trick is Ctrl+Alt+F1 to a prompt, use ps to find the vlc process and kill it, then Ctrl+Alt+F7 back to the Gui).

My only suggestion is to try kodi, as in my limited testing that seemed to work okay, so at least the hardware is fast enough.

Oh and you are correct, don't go above 128MB of graphics RAM on a pi4 or it will go stop booting (which I fixed by editing the config file with another operating system running from another card - you see two fs when connecting the problem card, bootfs and rootfs, edit config file on bootfs to change last line (for graphics memory) to 128). (I suspect you know this and am just mentioning for others that come here.)

Also I gather you shouldn't enable vnc for vlc to run windowed, I was told the reason is enabling vnc disables mutter (the two aren't compatible) and mutter is needed to avoid tearing within windows.

jc-kynesim, 2023-10-15:
If you want good fullscreen playback under Bookworm - either boot to the command line or swap out of Wayland to a non-Wayland console (e.g. with Ctrl-Alt-F2) and run nvlc (curses based) - that will get you playback via DRM which works well (and incidentally gets you HDR etc.).  The issue is that X allows the user to lease its DRM handles and do this automatically, but Wayland doesn't (currently).

sudo-splinter-cell, 2023-10-15:
@jc-kynesim

I am on "bullseye". Bookworm with its wayland backend probably solves this issue, however, i don't want to upgrade yet, as it has a lot of issues, as it is new.

@jon709

It is currently possible to have "tear free" fullscreen video playback with osd in VLC under Bullseye. You just select X11 Video output (XCB) as your output module. The only issue is that using this output module, you can not have subtitles. And i depend on subtitles.

sudo-splinter-cell, 2023-10-15:
> Does XCB work, I think I've tried all the modes and don't remember any reboots (could be content related of course) - I thought that was one of several choices that causes you to lose Keyboard and Mouse input after a while when full-screen (allegedly due to some hidden window grabbing the button/key inputs and sending them to nowhere).

There are two XCB output modes (X11 video and X video). X11 works. Shows OSD and with no screen tearing. The only caveat is that it can not show subtitles. The other xcb output mode has a huge flaw. After like 10 something seconds of video playback, the RPI just reboots itself. Probably a memory issue with this output module. Maybe i should post the logs. This one (X video output) actually is the best one as it shows osd and subtitles and has no screen tearing issue. But like i said, it just reboots the whole system :)

jon709, 2023-10-27:
I tried it in Bookworm as it seemed okay except the bounding box for the full-screen controls is wrong, so the controls are in the middle of the screen and if you dock them they stay at the same left/up position but shoot off the right edge of the screen (1080p). If that gets fixed I might use it again.

In the meantime I've given up and am using a NUC-alike, which is unsurprisingly on a level or two up for this stuff (£199 for a N100, 16GB DDR5, 521GB PCIe NVMe SSD, WiFi6, Dual 2.5G Ethernet, Bluetooth 5.2, Dual HDMI 2.0 plus a USB-C 3.2 gen 2 port that can output 4k/60, three USB-A 3.2 gen 2 ports @ 10Gbps, Win 11 Pro... VLC runs great at 4k full-screen, the controls are even at the bottom ;-)

popcornmix, 2023-10-27:
> PS3: Kodi was suggested. Kodi is fine but it takes over the whole system and you can not use your desktop for anything else while it is running. You can't minimize it. So it is not a viable alternative.

On bookworm, there are two options for launching kodi: "Kodi" and "Kodi Fullscreen".
The non-fullscreen one runs in a window.

sudo-splinter-cell, 2023-10-27:
> I tried it in Bookworm as it seemed okay except the bounding box for the full-screen controls is wrong, so the controls are in the middle of the screen and if you dock them they stay at the same [...]

It is an unfortunate fact of life that X86 will always be the stable option when it comes to running software. If only they could put an x86 chip on a RPI, now that thing would sell like ice cream.

sudo-splinter-cell, 2023-12-26:
I think it is time to close this one as the new Pi and the boorkworm has no issues anymore.

closed 2023-12-26 by sudo-splinter-cell

jon709, 2023-12-26:
Are you saying it works on the Pi5, as it isn't okay on my Pi4 and Bookworm?

sudo-splinter-cell, 2023-12-26:
> Are you saying it works on the Pi5, as it isn't okay on my Pi4 and Bookworm?

Well, i have been using only my Pi-5 since it arrived and i have been using it with KDE Plasma (which wasn't really viable on Pi-4) and i am not having any issues with video playback anymore with VLC. "kwin" (kde's screen compositor) does remove screen tearing in full screen even when i am on X11 (and i intend to stay on x11 because of all the issues we have with wayland). So yeah, there may still be issues with RPI-4, especially with the new bookworm release, but most of these issues are wayland related and i don't think VLC or the RPIOS is to blame here. I have realized that it is not just a video playback issue when it comes to "desktop use" with different distros including the RPIOS. It is a "wayland vs x11" issue. And that probably requires a separate bug report or even reportS.

jon709, 2023-12-26:
Okay, although I'd run it for a while full-screen and see if you lose keyboard and mouse control at some point....