RPi-Distro/vlc #92: VLC issue in CM4

ChayMahajj, 2023-11-20:
Hello, 

I am trying the cross-compile and  install VLC from source code for CM4. 
Unfourtunatelly VLC is not working on CM4 : No video output and the command line  return these errors : 
```
[018604f8] main decoder error: buffer deadlock prevented
[a28013f0] gl vout display error: parent window not available
[a28013f0] gles2 vout display error: parent window not available
[a28013f0] xcb vout display error: window not available
```
Description : I am using CM4 with Bullseye and Kernel 6.1
VLC steps I followed to build and install VLC : 

1. `./bootstrap`
2. `./configure  --enable-a52 --enable-aa --enable-aribsub --enable-avahi --enable-bluray --enable-caca --enable-chromaprint --enable-chromecast --enable-dav1d --enable-dbus --enable-dca --enable-dvbpsi --enable-dvdnav --enable-faad --enable-flac --enable-fluidsynth --enable-freetype --enable-fribidi --enable-gles2 --enable-gnutls --enable-harfbuzz --enable-jack --enable-kate --enable-libass --enable-libmpeg2 --enable-libxml2 --enable-lirc --enable-live555 --enable-mad --enable-matroska --enable-mod --enable-mpc --enable-mpg123 --enable-mtp --enable-ncurses --enable-notify --enable-ogg --enable-opus --enable-pulse --enable-qt --enable-realrtsp --enable-samplerate --enable-sdl-image --enable-sftp --enable-shine --enable-shout --enable-skins2 --enable-soxr --enable-spatialaudio --enable-speex --enable-srt --enable-svg --enable-svgdec --enable-taglib --enable-theora --enable-twolame --enable-upnp --enable-vnc --enable-vorbis --enable-x264 --enable-x265 --enable-zvbi --with-kde-solid=/usr/share/solid/actions/ --disable-aom --disable-crystalhd --disable-d3d11va --disable-decklink --disable-directx --disable-dsm --disable-dxva2 --disable-fdkaac --disable-fluidlite --disable-freerdp --disable-goom --disable-gst-decode --disable-libtar --disable-macosx --disable-macosx-avfoundation --disable-macosx-qtkit --disable-mfx --disable-microdns --disable-opencv --disable-projectm --disable-schroedinger --disable-sparkle --disable-telx --disable-vpx --disable-vsxu --disable-wasapi --disable-mmal-avcodec --enable-alsa --enable-dc1394 --enable-dv1394 --enable-libplacebo --enable-linsys --enable-nfs --enable-udev --enable-v4l2 --enable-wayland --enable-vcd --enable-smbclient --disable-oss --disable-mmx --disable-sse --enable-neon --enable-mmal --disable-vdpau --disable-libva --disable-altivec --enable-omxil --enable-omxil-vout --enable-rpi-omxil --enable-run-as-root'`
3. `make `
4. `make install`

Note:

- I want to use VLC without GUI, and  I don't use X server in my system 
- I already tested VLC installed from deb package using  : `apt-get install vlc ` and it is running well.

Thank you for your help.

Regards,
Chayma.

jc-kynesim, 2023-11-20:
I guess my 1st questions would be why don't you want to use the supplied working binaries?

You say cross-compile. Is that a typo? The configure above looks like what I would expect for a native compile. If you really mean cross-compile what is your host system?

Have you tried configuring without that massive list of options? Given you have no windowing system at least --enable-qt and --enable-wayland are pointless.

Are you sure you are running the code you've compiled? Make sure you've got rid of anything you installed with apt before trying to install your new version.

ChayMahajj, 2023-11-21:
Hello,

Why not working with the provided binaries beacause I have to add some events in VLC code to be linked with my Code moreover I have to use VLC dbus as system interface not session interface.

Yes I am crosscompiling it in my Host I export the envirnment path befor edoing all of that Am I wrong ? 
Th ebinary should be compiled for ARM7v Raspberry HW my Host is a Desktop Debian11.

I tried yes but with no Video output (in the other hand I get the audio autput)
What I have noticed from the dep package binary which is working that it uses drm_vout. 

I have two different images one with vlc from deb package and one compiled and installed from source code.
HW : Raspberry CM4 - Bullseye - Kernel 6.1 

I would apprerciate your help.

jc-kynesim, 2023-11-21:
Nothing wrong with env variable X-compile.

Sure. Yes - you want to make sure drm_vout is compiled & built. My 1st suggestion would be to compile the tree natively on the Pi (having enabled apt sources and done apt build-dep vlc) - it doesn't take that long on a Pi4 and see if that works better - sysroot creation has been my most common cause of cross-compile failure and the configure script is very opaque in what it enables & why.

Check you've built modules/.libs/libdrm_vout_plugin.so in whatever you are using as an output directory.

Having built it I'd recommend a test line like ./vlc -Irc -V drm_vout -vv mystream.mp4 and report back with the debug.

ChayMahajj, 2023-11-21:
Hello 
Regarding the cross-compilation is there a flag should be enabled to define the compiler ?  ( am asking about another why to do th ecross compilation ) 

I am already using apt build-dep vlc command to install all dependencies.

Unfourtunately after the compilation finished there is no libdrm_vout_plugin.so file

jc-kynesim, 2023-11-21:
Not sure. I used to X-compile vlc & ffmpeg back in Pi2 days as it was so much faster that some effort and confusion over sysroots was worth it. These days I compile on a Pi4 (I'd recommend 4 or 8G) as being much more likely to give me results I need.

ChayMahajj, 2023-11-21:
well noted.
for now I don't find  libdrm_vout_plugin.so
shall I compile with ./configure only or there is some important flags should be enabled ? 
Any tips ? 

Thank you for your help.

jc-kynesim, 2023-11-21:
The primary system dependency for libdrm_vout is libdrm; check you've got /usr/lib/aarch64-linux-gnu/libdrm.a installed in your environment (adjust path as required for your compile).

On a Pi ./configure works for me. I add a few flags but none of them affect building drm_vout. Once you've got that built successfully then add some flags.  Make a native compile work 1st then go for X-compile if you feel the need.

ChayMahajj, 2023-11-21:
I checked for libdrm which is no installed so I install it in my environment (RPI) apt-get install libdrm-dev
I got these files under /usr/lib/arm-linux-gnueabihf
```
libdrm.so                  libdrm_freedreno.so.1.0.0
libdrm.so.2                libdrm_nouveau.so
libdrm.so.2.4.0            libdrm_nouveau.so.2
libdrm_amdgpu.so           libdrm_nouveau.so.2.0.0
libdrm_amdgpu.so.1         libdrm_omap.so
libdrm_amdgpu.so.1.0.0     libdrm_omap.so.1
libdrm_etnaviv.so          libdrm_omap.so.1.0.0
libdrm_etnaviv.so.1        libdrm_radeon.so
libdrm_etnaviv.so.1.0.0    libdrm_radeon.so.1
libdrm_exynos.so           libdrm_radeon.so.1.0.1
libdrm_exynos.so.1         libdrm_tegra.so
libdrm_exynos.so.1.0.0     libdrm_tegra.so.0
libdrm_freedreno.so        libdrm_tegra.so.0.0.0
libdrm_freedreno.so.1
```
Under VLC installation unfourtunatlly there is no libdrm_vout_plugin.so under vlc/modules/.libs

Do you have any idea on how to make vlc build the drm_vout plugin ?

ChayMahajj, 2023-11-22:
Hello, 
It is the right question hhh. I didn't know about the patch.  There is no drmu in modules/video-output So.
Where is the patch that should be applied ?

jc-kynesim, 2023-11-22:
This is a debian distribution archive so the main tree is just the straight mainline source. Apply the patches in debian/patches to get what is actually built for the pi distro. Or lookup building debian packages on google.

ChayMahajj, 2023-11-22:
Hello 
After aplying the patch VLC works as magic.
Thank you for your support.

closed 2023-11-24 by ChayMahajj