RPi-Distro/vlc #96: VLC on the Raspberry Pi 4 (64-bit) playing YouTube freezes and buffers after 1-7s with an ffmpeg error.

JustAnEric, 2023-12-10:
VLC log:
```
pi@raspberrypi:~ $ vlc https://www.youtube.com/watch?v=sl09B87-SAY

VLC media player 3.0.20 Vetinari (revision 3.0.20-0-g6f0d0ab126b)
[00000055acac37f0] main interface error: no suitable interface module
[00000055ac9dc560] main libvlc error: interface "globalhotkeys,none" initialization failed
[00000055ac9dc560] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
[00000055acac37f0] skins2 interface error: cannot initialize OSFactory
[00000055acac37f0] [cli] lua interface: Listening on host "*console".
VLC media player 3.0.20 Vetinari
Command Line Interface initialized. Type `help' for help.
> [0000007f8c358000] lua stream error: Couldn't descramble YouTube throttling URL parameter: data transfer will get throttled
[0000007f78003420] gles2 generic error: parent window not available
[0000007f78003420] xcb generic error: window not available
[0000007f78001cc0] mmal_xsplitter vout display: Try drm
[0000007f78003420] drm_vout generic: <<< OpenDrmVout: Fmt=DPV0
[0000007f78003420] drm_vout generic error: Failed to get xlease
[0000007f8c627430] avcodec decoder: Using DRM Video Accel for hardware decoding
[0000007f98006050] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1000 ms)
[0000007f98006050] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1216 ms)
[0000007f8c627430] main decoder error: buffer deadlock prevented
[0000007f98006050] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1397 ms)
[0000007f8c627430] main decoder error: Timestamp conversion failed for 12133333: no reference clock
[0000007f8c627430] main decoder error: Could not convert timestamp 0 for FFmpeg
[0000007f98006050] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1901 ms)
[0000007f8c627430] main decoder error: buffer deadlock prevented
^C[00000055acac37f0] [cli] lua interface error: Error loading script /usr/lib/aarch64-linux-gnu/vlc/lua/intf/cli.luac: lua/intf/modules/host.lua:286: Interrupted.
[0000007f8c3d85f0] main tls client error: connection error: Resource temporarily unavailable
[0000007f8c3d85f0] gnutls tls client error: TLS handshake error: Error in the push function.
[0000007f8c3d85f0] main tls client error: TLS session handshake error
[0000007f8c3d85f0] main tls client error: connection error: Network is unreachable
[h264_v4l2m2m @ 0x7f8c651780] Failed to get coded packet: err=-1094995529
```
I am on the latest version of VLC, and trying to play a 1080p (HD) video from YouTube.

JustAnEric, 2023-12-12:
@jc-kynesim Alright, thanks! I will definitely try that out.

sudo-splinter-cell, 2023-12-26:
This is not a bug. Youtube itself blocks any third party requests coming from apps other than its own official app for playback and basically just limits the throughput to those ip addresses which ends up in buffering issues. There are some tricks that people came up with to get around it but it is cumbersome. So vlc for youtube not recommended.

JustAnEric, 2024-01-04:
@sudo-splinter-cell And it is really annoying that YouTube does this.

popcornmix, 2024-01-04:
Kodi's youtube addon works without buffering (it uses some of the tricks to avoid throttling).

JustAnEric, 2024-01-04:
@popcornmix It uses an API key.

sudo-splinter-cell, 2024-01-04:
I suggest Freetube.

[https://freetubeapp.io/](url)

This is what i use to play youtube videos on my Pi-5. It works perfectly. And you can subscribe to channels and even backup your subscriptions, indepentend from the Youtube api. Don't even need a google account.

JustAnEric, 2024-01-05:
@sudo-splinter-cell Yeah I tried FreeTube. But I tried it on my Raspberry Pi 4. It keeps buffering at 720p and higher.

sudo-splinter-cell, 2024-01-06:
> @sudo-splinter-cell Yeah I tried FreeTube. But I tried it on my Raspberry Pi 4. It keeps buffering at 720p and higher.

On RPI-4, you need to select "legacy codecs" in player settings because rpi 4 is not powerful enough for dash format. 60fps videos will suffer. If you upgrade to Rpi 5, you can go with dash 1080 and even 60fps.

sudo-splinter-cell, 2024-01-25:
> @sudo-splinter-cell Where can I find the Player Settings for FreeTube?

Seriously? There is a button on the left column with "settings" written on it.