RPi-Distro/vlc #97: Installing and removing `vlc` apt package causes permanent changes to sound configuration.

tonicsoft, 2023-12-11:
Environment: Raspberry pi 3B, RaspiOS Lite bookworm (latest release as of 9th December 2023)

Issue:

1. Plug in headphones using the built in 3.5mm jack
2. Run `speaker-test -c2` and hear the white noise, confirming that the sound is working
3. install vlc with `sudo apt-get install vlc`
4. Run `speaker-test -c2` and observe that there is no sound
5. remove vlc with `sudo apt remove --purge vlc && sudo apt autoremove`
6. Run `speaker-test -c2` and observe that there is still no sound
7. Reboot the Pi and observe that the problem is still not fixed.

Some background on my use case: I am trying to use `vlc` to play audio files from a headless raspberry pi. I can't get it to work at all, but the steps above seemed like something specific enough that it might be worth reporting as a bug. I'm sorry if this is completely useless info or if vlc is simply not intended to run on the Lite version of the OS, I don't know anything about Audio.

jc-kynesim, 2023-12-12:
Honestly I don't know the answer to this but I'm going to guess that installing VLC installs a bunch of the desktop environment maybe including pipewire (I don't know if that turns up in Lite by default) and you are now getting pipewires default config which I'm going to guess defaults to something other than the 3.5mm jack. Uninstalling VLC then doesn't uninstall pipewire.  You can test for it being there with `pipewire -h` which will produce a simple help test if it exists. You can try changing the audio out with raspi-config.

tonicsoft, 2023-12-12:
Thanks for the help, especially the tip that you can configure audio with raspi-config.

It's a new day and I don't know what has changed but I can't reproduce the issue anymore, at least not from a completely freshly imaged SD card. I have a feeling that maybe I confused myself and it was in fact pulseaudio that caused the issues. Anyway, vlc appears to work out of the box, despite error messages about pulseaudio. No idea if this is a good idea to be using VLC in a headless environment, and yes a boatload of extra stuff gets installed (but not `pipewire`) but it's working :)

closed 2023-12-12 by tonicsoft