RPi-Distro/vlc #98: Playing 4K videos with VLC failed !

ChayMahajj, 2023-12-12:
Hello everyone,

I am trying to play 4K videos with VLC but vlc fails.
Command line : 
`cvlc vide_name.mp4`
Result - Output : 
```
[a57106e0] avcodec decoder error: more than 5 seconds of late video -> dropping frame (computer too slow ?)
[a57106e0] avcodec decoder error: more than 5 seconds of late video -> dropping frame (computer too slow ?)
[a57106e0] avcodec decoder error: more than 5 seconds of late video -> dropping frame (computer too slow ?)
[a57106e0] avcodec decoder error: more than 5 seconds of late video -> dropping frame (computer too slow ?)
[a57106e0] avcodec decoder error: more than 5 seconds of late video -> dropping frame (computer too slow ?)
[a57106e0] avcodec decoder error: more than 5 seconds of late video -> dropping frame (computer too slow ?)

```
I am using Compute Module 4 and Bullseye

Best regards,
Chayma.

jc-kynesim, 2023-12-12:
What video type? HEVC/H265, H264, AV1? The only one that is expected to work is H265 (being the only one with h/w accel).

ChayMahajj, 2023-12-12:
The video Type is H264
I got this Info from VLC in my desktop
[image]

jc-kynesim, 2023-12-12:
Then I'm not surprised it doesn't work. Pi 4 doesn't have the CPU power to do that and its h/w H264 decode is only good to HD (its basically the same block that was in a Pi 1). This falls into the realm of "can't fix".

ChayMahajj, 2023-12-12:
So teh video should be scaled to lower resolution or we can encode it to H265  so vlc can play it ? 
What do you think ?

Thank you for your help!

jc-kynesim, 2023-12-12:
H265 4k should generally work as long as you are using DRM output (windowed output will fail), HD H264 will also work.

ChayMahajj, 2023-12-12:
Thank you for this valuable Information.